prusa3d/Prusa-Firmware #3086: [QUESTION] confused about line endings in serial communication

akhtariev, 2021-03-26:
**What is your question?**
Hi!

I am communicating with ender's Marlin as well as prusa's Marlin via serial port. I noticed that Ender's sense `\n` which is understandable. However, I observed that I receive `\r\n` from Prusa. For example, for command end I get `ok\r\n` and can't understand the reason why?

I've been looking into the code for Prusa's firmware but can't find where it decides to inject LF. Could someone please explain any reason as for why I get the line feed and if it is possible to disable it?

I looked at #2984 . However, on my Linux machine, this part of the code is disabled with a macro.

Thanks so much in advance!

leptun, 2021-03-27:
Have you tried using the latest firmware 3.10.0-rc1? In there I’ve addressed this issue.
Here is where \r is inserted: https://github.com/prusa3d/Prusa-Firmware/pull/2984/files

akhtariev, 2021-03-29:
Thanks! I'll try that

akhtariev, 2021-04-01:
This helped. Thanks!