prusa3d/Prusa-Firmware #3087: Dutch translation in 3.10.0-RC1 firmware

nielssteenvoorde, 2021-03-27:
I'm now using your 3.10.0-RC1 firmware in my prusa i3MK3S printer and the dutch translation (Nederlands) doesn't show up under community made? Am I doing something wrong? Is there an explanation why it doesn't show up? I really was looking forward to this.

DRracer, 2021-03-29:
@nielssteenvoorde did flashing the FW end up correctly? Including languages?

3d-gussner, 2021-04-01:
@nielssteenvoorde Can you please give us an update on this issue, or if solved close it?

3d-gussner, 2021-04-10:
[stale]  @JohnnyDeer please close this issue as it got stale and there is no reaction from owner.

JohnnyDeer, 2021-04-12:
@nielssteenvoorde I'm closing this issue for not responding.

closed 2021-04-12 by JohnnyDeer

nielssteenvoorde, 2021-04-12:
> @nielssteenvoorde did flashing the FW end up correctly? Including languages?

yes the flashing of the firmware was without any errors. I mostly do it by interface octoprint (firmware update plugin). No problems there

nielssteenvoorde, 2021-04-12:
> [stale] @JohnnyDeer please close this issue as it got stale and there is no reaction from owner.

sorry, i was in hospital for a while ;-(

3d-gussner, 2021-04-12:
Hope you are doing well again.
Octoprint Firmware flasher plugin can flash MK2.x with multiple languages BUT not MK3/S/+ as the languages are stored in the external flash and the plugin doesn't do the 2nd flash process which is implemented in PrusaSlicer.
Please use PrusaSlicer to flash MK3/S/+.