prusa3d/Prusa-Firmware #3095: [QUESTION]Will color "Change in mins" require a PrusaSlicer update?

CWStevens, 2021-03-30:
3.10.0-RC1 currently reports "Change in mins: -1" for an active print, although I have 4 layer changes for my current print.

Messages received:
Recv: NORMAL MODE: Percent done: 4; print time remaining in mins: 663; **Change in mins: -1**
Recv: SILENT MODE: Percent done: 4; print time remaining in mins: 664; **Change in mins: -1**

Is this a feature that will require a PrusaSlicer (gcode) update to provide pending color change details?

3d-gussner, 2021-04-01:
@CWStevens The g-code `M73` got two new parameters to trigger the Prusa-Firmware to show "Change/Pause/next action" time.

I guess next PrusaSlicer release will automatically add these new parameters when a color change is added to inform the user as described in the release notes.
Existing g-code files do **NOT** have these parameters and so the firmware FW3.10.0+ shows **Change in mins: -1**.

The printer processor isn't resourceful enough to parse the hole g-code and calculate the time to next `M600/M601/...`.

If you need/want to try this new feature on existing g-code files, you gonna need to modify them manually:
1. Search for `M600` in gcode file
2. Look for first `M73` above the found `M600` and add to `M73` the parameters `Cxxx Dyyy` , where `xxx,yyy` are seconds to change
3. Add after the `M600` a new line `M73 C0 D0` to turn it off.
4. Search for 2nd, 3rd, 4th, etc `M73` and add the two parameters with seconds to go.
5. Repeat Steps 1-4 for the next `M600` you have in the existing g-code. 

Here the documentation for the g-code `M73` https://github.com/prusa3d/Prusa-Firmware/blob/v3.10.0-RC1/Firmware/Marlin_main.cpp#L6371-L6384

```
    ### M73 - Set/get print progress <a href="https://reprap.org/wiki/G-code#M73:_Set.2FGet_build_percentage">M73: Set/Get build percentage</a>
    #### Usage
    
        M73 [ P | R | Q | S | C | D ]

    #### Parameters
        - `P` - Percent in normal mode
        - `R` - Time remaining in normal mode
        - `Q` - Percent in silent mode
        - `S` - Time in silent mode
        - `C` - Time to change/pause/user interaction in normal mode
        - `D` - Time to change/pause/user interaction in silent mode
```

Hope this answers your question, and if so please consider to close this issue.

CWStevens, 2021-04-08:
Yes, that answers my question.  Thanks!

closed 2021-04-08 by CWStevens