prusa3d/Prusa-Firmware #3096: Using a Prusa Mini Filament Sensor on a Mk3s

NickRno77, 2021-03-31:
Please, before you create a new question, please make sure you searched in open and closed issues and couldn't find anything that matches.

**What is your question?**
I’m using a E3D Hemera extruder which doesn’t have a filament sensor, I trying to use a Prusa Mini Filament sensor. Trouble is when filament is loaded the switch is set to 0 so I need to toggle this on the Firmware. I could redesign the filament sensor housing but ideally I’d like to do this in FW

Can somewhere point where and what to change in the firmware. 

Thanks

3d-gussner, 2021-04-01:
Duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/3047
@NickRno77 Please consider to close this issue.

NickRno77, 2021-04-01:
> Duplicate of #3047
> @NickRno77 Please consider to close this issue.

Ok I’ll update and close when later, thanks.

NickRno77, 2021-04-01:
Couldn't find a switch to Invert in MK3s FW... apparently there is an Invert switch in Prusa Mini FW. I give up with FW so I'll just remake the sensor enclosure to work like MK3s with no magnets,ball and lever.

closed 2021-04-01 by NickRno77