prusa3d/Prusa-Firmware #3102: [BUG] [3.10.0-RC1] SD card print initiated from OctoPrint will start again upon completion and crash

kazibole, 2021-04-06:
**Printer type** - MK3S
**Printer firmware version**-  3.10.0-RC1

**MMU Upgrade** - N/A

**SD card or USB/Octoprint**
  SD card print initiated from OctoPrint (gcode stored on printer's SD card)

**Octoprint Version** 1.5.3

**Describe the bug**
  When the SD card print completes, it immediately starts again and will crash on homing due to the purge line and part still on the bed.
  
**To Reproduce**
  Gcode is stored on printer's SD card. Connect OctoPrint to printer. Fron OctoPrint choose gcode from printer's SD card and click "load and print". SD card print will start and complete successfully. However immediately upon completion the SD card print will start again and crash on homing due to the purge line and part still on the bed. This does not occur on firmware 3.9.3.
  
**Expected behavior**
  SD card print does not start again without user intervention.
  
**G-code**
  https://gist.github.com/kazibole/ed3b0332cb7eca2e5e95c2254c90b794

**OctoPrint Serial Logs**
https://gist.github.com/kazibole/ed3b0332cb7eca2e5e95c2254c90b794

Note for 3.10.0-RC1.txt log: I pressed the printer's reset button when the print started again but before the crash.

JohnnyDeer, 2021-04-19:
Hi @kazibole, thank you for reporting. Can you please try it on 3.10.0-RC2?

kazibole, 2021-04-19:
@JohnnyDeer I tested with 3.10.0-RC2 - same behavior (i.e. not fixed).

JohnnyDeer, 2021-04-29:
This bug will be fixed in 3.10.0 final.

kazibole, 2021-05-07:
@JohnnyDeer @leptun I confirmed that this is resolved in 3.10.0 final.

PR #3137 (2021-04-29): Do not allow M23 to run subroutines
Fixes #3102 
PFW-1239
closed 2021-05-07 by kazibole