prusa3d/Prusa-Firmware #3106: [BUG] With crash detection off fails to auto home if Y=max before the print starts. It grinds the servos...

kozmaz87, 2021-04-07:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

MK3S
browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/89.0.4389.114 Safari/537.36
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 3959304192
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
printer.firmware: Prusa-Firmware 3.9.3 based on Marlin

**Describe the bug**
  I had to turn off crash detection because it was false alarming in Y axis I am yet to find out why. This however also seems to affect the auto home.
  
**To Reproduce**
 Push the bed to home position back to Y=0 the printer prints as expected everything works. At the end of print the print is offered to you as normal by moving to X=0 Y=Max. Start next print and the printer starts the auto home procedure in the back never pulls the Y axis forward, Not sure if this is the fw bug but it started acting up with the crash detection shortly after I updated fw... if it is a hardware thing I must look at I apologise for this ticket.
  
**Expected behavior**
  It should auto home from the home position not the back corner on the start of the print.
  
**G-code**
  It was sliced using prusa slicer and I did not tamper with the settings of custom gcode recently. The only thing I added there is the long intro line because that makes it way more reliable for PETG filaments that love to ooze and blob. Gcode is attached. It is a short print,

kozmaz87, 2021-04-07:
[Endpiece_0.2mm_PETG_MK3S_12m.zip](https://github.com/prusa3d/Prusa-Firmware/files/6275089/Endpiece_0.2mm_PETG_MK3S_12m.zip)

JohnnyDeer, 2021-04-12:
Hi @kozmaz87, does this happen to you only with printing via octoprint or also from SD? Does your printer pass the selftest successfully?

wavexx, 2021-04-17:
I might actually have fixed this indirectly in commit d2be40491b62ea2169c431a734a48319ff0940f7 where in some skew/offset configurations you can get a false crash detection if X or Y is also at the max position.

This should be available in the next 3.10 release

kozmaz87, 2021-04-20:
@JohnnyDeer I have not tried the SD print in a while to be honest.

kozmaz87, 2021-04-20:
@JohnnyDeer I have not tried the self-test but I will.

JohnnyDeer, 2021-07-29:
@kozmaz87 have you tried latest FW? Or have any updates on this?