prusa3d/Prusa-Firmware #3114: [ENHANCEMENT] Confirm filament type on automatic load, even if at previously set temp.

ggppjj, 2021-04-19:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - MK3S

**Is your enhancement related to a problem? Please describe.**
When swapping filament with Autoload enabled, I will often forget that the unload temps are currently active and the printer will automatically load filament without checking for filament type to change temps.
  
**Describe the solution you'd like**
Upon loading filament, I'd like a way to have the printer confirm the filament type that I'm loading, even if the printer is currently at-temp for, for example, PLA from a recent unload. 

A preferred sequence of user steps would be as follows:
1. Unload filament from "unload" menu
2. Insert new filament into load area
3. Printer display asks for confirmation of either current temps, or to change to one of the other standard temps, as it would if attempting to load cold, default option being to confirm current temps and continue to load.

hajosc, 2021-06-14:
I'd consider this a bug rather than an enhancement...

ggppjj, 2021-06-14:
@hajosc For clarification, do you mean that my proposed change to confirm filament temp on load when currently preheated would be considered buggy behavior if implemented, or that the current behavior of not confirming temps when loading is?

hajosc, 2021-06-14:
Sorry for not being clear -- I consider it a bug that the printer does not confirm temperature when loading a new filament. If you switch from "coldest" to "hottest" filament the loading process might very well fail.
So I think the issue should be tagged as "bug", not only as "enhancement"

ggppjj, 2023-11-01:
While the issue is stale, it is still open for consideration for future firmware improvement.

arekm, 2023-11-04:
Also just noticed that bug. 

How to reproduce:

- load filament, like PLA, select cooldown
- click unload filament, select PLA; it will heat up nozzle and ask you to remove filament after pushing knob; temperature stays at PLA level
- now put ABS filament into extruder. Filament sensor will detect it and extruder will pull in filament at PLA temperature which is incorrect.

It should ask what filament type is being loaded and heat to proper temperature. 

Tested on MK3S+ with 3.13.2.

ggppjj, 2024-01-10:
Hello! Please don't automatically close this issue, actions bot!

It may be old, but it's still relevant.

3d-gussner, 2024-03-12:
Thanks for the idea and sorry for answering so late.

Some (I would say most) users may find it annoying to be asked each time which filament type they want to load as they print mostly the same type of material. So it will not be easy to satisfy everyone.

Workflow idea for you:
When you switch from one material to another
- unload existing filament
- Select preheat of new material
- Auto/load filament

@arekm 
> load filament, like PLA, select cooldown
> click unload filament, select PLA; it will heat up nozzle and ask you to remove filament after pushing knob; temperature stays at PLA level

At this moment you tell the printer that the active filament is PLA to be unloaded

> now put ABS filament into extruder. Filament sensor will detect it and extruder will pull in filament at PLA temperature which is incorrect.

As most users change the same type of filament the printer at the moment assumes that you have PLA selected.

Keep also in mind that the un-/ load procedures are used during filament runouts, filament changes and MMU.

@hajosc 
>Sorry for not being clear -- I consider it a bug that the printer does not confirm temperature when loading a new filament. If you switch from "coldest" to "hottest" filament the loading process might very well fail.
So I think the issue should be tagged as "bug", not only as "enhancement"

It is NOT a bug as it works for the most used scenario, where users print mostly with the same material.

Changing materials needs extra attention in my opinion. Here two examples with most extreme temperature differences.

Change from PLA to PC:
As the temperature of PC is way higher than PLA you should try to clean the nozzle as clean as possible to prevent PLA filament cooking. One way is
- perform a "cold pull" to get most of the PLA out of the system
- Preheat to PC
- Load PC filament
- extrude few extra cm of PC to flush any PLA left overs out of the system.

Change from PC to PLA:
Same here, if you don't "clean" the nozzle from PC and after an unload you still may have some PC inside. But here you run a risk of a clog as this material will not melt at PLA temperatures. My approach for changing from high temp to low temp material is
- perform a cold pull
- set the temperature to PC temp
- wait to reach the temp
- select cool down
- open idler
- push PLA manually very fast to try flush any PC that may have left in the nozzle while it is cooling down
  - don't stop flushing the nozzle until PLA temp is reached. You don't want to cook the PLA
- perform an additional PLA cold pull to be "sure" i got everything out of the system
- load PLA
- ready to print PLA 

As this an edge case which most users may not use it we will not implement it. The workflow above should work for everyone just one additional manual step.

If a volunteer is willing to work on it and provide a PR we may consider to review this PR.
Please also don't forget my approach of changing from Low to High and back, but this would probably cost a lot of resources and the chances are low to get it reviewed or merged.

arekm, 2024-03-12:
I meantime I've switched to MK3.5 (so Prusa-Firmware-Buddy) and it
* remembers what filament was loaded, even after printer reset
* when clicking unload it doesn't ask for filament type (due to remembering it) and just heats up and unloads
* after unloading, when inserting filament with filament sensor detection (or choosing load from menu) it asks for filament type (so doesn't assume that you load the same filament)

No idea if the same procedure is used everywhere (mmu etc) like in 8bit firmware though. Also I'm ok with "wontfix" at this stage of 8bit fw life cycle coming (sooner or later) to the end.

3d-gussner, 2024-03-12:
@arekm Thanks for the feedback. The 32-bit is different and has way more resources to do things and there might be a different approach to this.

I think that especially switching from high temp material to lower temp needs several manual steps to prevent clogs.