prusa3d/Prusa-Firmware #3119: [ENHANCEMENT] Can you make the default/idle z-height just a little higher? 10mm or so?

rw3iss, 2021-04-21:
Prusa 2.3.0-64 linux
MK3S

This is a request to make the default/idle Z-height about 10mm higher, for multiple reasons:

If the nozzle is oozing, we generally want to clear it right before it starts printing... I do this with a quick wipe of the wire brush (see attached image). I can't fit the standard smallest width wire brushes under it... well just barely, but it's quite tight. Another 10-15mm would be perfect.
[image]

The second argument is when inevitable blobs of failed prints get caught under the print head... we typically want to stop print and pull them off, but there's barely any more room to do that at the idle height, and going into settings to move the Z axis is tedious all the time.

Please! Or maybe someone knows a way to edit the Gcode to do this?

wavexx, 2021-04-27:
I feel this. I also modified the default "stop/pause height" to be minimum 50mm higher than the print, simply because I need more space to remove the steel sheet, and just like you say, I had to raise Z almost every time.

Anybody else having the same need?

rw3iss, 2021-04-27:
> I feel this. I also modified the default "stop/pause height" to be minimum 50mm higher than the print, simply because I need more space to remove the steel sheet, and just like you say, I had to raise Z almost every time.
> 
> Anybody else having the same need?

Nice. Did you modify the default G-Code and have to re-flash the firmware? Or can you just modify it with PrusaSlicer or elsewhere?

Panayiotis-git, 2021-04-28:
This is one of the reasons I like the skelestruder: it provides plenty of room to access the nozzle.

wavexx, 2021-04-28:
> Nice @wavexx . Did you modify the default G-Code and have to re-flash the firmware? Or can you just modify it with PrusaSlicer or elsewhere?

It's done in multiple steps. I have this in the end-gcode:

```
{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+50, max_print_height)}{endif} ; Move print head up
```

For pause/stop, it's currently defined in the FW, so it needs a couple of simple changes (https://github.com/wavexx/Prusa-Firmware/tree/stop_raise_higher is one)

DaReaLDeviL, 2021-05-04:
I also whould prefer a higher standard value. Great suggestions.

rw3iss, 2021-12-27:
Hey Prusa, really, can you just make the default height higher, in all cases? It causes no issue at all to wait an extra moment to lower it during print, and would relieve a lot of pain not having to raise it manually to mess with the printhead, which is so often! 

I'm currently going through printing 36 individual pieces, and after each print, the printhead is sitting almost touching the printbed, and oozing. I can't start the next print without manually raising it and cleaning it. It's quite annoying, and would be a lot easier if it was just 10mm higher. 20mm would even be great. This might even be a nice setting custom setting to define in the Software or somewhere else!

Happy holidays :)

closed 2023-11-09 by github-actions[bot]