prusa3d/Prusa-Firmware #3122: [BUG] Changing feedrate during pause should affect the paused print *only*

wavexx, 2021-04-22:
**Printer type** - MK2.5+
**Printer firmware version**-  3.9+

**MMU Upgrade** - n/a
**MMU upgrade firmware version** - n/a

**SD card or USB/Octoprint**
n/a

**Describe the bug**
When the print is paused, changing the feedrate (either through the Tune menu, or by rotating the encoder from the status screen) should change the feedrate of the saved print, not the feedrate of the paused state.
  
**To Reproduce**
- Start print
- Pause print
- Change the feedrate (rotate the encoder)
- Resume print: resume moves are *incorrectly* affected (bug)
- New feedrate is lost as soon as printing resumes (bug)

**Expected behavior**
Feedrate set while paused should set the feedrate of the saved print _only_.

3d-gussner, 2023-11-07:
@wavexx Do you think this is still relevant?

gudnimg, 2023-11-12:
I created a PR for this. @wavexx could you take a quick look and check if my PR covers what you had in mind?

`feedmultiply` is updated for the UI, but this also makes the setting take effect immediately. I'm not sure if this is contradicting your description:

> Feedrate set while paused should set the feedrate of the saved print _only_.

commit ddbadf48 (2023-11-12): Fix #3122

If the feedrate multiplier is changed while the print is paused
update the saved feedrate multiplier in RAM
commit e74abb10 (2023-11-12): Merge pull request #4494 from gudnimg/issue3122

Fix #3122
closed 2023-11-12 by 3d-gussner