prusa3d/Prusa-Firmware #3128: [BUG] filament autoloading stopped and not finished when SD card inserted

stepheneb, 2021-04-26:
- Original Prusa i3 MK3S kit
- firmware: 3.10.0-RC2

I started to autoload filament and halfway through the procedure I inserted an SD card. The system displayed message about sorting files and the ongoing autoload procedure was terminated before any filament had been extruded.

stepheneb, 2021-04-26:
Perhaps a related issue: #2651

JohnnyDeer, 2021-04-27:
Hi @stepheneb, I'm unable to reproduce this issue. Printer behave as expected, load filament and show dialog about successful load even if i stay in SD menu. If i choose file to print during load printer does reset.
Can you please uplad video with this issue? Thank you

awenelo, 2021-04-27:
@JohnnyDeer, I can cause something similar to this issue on 3.10.0-RC1, but the only thing that happens is that the `Loading filament` message disappears, the autoload still brings up the `Is filament extruding and with the right color?` menu and continues to run the extruder as expected.

Expected message during autoload:
[image]

Actual message during autoload after SD is inserted:
[image]

stepheneb, 2021-04-30:
@JohnnyDeer, I tried reproducing today and wasn't able to. The printer stays on 24x7 and is printing almost all that time. I pressed the reset button today before I tried to replicate.

I will try a few more times.

JohnnyDeer, 2021-07-19:
@stepheneb did you reproduce this issue since? If no could we close it as "unable to reproduce"?

JohnnyDeer, 2021-08-19:
I'm closing this issue because there are no responses.

closed 2021-08-19 by JohnnyDeer