prusa3d/Prusa-Firmware #3129: [BUG] Removing/Inserting the SD card during heating removes the last 3 lines

awenelo, 2021-04-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.10.0-RC1 (Haven't had time to update yet, don't think that anything will ch

**SD card or USB/Octoprint**
SD card

**Describe the bug**
Removing or inserting the SD card during the heating message triggered by autoload causes the bottom of the screen to disappear. They re-appear when the screen switches to the `Click wheel to load filament message`. Similar behaviour to that described in #3128, only it doesn't stop anything in the printer, just causes graphical issues.
  
**To Reproduce**
1. Insert SD card
2. Insert filament
3. Select filament temperature preset
4. Wait for the `Preheating to load` and `>Cancel` messages to both be on the screen
5. Remove SD card
6. `Preheating to load` and `>Cancel` messages disappear from the screen
7. Click wheel
8. Heating cancels
---
9. Remove and re-insert filament
10. Select filament temperature preset
11. Wait for the `Preheating to load` and `>Cancel` messages to both be on the screen
12. Insert SD card
13. Wait for SD menu to appear
14. Click "Back"
15. `Preheating to load` and `>Cancel` messages are no longer on the screen
  
**Expected behavior**
`Preheating to load` and `>Cancel` messages remain on the screen, until it reaches the expected temperature unless heating is cancelled.

**Picture**
[image]

leptun, 2021-04-27:
@awenelo I think It’s a bug with the waiting for temperature message which was recently fixed it in one of @wavexx’s PRs.

wavexx, 2021-04-27:
Yes, looks another way trigger #3120
Hopefully it's already fixed in the current branch

awenelo, 2021-04-27:
@wavexx, compiling and flashing [wavexx/Prusa-Firmware/fix_lcd_redraw_autoload](https://github.com/wavexx/Prusa-Firmware/tree/fix_lcd_redraw_autoload), this still happens.

wavexx, 2021-04-27:
Nasty... ok, going to look into that.

wavexx, 2021-04-28:
@awenelo could you check https://github.com/prusa3d/Prusa-Firmware/pull/3134 ?

awenelo, 2021-04-28:
@wavexx #3134 fixes this issue.

JohnnyDeer, 2021-07-19:
I´m closing this issue as solved by mentioned #3134.

JohnnyDeer, 2021-07-19:
I´m closing this issue as fixed by mentioned #3134.

closed 2021-07-19 by JohnnyDeer