prusa3d/Prusa-Firmware #3131: Extruder head continues to move after 'stop print'

michgsy, 2021-04-28:
Started to notice an issue with my mk3s. I select 'stop print' part way through a print, and the print stops, but the extruder head does not go home. Instead, it stays where it is and stops moving/extruding. The next time the extruder head is prompted to move - either through a manual movement of the z-axis, or starting the next print, the extruder head will print the next few movements of the print that was stopped, stopping, lifting and homing after about 5 seconds, almost as if the print gcode has gotten stuck in its memory somehow

wavexx, 2021-04-28:
Which firmware version and on what printer this happens?

michgsy, 2021-04-29:
Latest firmware (V3.9.3 07/01/21) on an MK3S

Prusa-Support, 2022-06-13:
Hello! Sorry for intervening on this so late. Was this resolved on the subsequent firmware released in the meantime? Let me know, in case of current issues with this we'll keep looking into it. 

Alessandro Pantaleo
Prusa Research

closed 2023-11-09 by github-actions[bot]