prusa3d/Prusa-Firmware #3133: [QUESTION] Getting compile errors on linux? Have gawk installed.

rw3iss, 2021-04-28:
Am on Ubuntu 20.10.
I just cloned the repo, installed gawk:
`sudo apt install gawk`

Then
`./build.sh`

...but it's giving this... any ideas what I could be missing? Bad JAVA install?

> rw3iss@G14:~/Sites/prusa-firmware-original$ ./build.sh
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
In file included from sketch/w25x20cl.c:6:0:
sketch/w25x20cl.c: In function 'w25x20cl_init':
sketch/io_atmega2560.h:359:20: error: '__DDR_PW25X20CL_PIN_CS' undeclared (first use in this function)
 #define __DDR(pin) __DDR_P##pin
                    ^
sketch/io_atmega2560.h:363:18: note: in expansion of macro '__DDR'
 #define DDR(pin) __DDR(pin)
                  ^
sketch/io_atmega2560.h:366:22: note: in expansion of macro 'DDR'
 #define PIN_OUT(pin) DDR(pin) |= __MSK(pin)
                      ^
sketch/w25x20cl.c:48:2: note: in expansion of macro 'PIN_OUT'
  PIN_OUT(W25X20CL_PIN_CS);
  ^
sketch/io_atmega2560.h:359:20: note: each undeclared identifier is reported only once for each function it appears in
 #define __DDR(pin) __DDR_P##pin
                    ^
sketch/io_atmega2560.h:363:18: note: in expansion of macro '__DDR'
 #define DDR(pin) __DDR(pin)
                  ^
sketch/io_atmega2560.h:366:22: note: in expansion of macro 'DDR'
 #define PIN_OUT(pin) DDR(pin) |= __MSK(pin)
                      ^
sketch/w25x20cl.c:48:2: note: in expansion of macro 'PIN_OUT'
  PIN_OUT(W25X20CL_PIN_CS);
  ^
sketch/io_atmega2560.h:354:20: error: '__BIT_PW25X20CL_PIN_CS' undeclared (first use in this function)
 #define __BIT(pin) __BIT_P##pin
                    ^
sketch/io_atmega2560.h:355:26: note: in expansion of macro '__BIT'
 #define __MSK(pin) (1 << __BIT(pin))
                          ^
sketch/io_atmega2560.h:366:34: note: in expansion of macro '__MSK'
 #define PIN_OUT(pin) DDR(pin) |= __MSK(pin)
                                  ^
sketch/w25x20cl.c:48:2: note: in expansion of macro 'PIN_OUT'
  PIN_OUT(W25X20CL_PIN_CS);
  ^
sketch/io_atmega2560.h:358:21: error: '__PORT_PW25X20CL_PIN_CS' undeclared (first use in this function)
 #define __PORT(pin) __PORT_P##pin
                     ^
sketch/io_atmega2560.h:362:19: note: in expansion of macro '__PORT'
 #define PORT(pin) __PORT(pin)
                   ^
sketch/w25x20cl.c:35:20: note: in expansion of macro 'PORT'
 #define _CS_HIGH() PORT(W25X20CL_PIN_CS) |= __MSK(W25X20CL_PIN_CS)
                    ^
sketch/w25x20cl.c:49:2: note: in expansion of macro '_CS_HIGH'
  _CS_HIGH();
  ^
In file included from sketch/w25x20cl.c:3:0:
sketch/w25x20cl.h:18:40: error: 'W25X20CL_SPCR' undeclared (first use in this function)
 #define W25X20CL_SPI_ENTER() spi_setup(W25X20CL_SPCR, W25X20CL_SPSR)
                                        ^
sketch/w25x20cl.c:50:2: note: in expansion of macro 'W25X20CL_SPI_ENTER'
[... 260 lines cut ...]
sketch/io_atmega2560.h:358:21: error: '__PORT_PW25X20CL_PIN_CS' undeclared (first use in this function)
 #define __PORT(pin) __PORT_P##pin
                     ^
sketch/io_atmega2560.h:362:19: note: in expansion of macro '__PORT'
 #define PORT(pin) __PORT(pin)
                   ^
sketch/w25x20cl.c:34:20: note: in expansion of macro 'PORT'
 #define _CS_LOW()  PORT(W25X20CL_PIN_CS) &= ~__MSK(W25X20CL_PIN_CS)
                    ^
sketch/w25x20cl.c:170:2: note: in expansion of macro '_CS_LOW'
  _CS_LOW();
  ^
sketch/io_atmega2560.h:354:20: error: '__BIT_PW25X20CL_PIN_CS' undeclared (first use in this function)
 #define __BIT(pin) __BIT_P##pin
                    ^
sketch/io_atmega2560.h:355:26: note: in expansion of macro '__BIT'
 #define __MSK(pin) (1 << __BIT(pin))
                          ^
sketch/w25x20cl.c:34:46: note: in expansion of macro '__MSK'
 #define _CS_LOW()  PORT(W25X20CL_PIN_CS) &= ~__MSK(W25X20CL_PIN_CS)
                                              ^
sketch/w25x20cl.c:170:2: note: in expansion of macro '_CS_LOW'
  _CS_LOW();
  ^
exit status 1

rw3iss, 2021-04-28:
@wavexx Worked like a charm, thank you. Maybe should update README to suggest trying PF-build.sh on Ubuntu/Debian systems - I see the note further down under WSL for Debian/Ubuntu to try PF-build.sh.
Cheers!

closed 2021-04-28 by rw3iss

wavexx, 2021-04-28:
``build`` should also work, but it doesn't do all the checking done by PF-build and it requires some customization to work correctly. Looking at the error message it almost looks like the wrong arduino environment or arduino board is being used to rebuild the FW.

But you're right: @3d-gussner maybe the README could be improved?

rw3iss, 2021-04-28:
@wavexx Hmm, well the build supposedly succeeds... but I don't see it output anywhere, strangely.

Here is the output second time running (first time it did a bunch, no errors):

> rw3iss@G14:~/Sites/prusa-firmware-original$ ./PF-build.sh
Linux 64-bit found

Script path : /home/rw3iss/Sites/prusa-firmware-original
OS          :
OS type     : linux

Arduino IDE : 1.8.5
Build env   : 1.0.6
Board       : prusa_einsy_rambo
Package name: PrusaResearch
Board v.    : 1.0.3
Specific Lib: PrusaLibrary

1) Firmware/variants/1_75mm_MK2-RAMBo13a-E3Dv6full.h	  6) Firmware/variants/1_75mm_MK2-RAMBo10a-E3Dv6full.h
2) Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h	  7) Firmware/variants/1_75mm_MK25S-RAMBo10a-E3Dv6full.h
3) Firmware/variants/1_75mm_MK25S-RAMBo13a-E3Dv6full.h	  8) Firmware/variants/1_75mm_MK25-RAMBo13a-E3Dv6full.h
4) Firmware/variants/1_75mm_MK3S-EINSy10a-E3Dv6full.h	  9) All
5) Firmware/variants/1_75mm_MK25-RAMBo10a-E3Dv6full.h	 10) Quit
Select a variant: 2

Which lang-build do you want?
1) Multi languages
2) English only
Select a language: 2
Nothing to clean up
 
Variant    : 1_75mm_MK3-EINSy10a-E3Dv6full
Firmware   : 3100-RC2
Build #    : 4104
Dev Check  : RC2
DEV Status : RC
Motherboard: BOARD_EINSY_1_0a
Languages  : EN_ONLY
Hex-file Folder: PF-build-hex/FW3100-RC2-Build4104/BOARD_EINSY_1_0a

 
English only language firmware will be built
 

Start to build Prusa Firmware ...
Using variant 1_75mm_MK3-EINSy10a-E3Dv6full
Sketch uses 233656 bytes (92%) of program storage space. Maximum is 253952 bytes.
Global variables use 6152 bytes of dynamic memory.

Copying English only firmware to PF-build-hex folder
1
 
 
Build done, please use Slic3rPE > 1.41.0 to upload the firmware
more information how to flash firmware https://www.prusa3d.com/drivers/ 

... but I don't see any new folders in the project directory, no PF-build-env directory. build-env only has the original PF-build-env-Linux64-1.0.6.1.zip file, nothing in lang folder... 

Any ideas where it is supposed to be built or what could confirm it isn't?
Perhaps some note in the PF-build.sh output to show the final file location would be helpful ;)

It does mention:
`ex-file Folder: PF-build-hex/FW3100-RC2-Build4104/BOARD_EINSY_1_0a`

.. but this folder doesn't exist or isn't getting created... I did create it manually with full permissions and ran again, but nothing is put there.

leptun, 2021-04-28:
Go to ~/Sites and it should be there in theory alongside the local clone of the repo

rw3iss, 2021-04-28:
> Go to ~/Sites and it should be there in theory alongside the local clone of the repo

Ahhhhhhhhhh :-))))) Thanks!

3d-gussner, 2021-04-28:
> But you're right: @3d-gussner maybe the README could be improved?

It is on the list of ToDos