prusa3d/Prusa-Firmware #3142: [ENHANCEMENT]M300 To Override 'SILENT MODE'

sarf2k4, 2021-05-07:
**MK3S**
**MMU2S**

**Problem/Issues**
M300 only works if the printer aren't set into 'silent mode' via octoprint terminal. I am able to produce beep/sound that I want if the printer aren't set into 'silent mode' in the settings menu. Generating a tone is a great way to notify users for an important event happened while still keeping mostly silent, for example, after print has finished, filament change sequence, etc
  
**Proposal**
It is great that the current printer's firmware able to filter out M300 commands when it is in 'silent mode'. However, an M300 issued command should able to override or bypass the 'silent mode' filter.

My thought is to have similar like this `M300 S<Hz> P<ms> F<Bool>` where F parameter is flag, when it is 0, it will respect 'silent mode' and will not generate any noise, if set to 1, it will ignore 'silent mode' and will play a tone.

3d-gussner, 2021-05-08:
Why would you set the printer to silent speaker and then edit lot of gcode files to overwrite it?
The `Silent` mode has been implemented to enable the user to choose if they want sound or not.

As a "hack" workaround you can modify the EEPROM.
- `D3 Ax0ed7 X00` to set it loud  
- `D3 Ax0ed7 X02`to set it silent.
See https://github.com/prusa3d/Prusa-Firmware/blob/v3.10.0/Firmware/eeprom.h#L216-L219

**WARNING: Modifying the EEPROM manually can break things.**

sarf2k4, 2021-05-08:
I didn’t edit a lot of gcode files to overwrite it, I set my printer to silent mode to avoid unwanted noise. I also would like to keep avoiding from modifying the eeprom values as what you said in your warning. It’s actually not to override the ‘sound mode’, the printer actually didn’t beep when an M300 issued manually via terminal when the printer in ‘silent mode’ only From: 3d-gussnerSent: Sunday, 9 May, 2021 4:11 AMTo: prusa3d/Prusa-FirmwareCc: sarf2k4; AuthorSubject: Re: [prusa3d/Prusa-Firmware] [ENHANCEMENT]M300 To Override 'SILENT MODE' (#3142) Why would you set the printer to silent speaker and then edit lot of gcode files to overwrite it?The Silent mode has been implemented to enable the user to choose if they want sound or not.As a "hack" workaround you can modify the EEPROM.D3 Ax0ed7 X00 to set it loudD3 Ax0ed7 X02to set it silent.See https://github.com/prusa3d/Prusa-Firmware/blob/v3.10.0/Firmware/eeprom.h#L216-L219WARNING: Modifying the EEPROM manually can break things.

closed 2023-11-09 by github-actions[bot]