prusa3d/Prusa-Firmware #3144: [BUG]<Enter comprehensive title>

andreapede, 2021-05-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S+
**Printer firmware version**-  3.10.0

**Describe the bug**
  After firmware update, if you select "italian" in the language menu, the message o copy language appears but loop in the change language settings. 
 With english works correctly. Didn't try other languages.

3d-gussner, 2021-05-08:
@andreapede How did you flash the firmware?
At this moment only PrusaSlicer Firmware upload will work correctly for Multi-language firmware on MK3/S/+

Octoprint and other firmware updater DO NOT flash the external flash of the EINSY where the multi-languages need to be so the firmware can copy them correctly.

3d-gussner, 2021-05-08:
If it works please close the issue.

andreapede, 2021-05-09:
> If it works please close the issue.

Doing from prusaslicer works perfectly. Ty

closed 2021-05-09 by andreapede