prusa3d/Prusa-Firmware #3145: [BUG]Missing LF/CR on Sensor Info Display

Rebel154, 2021-05-12:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

Mk3S 
Firmware 3.10.0

MMU2S
1.0.6

Usually Octoprint but occasionally SD card

**Describe the bug**

After updating to firmware 3.10.0, on the Support/Sensor Info menu display, there is a missing line feed/carriage return between PINDA 0 and FINDA 0
  
**To Reproduce**
 Select Support/Sensor Info from printer display
  
**Expected behaviour**
  
I would expect each sensor to be on its own line
  
**G-code**
Not applicable

**Video**
Photo attached
[image]

3d-gussner, 2021-05-12:
The Sensor support info screen has to work for MK3 and MK3S/+ that is why the PINDA and FINDA share the first line.
The MK3 "laser" sensor shows additional informations which need the 3rd 4th line of the screen.

So this is not a bug more a feature😉

Please consider to close this issue.

Rebel154, 2021-05-12:
OK, thanks for the explanation.

closed 2021-05-12 by Rebel154