prusa3d/Prusa-Firmware #3146: [BUG] unload function keep the temperature on also after filament is out

andreapede, 2021-05-13:
**Printer type** MK3S+
**Printer firmware version** 3.10.0
**MMU Upgrade** - [e.g. MMU2S, MMU2, MMU1]
**MMU upgrade firmware version [e.g. 1.0.6, 1.0.6-RC2, ...]

**Octoprint**
  connected but the operation is done manually on the printer.

**Describe the bug**
Selecting the unload function, after the temperature rose at the proper filament kind.
Pushing the control button and extracting the filament out on the beep.
Menu is correctly restored in the main screen but the bed and the nozzle temp remain at the same level.
It is necessary to reset or give the stop command from menu.

JohnnyDeer, 2021-05-17:
Hi @andreapede, I´m trying it and... settings -> unload filament -> confirm after temp reached -> filament unloaded from nozzle to MMU unit -> printer jumps back to settings and stays on temp. Same for 3.10.0, 3.9.3, with and without OctoPrint connected. Can you please describe step by step how this issue happen? Thank you

andreapede, 2021-05-17:
The steps are exactly the same settings -> unload filament -> confirm after
temp reached -> filament unloaded from nozzle to MMU unit ->  (ASKED FOR
PULL OUT THE FILAMENT) -> printer jumps back to settings and stays on temp the
temp rise up to the PETG temperature settings (even if PLA is set).
But  the temperature, after the unload operation, shouldn't be set to zero?
Otherwise i will only check which will be the target temperature after
filament removal.

JohnnyDeer, 2021-05-17:
No, printer stay on temp after unload in all FW versions as far I can remember. So I would not say this is bug. We can change this issue to feature request if you agree.

andreapede, 2021-05-17:
I see. You can modify to feature request. (I'll check the thing about the
target temperature just to be sure)
I still have the possibility to use an octoprint pluging,  but a native
timeout could be a good compromise.
Thank you.

andreapede, 2021-05-17:
Checked again. The temperature that continues to rise up is related to the plate (that if you start from environment temperature is slower than the nozzle) but the ending temperature is the targeted filament kind.
I think that your suggestion to change in new feature is good. 
It can be set a timeout to the heating in case no print started in some n minutes.
thank you.
Andrea

closed 2023-11-09 by github-actions[bot]