prusa3d/Prusa-Firmware #3147: [BUG] 3.10 Preheat menu loops when power on with loaded filament

Lyr3x, 2021-05-13:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [e.g. MK3S, MK3, MK2.5S, MK2.5, MK2S, MK2]
Mk3s+
**Printer firmware version**-  [e.g. 3.8.1, 3.8.1-RC1, ...]
3.10 / 3.9.3

**SD card or USB/Octoprint**
SD

**Describe the bug**
  When power on the printer with loaded filament the preheat menu is appearing again and again (inconsistent delay between). This only occurs if filament is loaded before power on the printer. As soon as i remove the filament once the issue is gone until the next power switch. Filament sensor is detected correct and the lever etc is working flawless. 
Flashed 3.9.3 back and the issue is gone. 

Tried to factory reset before and after flashing 3.10 with no success. Sensor voltages and states are correct. When unplugging the filament sensor the pop up does not appear. 

I cant identify a hardware issue as with 3.9.3 everything is working without any issues. The preheat menu appears also if autoload is disabled. 

When unloading and reloading the filament while the printer is powered on, the sensor is working as expected.
  
**To Reproduce**
1. Load filament
2. Power off the printer
3. Power on the printer
4. Preheat/Autoload Menu appears again and again
  
**Expected behavior**
1. Load filmanet
2. power off -> on
3. Nothing

If you need any more information, i am happy to provide them ASAP.

wavexx, 2021-05-18:
@Lyr3x can you make a video of the behavior?

If I understand correctly, the Preheat menu shows up, you *exit* the menu, but after a few seconds it pops up again?

Lyr3x, 2021-05-18:
Sure. Follow the link: https://photos.app.goo.gl/trkSfidprWqBfEJdA

I printed over the weekend with 3.9.3 without the issue. Flashed `prusa3d_fw_MK3S_MK3S+_3_10_0_4481.hex` for the video and as you can see, the issue is instantly present.

wavexx, 2021-05-18:
Might an issue with the fsensor version detection?

Start from reset with the filament unloaded. Under the support support, just under the date what does the "Fil. sensor v.:" say? Then go to voltages and note down the value. Then insert the filament just to trigger the sensor (ignore autoload - go back to voltages). What does it read now?

Then re-do the same, but with the filament inserted and reset. Detected version under support? Initial voltage?

Lyr3x, 2021-05-18:
**Reset with unloaded filament:**
Fil. sensor v.: 0.4 or newer
voltage unloaded: 4.4V
voltage loaded: 0.4V
IR Sensor state: 0

**Reset with loaded filament:**
Fil. sensor v.: 0.4 or newer
voltage loaded: 0.4V
IR Sensor state: 1
voltage unloaded: 4.4V

wavexx, 2021-05-18:
Looks good. I assume you don't see any weird fluctuations in voltage right if you leave it on for several seconds? (sadly, the display doesn't refresh fast enough to see fast peaks, but that would be a good indication).

Too me it the behavior would match if the was a random disconnect from the cable, maybe. Although why this changes with 3.9 is a bit weird but I need to look at a diff again to be sure.

I know it sounds stupid considering 3.9 works, but just as an obvious suggestion for the above, did you try to unplug/replug the sensor cable on both the board and extruder just to reseat the contacts? Connector or cable issues are always annoying.

Lyr3x, 2021-05-18:
The voltage of the IR unfortunately sensor stays stable at 0.4 or 4.4v. Any way to get more/faster debug output of via terminal?

>Too me it the behavior would match if the was a random disconnect from the cable, maybe. Although why this changes with 3.9 is a bit weird but I need to look at a diff again to be sure.

I absolutely thought the same and checked the wiring, measured continuity and unplugged the sensor (both at the sensor and on the MB). I also checked if with heating and with fans on something is changing. 
The fact that 3.9.3 is working fine (I flashed now 20+ times back and forth) doesn't look like a typical wiring issue. I am an engineer but not a C++ developer and do not understand the diff regarding the sensor, so i cant really support there.

I'll quickly check the wiring again and see if i can observe something new. I'll measure the sensor voltage by hand and see if i can see any fluctuation

wavexx, 2021-05-18:
Since you're going to do that, measure directly at the connector on the einsy and move the extruder cable bundle by hand (just in case the cable itself has issues). If you have a min/max feature on your meter you could catch a potential issue without having to look at the display.

Another question sprung to mind: so if insert the filament _after_ the printer has been turned on, then cooldown until all fans are off, does it also trigger randoly? Or it's only triggered from a cold-start with the filament pre-inserted?

Lyr3x, 2021-05-18:
We checked with 4 eyes: stable 4.39V and little more "unstable" 0.38-0.39v which looks fine for me and can even be my low cost meter. Nothing changes if moving the cable bundle. 

I do not even need to hot load the filament, just trigger the sensor. If filament is inserted AFTER the printer is turned on i don't see the issue. Only if I power on while the sensor is triggered. Does that make sense?

wavexx, 2021-05-18:
Yes it does, maybe some initialization is skipped. So if you poweron with the filament inserted (issue present), then remove and re-insert the filament the issue is also gone?

Lyr3x, 2021-05-18:
Exactly.

wavexx, 2021-05-18:
Ok this narrows down the issue a lot. For some reason I cannot trigger the issue, but with these set of conditions there are not many places where this can happen.

Lyr3x, 2021-05-18:
I am more than pleased if I can test anything (like changes on a branch) or provide info on short notice.

diluno, 2021-05-18:
I'm experiencing this problem as well! Already tried flashing the firmware again - now I guess I'll downgrade to V3.9.3.

Also, when I select a filament type in the preheat menu, it just keeps on "loading". It just doesn't stop oozing filament, even after selecting "Yes" when asked if it loaded correctly.

Lyr3x, 2021-05-18:
Just checked and after the second attempt that happened to me also. LCD went blank and extruder keeps extruding.

wavexx, 2021-05-18:
@Lyr3x  @diluno You don't have an MMU right?

@diluno I assume you also have a "v0.4 or newer" filament sensor type?

diluno, 2021-05-18:
Yes, I have a new (bought about 2 months ago) MK3S+ without MMU.

Lyr3x, 2021-05-19:
I have an MMU but not connected. My MK3S+ is also fairly new

Lyr3x, 2021-05-19:
I checked what changed in 3.10 vs 3.9.3 and switched back to SW i2c. https://github.com/prusa3d/Prusa-Firmware/blob/f5ca79926ee5b1fc887ff64852ad2448c6ad8c86/Firmware/config.h#L26-L27

that build is working fine for me again. So there might be an issue with the introduced hardware i2c mode. Maybe it helps you to narrow it further down

wavexx, 2021-05-19:
@Lyr3x could you instead leave it in HWI2C and change 

https://github.com/prusa3d/Prusa-Firmware/blob/f5ca79926ee5b1fc887ff64852ad2448c6ad8c86/Firmware/Marlin_main.cpp#L9733

to enable logging on the serial, and attach a log from the serial with the printer from reset and the filament missing, and one reset+present?

@leptun in SWI2C both SDA/SCL are set to output/1 mode, while in HW mode they're left in input with pullups, so they're floating. Are we picking up noise, or the adc settings are incorrect somewhere?

wavexx, 2021-05-19:
@leptun  The fact that it works correctly after the filament is removed makes me think more of the minVolt/maxVolt ranges being the same and starting low instead of the opposite.

Lyr3x, 2021-05-19:
Reset without filament:
```
start
echo: 3.10.0-4481
echo: Last Updated: May 19 2021 16:29:38 | Author: (none, default config)
Compiled: May 19 2021
echo: Free Memory: 2035  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Extruder fan type: NOCTUA
CrashDetect ENABLED!
FSensor ENABLED (sensor board revision: 0.4 or newer)
Sending 0xFF
echo:SD card ok
echo:busy: processing
echo:busy: processing
echo:busy: processing
LCD status changed
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.39V (min=4.39 max=4.39)
fs volt=4.38V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.38V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.38V (min=4.38 max=4.39)
fs volt=4.38V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.38V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.38V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.39V (min=4.38 max=4.39)
fs volt=4.38V (min=4.38 max=4.39)
```

Reset with filament:
```
start
echo: 3.10.0-4481
echo: Last Updated: May 19 2021 16:29:38 | Author: (none, default config)
Compiled: May 19 2021
echo: Free Memory: 2035  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Extruder fan type: NOCTUA
CrashDetect ENABLED!
FSensor ENABLED (sensor board revision: 0.4 or newer)
Sending 0xFF
echo:SD card ok
echo:busy: processing
echo:busy: processing
echo:busy: processing
LCD status changed
fs volt=0.43V (min=0.43 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.43V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
fs volt=0.42V (min=0.42 max=0.43)
```
But we have a problem here. I just figured that the issue is gone with my own build. Thats confusing to me.

wavexx, 2021-05-19:
Wouldn't be the first time I've seen weird issues depending on the compiler version. Did you use PrusaSlicer to flash it? (this is needed to *also* update the language data in a multilang fw).

@DRracer ideas?

Lyr3x, 2021-05-19:
I flashed the released version with PrusaSlicer on my M1 Mac and on my windows maschine. My own build only via Arduino, but testing prusa slicer now

wavexx, 2021-05-19:
Ok, this was just to exclude problems with the language data. Any version of PS should do

wavexx, 2021-05-19:
Since we're here, can you also try to build a multilanguage version and flash it? Use the PF-build.sh script, should be easy. Just to rule out a difference between the two (multilang takes more ram)

Lyr3x, 2021-05-19:
Here we go. Now the issue is back. I build with WSL2 as i cant build on my mac.  I'll re test with the IR debug output

Lyr3x, 2021-05-19:
With the debug code enabled the issue is gone again. Voltages are the same as above.

wavexx, 2021-05-20:
So just to be clear, you can reproduce it by rebuilding it in the mutilang build even without the debug, right? (so, exactly like the official build).

Lyr3x, 2021-05-20:
Yes. Tried it two more times yesterday.

wavexx, 2021-05-20:
@Lyr3x this is great ;), means we can reliably reproduce the issue.

Can you check the revision of your Einsy? It's written next to the powerpanic connectors, see this picture:

[image]

wavexx, 2021-05-20:
@Lyr3x It would be great if you also provide an EEPROM dump. If you issue a "D3" gcode on the serial, a complete dump will follow. Posting the complete dump here would be great.

Lyr3x, 2021-05-21:
There is actually no revision printed on, but only a tiny mini QR code. 

```
D3 - Read/Write EEPROM
0000  50 52 55 53 41 33 44 46 57 00 03 00 0a 00 00 00
0010  7f 00 ff ff 56 32 00 00 00 00 c8 42 00 00 c8 42
0020  00 00 c8 43 00 00 8c 43 00 00 48 43 00 00 48 43
0030  00 00 40 41 00 00 f0 42 e8 03 00 00 e8 03 00 00
0040  c8 00 00 00 88 13 00 00 00 40 9c 44 00 40 9c 44
0050  00 00 00 00 00 00 00 00 20 4e 00 00 00 00 20 41
0060  00 00 20 41 cd cc cc 3e 00 00 90 40 00 00 00 00
0070  00 00 00 00 00 00 00 00 cd cc cc 3e 5c 8f 7e 41
0080  0b 34 69 3e 50 75 8a 43 8f 42 fc 42 e6 5a 34 3f
0090  4c 62 b0 45 00 00 00 00 00 40 40 00 00 34 42 00
00a0  00 00 00 00 00 00 00 00 00 00 41 00 00 00 e0 3f
00b0  00 00 c8 42 00 00 c8 42 00 00 40 41 00 00 f0 42
00c0  c0 03 00 00 c0 03 00 00 c8 00 00 00 88 13 00 00
00d0  10 10 10 20 00 40 9c 44 ff ff ff ff ff ff ff ff
00e0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
00f0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0100  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0110  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0120  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0130  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0140  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0150  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0160  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0170  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0180  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0190  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
01a0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
01b0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
01c0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
01d0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
01e0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
01f0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0200  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0210  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0220  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0230  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0240  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0250  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0260  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0270  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0280  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0290  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
02a0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
02b0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
02c0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
02d0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
02e0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
02f0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0300  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0310  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0320  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0330  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0340  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0350  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0360  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0370  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0380  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0390  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
03a0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
03b0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
03c0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
03d0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
03e0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
03f0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0400  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0410  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
[... 167 lines cut ...]
0e90  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0ea0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0eb0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0ec0  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 00
0ed0  00 00 00 06 00 ff 00 01 ff ff ff ff ff ff ff ff
0ee0  ff ff ff ff ff ff ff ff ff ff ff ff 36 01 2e 01
0ef0  ff ff ff ff 00 00 00 00 01 ff ff 31 ff ff 1d 00
0f00  00 04 00 01 00 01 00 01 00 00 ff ff ff ff ff ff
0f10  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0f20  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0f30  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0f40  ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
0f50  ff ff ff ff ff ff ff ff ff ff 00 1c 01 e6 00 00
0f60  d7 ec b7 38 00 00 00 01 00 ff ff ff ff ff 00 00
0f70  9a c2 01 ff ff ff ff ff ff ff ff ff ff ff ff ff
0f80  ff ff ff ff ff ff ff ff ff ff ff ff ff 9a 99 19
0f90  3e 00 00 00 00 6d 6d 75 5f 62 6c 7e 31 ff ff ff
0fa0  ff ff ff ff ff 00 01 ff ff ff ff ff ff ff ff 00
0fb0  00 00 00 00 00 00 00 00 00 00 ff 00 00 00 00 00
0fc0  01 ff ff ff ff 04 00 f2 ff 06 00 10 00 fe ff 0a
0fd0  00 15 00 05 00 35 a5 1b 39 00 00 80 3f 00 00 80
0fe0  3f 35 a5 1b b9 27 1a 29 be b5 1c 27 bf b2 0b 00
0ff0  00 09 30 00 00 ff ff 01 ff ff ff ff ff ff 00 01
ok
```

DRracer, 2021-05-27:
@Lyr3x thank you for your effort in investigation of this issue. Our tech support will contact you regarding an exchange of your EINSY. 

Your EINSY may not be HW-defective, but your test case is the most reproducible one probably related to some memory corruption some users have reported recently.

Before returning the board, please do the following:
- flash 3.10
- reset all settings
- disable filament sensor
- disable fan check
- power off
- remove the following cables: all motors, fan cables (hotend, print), filament sensor
- power on again
- **verify problem still happens**
- power off
- disconnect and remove einsy without further changes

Lyr3x, 2021-05-27:
I did all you suggested except the filament sensor removal. It needs the connection to trigger the problem. I can disable the sensor in the settings, but the wire needs to be connected.
 With all other parts disconnected the problem still happens.

leptun, 2021-05-27:
Please take a photo of the fs cable connected to the einsy. Just want to check the pinout

Lyr3x, 2021-05-27:
[image]

DRracer, 2021-05-27:
@Lyr3x so it really needs to have something on the pin of the CPU - that's an interesting piece of information

DRracer, 2021-05-27:
@Lyr3x do you have another thermistor at hand you could try? Once we saw a situation on the MINI when badly connected wires inside the connector of the termistor cable caused watchdogs (there was some disturbance in the thermistor signal crashing the FW)

leptun, 2021-05-27:
@DRracer it needs something connected because the fsensor is active low. So with filament inserted the pin should be pulled low.

Lyr3x, 2021-05-27:
That's what i discussed with @wavexx also. 
I mean there are more things happening regardless the filament sensor but in connection with the MMU. (Which I do NOT had connected while troubleshooting (not Power nor the connection above the FSensor!)

With 3.10 the printer resets right after the second filament change. Always, on all tested models. 

> @Lyr3x do you have another thermistor at hand you could try? Once we saw a situation on the MINI when badly connected wires inside the connector of the termistor cable caused watchdogs (there was some disturbance in the thermistor signal crashing the FW)

I do have thermistors but non from prusa, but those cheaper chinese ones

DRracer, 2021-05-27:
@Lyr3x ok, thank you

Lyr3x, 2021-06-02:
New board arrived. Just quickly swapped and checked 3.10. It seems to work fine. Should we keep the issue open or do you only want to investigate internal?

wavexx, 2021-06-02:
IMHO this should be kept open until the cause is known. There's also @diluno experiencing the same issue.

diluno, 2021-06-02:
So, I haven't been following this very closely - do I need to return my board as well, or should I wait?

Lyr3x, 2021-06-02:
@wavexx I would also be interested in the root cause, if its found ever :) 
@diluno I think you should also contact the customer support.

DRracer, 2021-06-07:
@diluno We still believe this is a software issue at this time, but we decided to sample one board to investigate after a lot of help by @Lyr3x because we couldn't reproduce the problem. If you could wait a little bit and stick with 3.9.3 for the moment, we're going to report what's causing this as soon as we get it back and let you know if you should ask for a return.

DRracer, 2021-06-17:
@Lyr3x sad news, I've been playing with your EINSY for the whole day and I haven't been able to reproduce the issue at all (tried all kinds of setups - including thermistors and different fsensor boards). 

The only way to trigger the preheat menu (btw. a preheat before filament autoload) was to pot in place with the fsensor connector on the EINSY. If I disable fsensor in the menu, not even this test case causes the preheat menu to occur.

@diluno please verify that your fsensor connectors are properly connected on both sides and report back.

Lyr3x, 2021-06-17:
That is very unfortunate. I tripple checked the connection especially on the fsensor and could reproduce that issue with a 100% success rate. 
It did not occur once with the replacement EINSY. 
* You flashed the latest FW
* Inserted filament (just trigger the sensor)
* Power off -> Power on

And it does not trigger the preheat menu?

DRracer, 2021-06-17:
@Lyr3x I completely believe you ;)

- yes, I flashed the 3.10.0 stable final FW from github
- yes, I verified, that the fsensor works and left the filament inside
- yes, power off -> power on ... at least 50 times ... not a single occurrance

I suspect the fsensor cable - you might have it "half-broken". As written above - I was only able to cause the preheat menu to occurr when playing with the fsensor connector on the EINSY side.

diluno, 2021-06-27:
any news on this?

i checked my fsensor cable, even reconnected it on both sides. problem still persists when i flash 3.10 again.

wavexx, 2021-06-27:
@diluno Yes, we believe to have found the cause due to be a compiler bug affecting one of the main handling functions. The fix is already merged in master, but I've rebuilt the firmware at 56e531d40a1c81aff1004ab3fd3f8b1b8519718c for you here:

[1_75mm_MK3S-EINSy10a-E3Dv6full.zip](https://github.com/prusa3d/Prusa-Firmware/files/6721605/1_75mm_MK3S-EINSy10a-E3Dv6full.zip)

It would be great if you could already give it a shot and see if everything works as expected.

diluno, 2021-06-27:
Looks like it's working! Thank you so much!!

But now when I start a print, it says "New firmware available: 3.9.3", and I have to confirm first.

wavexx, 2021-06-27:
Ok, that *might* be my fault (I didn't set any FW version here), but I'll check that later. This shouldn't happen on offical versions. Meanwhile you should be able to disable the check in Settings->HW Setup->Checks->Firmware so you can test without having this at every print.

wavexx, 2021-06-27:
Note to @DRracer, checking for differences with #3093

wavexx, 2021-06-27:
@diluno found the problem for the version check as well (see #3211). Updated fw:

[1_75mm_MK3S-EINSy10a-E3Dv6full-2.zip](https://github.com/prusa3d/Prusa-Firmware/files/6722082/1_75mm_MK3S-EINSy10a-E3Dv6full-2.zip)

you should be able to re-enable the check after this.

DRracer, 2021-07-30:
@diluno thank you for your help, is your issue solved with FW 3.10.1-RC1 now? Can we close this issue?

diluno, 2021-07-30:
I still have the firmware provided by @wavexx installed (56e531d), which solved my issue.
Let me know if I should test 3.10.1-RC1 as well.

wavexx, 2021-07-30:
3.10.1 contains the fix, so it would be nice if you can test that so we're sure the bug is gone for then next official FW.

diluno, 2021-08-02:
Looks good, just finished a print with 3.10.1 without any issues. I think you can close this issue. Thank you!

DRracer, 2021-08-02:
@diluno thank you, closing

closed 2021-08-02 by DRracer