prusa3d/Prusa-Firmware #3148: [BUG]3.10 (M601) doesn't heat after resume

redtif, 2021-05-14:
Printer type - MK3
Printer firmware version -  3.10

Octoprint

Describe the bug
Insert Pause via Prusa Slicer. Printer will pause at the desired Z. If you resume either at the printer or via Octoprint, it will start immediatily without heating up again. Also it will not extrude, because temperature are to low. It will finished the print without extruding. 
Going back to 3.9.3 will fix this.

If you need any more information, i am happy to provide them.

riedlingerm, 2021-05-15:
Identical Issue with my setup.
Printer type - MK3S
Printer firmware version - 3.10
Octoprint version - 1.6.1

I will stick with 3.9.3 as well since color change prints are practical impossible with this issue.

JohnnyDeer, 2021-05-17:
Hi @redtif, I can't reproduce this issue, work as expected. For how long you leave printer in pause? Can you please upload the Gcode where this is happening?

redtif, 2021-05-17:
Hi @JohnnyDeer, I tried the whole testcase again, 
the M601 ist by line 5344
first firmware 3.9.3 stops before 8.2, resume at printer, display shows again the printing temperature, all OK
Update Firmware
second firmware 3.10-4481, stops at 8.2, resume at printer, **No Display for a short time (5-10 seconds) but printing continues, the display normal but 0° temperature and printing continues........**
here the related gcode 

[TestInsert_0.2mm_PLA_MK3_11m_3.10.zip](https://github.com/prusa3d/Prusa-Firmware/files/6494429/TestInsert_0.2mm_PLA_MK3_11m_3.10.zip)

I revert to 3.9.3!

Rgds redtif

kg4iae, 2021-05-17:
I'm having the exact same issue.  Sliced the project with Prusa Slicer (V2.2).  Added a "Pause".  Printing using Repetier Server (1.04).  

In the previous firmware 3.9, the print would pause at the requested location.  The hotend would go to 0, and wait for me to put in the washer.  Then you have to do "Resume Print" on the printer.  The printer will heat back up and move into position.  Then you have to click "Continue" on Repetier Server.

In this new firmware, 3.10, the printer again pauses at the the requested location.  The hotend goes to 0, and it waits for me as usual.  When I do "Resume Print" on the printer, the screen goes blank, and appears to be non-responsive.  Then it immediately starts without heating the hotend.  Repetier Server goes from "Pause" to "Printing" without me doing anything.  (I actually like this change.)   

After the print is complete, the printer still appears to have a job running.  I can see the option to "Resume Print" in the menu.  Selecting it makes the screen go blank and non-responsive.  From there I usually get frustrated and turn it off.  

Hope this gets fixed soon, I use this feature often.

3d-gussner, 2021-05-18:
@redtif @kg4iae 
I also tried to reproduce the issue in the MK404 simulator and everything went fine. Tried my own `M601` gcode as well yours.
Also paused and resumed the printer manually in some combination of printer and Octoprint.
To use FW3.10.0 correctly with an USB host system like Octoprint you need to add in GCODE scripts
- `M601` in `After print job is paused`
- `M602` in `Before print job is resumed`

Without these I could reproduce your issues with "blank" screen.
- Modified TestInsert gcode to trigger at 0.4 a pause
- Copied TestInsert gcode to Octoprint
- Started TestInsert gcode in Octoprint
- Resuming via LCD generated a "blank" screen for few seconds and continued to print without heating.

Started another `M601` test gcode from SD card with Octoprint connected
- The printer stopped at the correct layer
- Resuming the print from LCD resumed correctly waiting to heat up

At this moment the printer doesn't know what the host system is capable. The GCODE scripts `M601 M602` is a workaround to establish kind of "two-way" communication for pause/resume.

There may be some printer/host negotiation in the future what both (printer/host) are capable of and this workaround isn't needed anymore, but for the meanwhile that is the "only" option to use `M601` with USB hosts like Octoprint and FW3.10.0

Other USB host systems may work fine.

I hope that helps.

gigios, 2021-05-19:
@ 3d-gussner If I understand correctly it is necessary, until a probable Octoprint update, to add the M602 command after M601 in the gcode pause in the prusaslicer (in te custom gcode page of the printer).
It's correct?

3d-gussner, 2021-06-01:
@redtif Would you mind to close this issue as we have a working solution?

gigios, 2021-06-03:
It's clear that we have a possible 'workaround', but, for me, it's better if the firmware is able to works in the correct way also without the changes in the octoprint settings (of course, without these changes the octoprint, probably, is not correctly synchronized but if the user resume the print from the display the print restart correctly).
This is my point of view, of course 😊

PS: I think it's necessary to add in the release notes how to setup octoprint to works correctly with the new firmware.

backstube, 2021-06-04:
Exact same issue here, right after Upgrade to v3.10.0 from v3.9.3. 

MK3S, v3.10.0, OctoPrint 1.5.2

3d-gussner, 2021-06-05:
Let me try to explain why it is like it is now.

Terms used:
- `brain` is the unit sends instruction
- `muscle` is the unit executes the instructions

Gcode (simplified):
- `M601` pause
  - finishes the queued movements
  - sends an `OK`
  - stops and saves print values to ram
  - sends `//action:paused` via serial line.

- `M602` resume
  - finishes the queued movements
  - restore saved print from ram and continues
  - sends `//action:resumed` via serial line.

**SD printing**
During a SD print the printer firmware is the `brain` and the `muscle`.
It is reading the instructions from the SD card and according to its rules/functions executes these.

As soon the SD print contains a `M601` it will execute the command.
If user manually chooses to pause the print via LCD it executes the `M601` command.

If the user manually chooses to resume the print via LCD it executes the `M602` command.
If the printer receives via serial `M602` it executes this command.

**Octoprint / USB Host printing**
During an Octoprint/USB print the host is the `brain` and so responsible to instruct the `muscle` to do things.

As soon the Octoprint/USB print sends a `M601` from the gcode file the `muscle`/printer will execute the `M601` command.

If the user manually press the `pause` in Octoprint IT as the `brain` has to instruct the `muscle` to do things:
- Sending a `M601` (via the G-Code Script settings) will instruct the `muscle` to do it.
- You can setup your own pause procedure (example is similar what the `M601` does)
```
M400 ;Wait for current moves to finish
M105 ;Get Extruder Temperature

_and somehow store it to some variable? I am not an Octoprint expert_ 

M104 S160 ;Set Extruder Temperature to 160 to prevent ooze during pause
M107 ;Turn off print fan
M114 ;Get current position 

_and somehow store it to some variables? I am not an Octoprint expert_ 

M91 ;Set to Relative Positioning
G1 Z20 F300 ;Raise the nozzle by 20mm
M90 ;Set to Absolute Positioning
G1 X50 Y190 F1000 ;Move to pause position X50 Y190
M400 ;Wait for current moves to finish
M117 Print paused ;LCD status output
```

If the user manually press the `resume`in Octoprint IT as the `brain` has to instruct the `muscle` to do things:
- Sending a `M602` (via the G-Code Script settings) will instruct the `muscle` to do it.
- You can setup your own pause procedure (example is similar what the `M602` does)
```
M400 ;Wait for current moves to finish
M109 S<previous store variable> ;Set Extruder Temperature and wait to previous saved value
M106 ;Turn on print fan
M90 ;Set to Absolute Positioning
G1 X<previous stored x pos value> Y<previous stored y pos value> F1000 ;Move to stored position
M91 ;Set to Relative Positioning
G1 Z-20 F300 ;Lower the nozzle by 20mm
M400 ;Wait for current moves to finish
M117 Resumed ;LCD status output
```

**Please DON'T use these examples as I am not an Octoprint expert.**

To keep Octoprint/USB host as flexible as possible it makes sense to keep it that way.

So you can setup Octoprint GCODE Script to what ever you want, BUT as the printer isn't the `brain` during the USB print the host system has to instruct the `muscle` what to do.

3d-gussner, 2021-06-05:
Started to test again and Octoprint sends the `M601` command, which is in the gcode file, to the printer.
The printer executes the command.
Need to think about this as the "own" scripts will not work correctly as the printer already moved to park position and so on.

sarf2k4, 2021-06-10:
I actually tried to pause prints from the printer's lcd, but I resumed via octoprint just to see if both host and slave really interact each other. The printer continued to print but the nozzle stopped heating like anyone else before this too. I decided to let the print continue until the M601 in my gcode being sent over to the printer to see if the printer goes to parking position, it didn't, because my octoprint "after pause" script is only M601.

After I put up together "after pause" and "before resume" script, I started another print of similar gcode. I decided to try again the above, by pausing the print job via lcd panel of the printer itself. To my surprise, it states "resume print" instead of "pause print". I pressed the "resume print", I forgot the exact behavior at this point either I cancelled the print via octoprint or the printer's lcd panel itself.

Then I started the same gcode file again, for whatever weird reason, during the purge gcode sequence, the printer's nozzle goes all the way to the back and I immediately cancelled the job via octoprint, knowing that there will be delays, I immediately pressed the reset next to the lcd. As far as I remember at this point, the printer executes an "unload filament" sequence, which I also put another in the octoprint "after print job cancelled" script.

Printer made weird behavior where it somewhat received a job to start prints yet the host already cancelled it. In the end I did octoprint full system reboot.

broodro0ster, 2021-07-14:
Same issue here with Octoprint (lastest stable version) and 3.10 firmware on my MK3S.
I inserted a pause like I normally do, but the hotend wasn't heating anymore after resuming and no filament was extruded.

tre4b, 2021-08-05:
I was on 3.9 when I first started trying to do this and had this issue.  I upgraded to 3.10 as I read comments somewhere that the reason it failed was down to firmware so I upgraded to 3.1 the most recent.   Now I can't remember what version my previous 3.9 was.  Did this happen with older versions of 3.9 too?

broodro0ster, 2021-08-05:
I went back to 3.9.3 and I can use the pause function again. I only had the issue with 3.10. I still need to try the 3.10.1 beta.

tre4b, 2021-08-05:
What version of PrusaSlicer are you using?  Maybe this is a combination thing?  Mine is 2.3.3

And I suddenly wonder.  I have an MMU so I used the firmware for that.  I wonder if this is why 3.9.3 works for you but not for me?

broodro0ster, 2021-08-05:
I don't have an MMU and I'm using PrusaSlicer 2.3.0.

tre4b, 2021-08-05:
I wonder if it is the MMU, that might make sense.  The slicer just seems to add the M601 and it is the printers reaction to that which is the issue.

SgtSeppel, 2021-09-07:
I can confirm this behavior with a combination of 
- FW: 3.10.1
- PrusaSlicer 2.3.3

Also, it is not showing the custom message which was added in PrusaSlicer (like: "Insert Magnet"). It just pauses.

JanKalin, 2021-09-24:
I can also confirm this behaviour. FW: 3.10.0-4481, PrusaSlicer 2.3.3, Octoprint 1.6.1

@3d-gussner's solution works. You can resume from both the Octoprint and from the printer itself.

3d-gussner, 2021-11-11:
I just have re-tested the FW3.10.0 with Octoprint and the workaround above.

It still works see https://youtu.be/N7FnXiOZhwM

@schraufi @vjirovsky @AmauryBJ have you read the complete issue and the workaround https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-843442610 ?

3d-gussner, 2021-11-16:
@schraufi @vjirovsky @AmauryBJ Can please try the workaround and give us some feedback if it solves your issue?

gjpc, 2021-11-23:
I experienced this problem this week, and I believe it is an Octoprint problem not a Prusa problem.

kg4iae, 2021-11-23:
It happens to Repetier-Server too.  I downgraded to 3.9 to fix the problem.

3d-gussner, 2021-11-24:
@gjpc @kg4iae Have you watched and read https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-966463702 ?

gjpc, 2021-11-24:
There was a fix on the Prusa forum that I adapted into Octoprint see the details [here](https://community.octoprint.org/t/resume-does-not-check-tool-bed-temperatures/39107)

ulope, 2022-03-26:
I think the discussion has gone of topic here. The original report (and AFAICS what others also report) is that a pause (M601) added *via PrusaSlicer* and sent via octoprint causes the printer to behave erratically (i.e. not heating again, blank display, etc.).

Can this please be fixed? It's clearly not working.

gjpc, 2022-03-26:
OK, I have the solution to this problem that only works on Prusa firmware 3.01 or later. If you do have a Prusa MK with updated firmware perform the following steps in your Octoprint.

select the Spanner (wrench) at the top of the OctoPrint page
select GCODE Scripts in the left hand column
enter **M601 ; pause the print** in the After print job is paused text box
enter **M602 ; resume the print** in the Before print job is resumed text box
enter **G4 P5000 ; Dwell in the Before print job is resumed** as the second line in the text box
select the Save button at the bottom of OctoPrint Settings

You will find two new scripts in your scripts/gcode directory afterPrintPaused and beforePrintResumed

When you select the OctoPrint Pause button the print will pause, the hot end heater will be turned off and the the print head will politely move away from the print. The bed heater will stay on to keep the print adhered to the build surface.

Selecting Resume will pickup the print exactly where it left off.

kg4iae, 2022-03-26:
I have seen this solution posted else ware.   I tried it with no success on Repetier-Server.  Not sure what's going wrong, my printer still behaves badly.  The solution I found is to do a "filament change" instead of a "pause".  The "filament change" does not shut down the hot end like "pause" does.  It's an extra step removing and replacing the same filament, but it's a lot easier than fighting with the un-predictable nature of the broken firmware pause function.

sarf2k4, 2022-03-27:
If I remember correctly this issues are somewhat 'partially' fixed? especially with the printer reporting back its status such as `:paused` or similar. I did this test months back and I'm sorry to tell you that I forgot about the outcome of the tests

derpicknicker1, 2022-04-21:
> OK, I have the solution to this problem that only works on Prusa firmware 3.01 or later. If you do have a Prusa MK with updated firmware perform the following steps in your Octoprint.
>
> select the [...]

This solution does not work for me. The Printer ignores the M601 and prints the object without pausing the print.
Using OctoPrint 1.7.3
Prusa FW 3.10.1-4697

gjpc, 2022-05-01:
Too bad for you. You obviously cannot follow directions.

bartmeuris, 2022-10-26:
> When you select the OctoPrint Pause button the print will pause, the hot end heater will be turned off and the the print head will politely move away from the print. The bed heater will stay on to keep the print adhered to the build surface.

This is **not** the issue. As @ulope mentioned, the problem happens when you insert a pause in the G-Code before a certain layer is printed using prusaslicer, which injects an M601 command there.

The printer pauses before that layer, and cooled down the nozzle, probably because I waited for too long. When I then resumed the print on the printe*, the nozzle was not re-heated, and it just started moving the printhead without extruding due to a too low nozzle temp.

It's pretty weird that somehow this would be an OctoPrint issue when this did work correctly with older firmwares. It's the firmware's choice to disable the nozzle heating, the "brain", as @3d-gussner referred it to, never decided this, and just assumes the printer is still in the state it the G-code it sent had configured, only paused. So from my pov, the moment the printer unpauses - it's the firmware's job to restore the previous state the "brain" expects before continuing.

gjpc, 2022-10-26:
Hey, @bartmeuris I gave up on Octoprint months ago. I paid the Repetier registration fee, and am very happy with a stable, professionally maintained 3d printer management system. I've had enough of the constant bickering on these forums. 

I do wish the best of luck to you folks that want to constantly tinker with your 3d printer code, but as far my 3d printers are concerned, I just want them to function as well as my toaster oven with minimal fiddling.  I retired from programing years ago.

Prusa-Support, 2022-11-02:
> It's pretty weird that somehow this would be an OctoPrint issue when this did work correctly with older firmware. 

If I'm not wrong, the firmware didn't use to work in both "directions" back then: pause/resume via host didn't work if the print was paused/resume via the printer LCD panel and vice versa. I can only speculate that something occurred there, but this was probably necessary to unlock further functionalities and scenarios of use.

Now certain commands can be sent from both ends. This also opened the doors to Prusa Link and Prusa Connect.
The idea is that Prusa Link and Prusa Connect in the future will offer a complete workspace - especially fully directly supported and maintained by us - for remote printer control and farm management.

Our developers have plans to improve the firmware and possibly solve issues like this, so the issue can stay open.
As explained in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/283, the solution to this issue though should not pose an obstacle to the company's main development plans. Please bear with us.

Meantime, specific configurations on the host side, may address the problem [as explained above](https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-843442610).

Michele Moramarco
Prusa Research

sarf2k4, 2022-11-02:
I totally forgot to provide feedback that the latest `M601` inserted from prusaslicer (latest stable release to this comment's written) causing the nozzle not being reheated. 

I am running on [3.11.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.11.0) firmware.

I decided to try out the `M601` inserted from the prusaslicer yesterday, because I have a part that requires item to be inserted and closed in it, hence requires the `M601`. Tested with few simple scenarios. All of these has `M601` inserted in prusaslicer.

> 1
> print paused as intended
> printer main menu > resume print
> printer moved to the original location then printer paused again for a brief moment
> printer moved to the original location with heater being properly heated up
> 
> In this scenario, I forgot that I have additional `M601` in octoprint's `after print paused` g-code command while having `M602` in `before print resumes`
> 
> I think the above `M602` in `before print resumed` wasn't sent because I resumed the print from the printer's lcd panel

> 2
> similar to [1] above except I emptied both "after print paused" and "before print resumed" in octoprint
> 
> the nozzle didn't heat up after resuming print from the lcd panel and the printer didn't do [1.3] above; "printer moved to the original location then printer paused again for a brief moment"

> 3
> Printer paused as intended
> 
> I only have `M602` command on "before print resumed" in octoprint and pressed "resume" from the octoprint web interface. Everything works okay, nozzle heated up properly before continuing the print job

In summary, we only need to issue `M602` command in octoprint's `before print resumed` configuration. However, I highly encourage that you guys would do further tests to confirm my scenario, that this is 'perhaps' the simplest solution

I also not sure what would happen if prints directly from sdcard

kg4iae, 2022-11-02:
Not sure if this helps your troubleshooting process... But it works correctly if you do a "Color Change" instead of a pause/resume.  I use Repetier-Server.  When the printer pauses for a color change, it does not cool down.  The server shows "Paused".  Once you replace the filament, and click Resume (on the printer) or click Resume on the server, it will continue where it left off without any trouble.  If you happen to wait for a while, the hot-end will cool down due to a time-out.  As soon as you click the button on the printer, it will heat back up and continue the "Color Change" process.

Hope that helps.

KilianB, 2022-12-13:
> In summary, we only need to issue `M602` command in octoprint's `before print resumed` configuration. However, I highly encourage that you guys would do further tests to confirm my scenario, that [...]

As visuals are a bit easier to follow, the following step worked for me. After resuming the print from within Octoprint the nozzle heats up again before resuming the print.

[image]

nonix, 2023-05-14:
Thank you for the explanation and the WO, but I can't resist to ask a silly question. Just few days ago, I was printing thing where two colors are required, using the OctoPrint (pause, replace filament, resume). All went just perfect!

Why does the pause/resume works when changing filament?
Cheers,
 N.

Prusa-Support, 2023-06-14:
Unfortunately, certain behaviors are not totally predictable for us because Octoprint is not maintained directly by us, especially because the Octorpint functionality may even slightly change update after update.

I'm sorry if this is not much of help but please consider reaching out to the [Octoprint](https://octoprint.org/)'s community.

Michele Moramarco
Prusa Research

ulope, 2023-11-02:
I don't think this is resolved. There's a workaround but people will probably only discover this after at least one failed print (if they manage to find this buried in some random github issue).

abatkin, 2023-11-02:
If the user presses the "Resume" button on the printer, what is supposed to happen? I don't understand how the serial protocol works, so here are the options that I can think of (and someone from Prusa can maybe chime in with their expectation)

1. User presses Resume. Printer sends some command over serial line and whatever is controlling the printer is responsible for doing stuff
2. User presses Resume. Printer is responsible for heating back up to the right place, putting the print head back where it was before, etc - at which point it _then_ somehow hands control back to the serial console
3. User presses Resume. Printer is responsible for heating back up to the right place, putting the print head back where it was before, etc, but as that is happening, those commands are echo'd back to the serial console, so whatever is controlling things over serial should be monitoring those commands so that it doesn't send more commands (like to start printing again)

Based on how this process is supposed to work (which has nothing to do with OctoPrint, and everything to do with Prusa's protocol for sending/receiving gcode over serial), we can figure out where the bug(s) lie, and where fix(es) should be

abatkin, 2023-11-02:
While I think 2 makes the most sense  (because buttons on the printer are effectively an "out of band" control mechanism, so it should put the rest of the world on hold while it does its thing, then resume the rest of the world once it is back to where it needs to be) - I don't think *we* get to choose which one we prefer. There is (presumably) an existing "standard" for how the serial protocol works, so we all need to know which one is expedted.

If it's (1), the bug is 100% in OctoPrint. If it's (2), the bug is 100% Prusa. If it's (3) then it could be either one.

nonix, 2023-11-02:
I will chime in as I have reported it at first. IMHO, I am not convinced (but 'm not the expert) that the bug is in the OctoPrint. As far as I know, the OP just feeds the commands over serial.
Where I would be looking at is the fact, that when you do like to print with two color string on single head printer you are prompted (on Prusa display) to change the filament, than the print resumes whiteout any issue.
However, if you like to do the insert it does cool down then restarts. IMO it is very similar from operation point of view, so perhaps if you could compare the two sections of gcode interpretation, could help to shed some light.
Also, I have tested it with the latest version of Prusa fw.
Anyway, thank you very much for looking into it.

closed 2024-02-14 by github-actions[bot]