prusa3d/Prusa-Firmware #3151: [BUG] Prusa MK2.5S firmware 3.10.0 (with MMU2S 1.0.6) randomly repeats a short section of gcode several times before failing with "static memory overwritten" error

IntegersOfK, 2021-05-20:
**Printer type** - MK2.5S with MMU2S
**Printer firmware version**-  3.10.0-4481

**MMU Upgrade** -  MMU2S
**MMU upgrade firmware version** - 1.0.6-372

**SD card or USB/Octoprint**
SD Card

**Describe the bug**
Printer has an odd movement, then repeats a short section of the gcode and eventually fails entirely with "static memory overwritten" error. If you restart the printer and do the print again, with the same print file, the failure may happen in a different section/different time.

After getting this error multiple times (in different sections of the same print), for my troubleshooting I reset the printer and MMU to factory defaults and reflashed the latest firmware on both. I did the calibration wizard all fresh and am using PrusaSlicer to create the gcode. This is a single colour print (using PrusaSlicer default profile "Single").

I have seen the problem occurs on both of these files:
[Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU2S_3h37m.zip](https://github.com/prusa3d/Prusa-Firmware/files/6512356/Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU2S_3h37m.zip)

I have replaced the SD card as a last resort in case there was some kind of read error or corruption issue with that, using the same gcode files exported from PrusaSlicer. It does not appear to be related to the SD Card.

It's worth noting that after I see the issue happen, I can pause the print from the menu and when it resumes it appears to continue until the issue happens again eventually in another section or on the next print of the same file (but again, not always in the same place).

Here is a picture which shows the section of gcode has repeated itself several times before shutting down entirely:
[image]

**Video**
https://youtu.be/z2-jaG_GTXU

The problem was also happening on the 3.10.0 RC1, but I had never had it happen on on earlier firmware versions (3.9.3) though I did not try these specific gcodes on that firmware version (I will reflash older firmware and try it now though).

#2964 is another issue which appears to mention this type of error, though that issue report it happening at the start of the print.

#2791 appears to be somewhat related to solving a similar issue, but is related to sensors?

I think some changelog referenced "strange movements" before this error, which feels close to this problem too.

wavexx, 2021-05-20:
Using 3.10 could you try to disable the filament sensor (settings -> filament sensor [off]) and see if that happens again or if the problem disappears?

IntegersOfK, 2021-05-20:
I would like to clarify that with 3.10 I am not noticing the issue on every print, even when using the same gcode. In particular, after I flash the firmware fresh to 3.10, it always seemed to handle the first print normally, but when I retry the same gcode again right away, I get the error eventually somewhere different each time. That intermittency/inconsistency in when it occurs, is probably going to make it difficult to give a full overview of how to reproduce it, though I can say on 3.10 it almost always happening the second and subsequent prints after flashing the firmware (which might be a coincidence that the first one has always worked after flashing, I don't know if it was worth mentioning, just something I noticed). Again, always using that same gcode, the 10x10 "fabric" link model.

> Using 3.10 could you try to disable the filament sensor (settings -> filament sensor [off]) and see if that happens again or if the problem disappears?

That's a good suggestion, I will give it a try next! Today I went back to 3.9.3 with the same gcode and filament, and have done two copies of the same print now without getting that error. I would like to do a few more with 3.9.3 due to that intermittency I mentioned, just to confirm that we really aren't seeing it happen on 3.9.3 and I think the sample size needs to be higher to say it with confidence. After a few more and I'm satisfied that it's not an issue on 3.9.3, I will reflash 3.10, wait for the problem one more time, and then disable the filament sensor for some testing after that. Will report back (probably in a day or two) with those results.

wavexx, 2021-05-20:
From the looks of it, it could be either some memory corruption issue (triggered by timing more than gcode IMHO), or some (incorrect) repeated call to the pause/resume machinery in an incorrect moment. Disabling the filament sensor mostly removes one source that *could* cause pause/resume internally.

These are nasty to debug. Any detail/step you can provide is useful. FW 3.10 is actually faster in a few places and uses less memory as well, which incidentally can cause an old timing bug to surface now, or a memory overwrite bug to overwrite something which is now critical :(

wavexx, 2021-05-20:
As an additional note, to help making the bug more reproducible, try resetting the printer at each print. This ensures the starting conditions are more similar and could result in a more reproducible scenario.

IntegersOfK, 2021-05-21:
After several more prints of the same gcode yesterday on 3.9.3, I feel confident that 3.9.3 does not contain the bug. Later today I will reflash 3.10, wait for the error, then disable the filament sensor and try again a few times. More to come!

leptun, 2021-05-21:
The SD card reading was changed in 3.10 to be more efficient. Maybe something broke along the way. There have been other issues relating to this that were fixed in the RCs, but maybe there is still some edge case that wasn't encountered so far.

IntegersOfK, 2021-05-23:
I turned off the filament sensor before starting the print again with 3.10, and I am sorry to report that I encountered the error again (in a different place as previously described) with that same print. I confirmed after the strange repeat movement started that the filament sensor was indeed off the whole time.

> The SD card reading was changed in 3.10 to be more efficient. Maybe something broke along the way. There have been other issues relating to this that were fixed in the RCs, but maybe there is still some edge case that wasn't encountered so far.

Should I install OctoPrint on my Raspberry Pi and try a print from that? Would that be a valid test to determine if it's SD-card related? As I said before, I have already tried two different brands of SD card, so I really don't think it could be the card itself.

IntegersOfK, 2021-05-24:
I did two prints with OctoPrint and did not get the error. I will try a few more tomorrow and turn the filament sensor back on. But preliminary testing indeed feels like an SD Card read issue then I suppose. Will report back again after I can confirm a few more with OctoPrint.

IntegersOfK, 2021-05-24:
After a few more prints, it works fine from OctoPrint. Same gcode, which I even copied from the SD card directly in case there was some kind of corruption that I was missing. So I guess @leptun is onto something when suggesting it could be the SD read logic. Is there anything else I can do to test for you about this? Are there logs anywhere I can record while it happens maybe?

djakkk, 2021-05-25:
I have the same problem after updating my MK2.5 to 3.10.0 (no MMU2 connected)

The first print I tried ran to about 95% and then the printer started the gcode over from line 1 with bed leveling.
The 2nd time I tried the same print it did the exact same thing.

When I tried to print another file I got the "static memory overwritten" error"

After that I ran a factory reset (clear all) and flashed the firmware again (the same version)
After running the Wizard, first layer calibration and trying the same files from the same SD-card (a Kingston 8GB Class 4)

This time the 1st print (15 min) worked, but the 2nd print (30 min) only ran for about 40% before the printer restarted the gcode.

After this I copied the gcodes to a new SD-card (Verbatim 4GB Class 10), but this time the 2nd print only ran for about 5% untill the printer restarted the gcode.

So now the printer is useless... not sure if I should downgrade the firmware, wait for a fix, or try using octoprint.

IntegersOfK, 2021-05-25:
> not sure if I should downgrade the firmware, wait for a fix, or try using octoprint.

Well I can confirm that both options will work in the meantime. 3.9.3 has no issues, and OctoPrint has been working great. 

Obviously I agree that this is a frustrating issue and absolutely must be fixed (otherwise we can't print from SD Card). Let's give them some time, I'm confident with our detailed reporting, the Prusa team and contributors will be able to solve it.

djakkk, 2021-05-25:
> Well I can confirm that both options will work in the meantime. 3.9.3 has no issues, and OctoPrint has been working great.
I'm gonna dig out an old Raspberry pi and install octoprint and see if that helps here too.

> Obviously I agree that this is a frustrating issue and absolutely must be fixed (otherwise we can't print from SD Card). Let's give them some time, I'm confident with our detailed reporting, the Prusa team and contributors will be able to solve it.

I hope so, any idea if the MK3 has the same problem?

djakkk, 2021-05-25:
Just printed the 30min gcode via octoprint and that worked perfect, so the issu seems to be with the SD-cards.

DRracer, 2021-05-26:
Sadly we haven't been able to reproduce the issue so far: 
[image]

From your reports it looks like something related to reading from the SD card + doing a second print (may be even stop the first print and then start another one).

DRracer, 2021-05-26:
@djakkk if you need to print, please downgrade to 3.9.3 for now. Yet, we have not enough information what may be happening on your and @IntegersOfK machines to fix the issue soon.

djakkk, 2021-05-26:
@DRracer 
I spent the morning trying to print the same file over and over from two different SD-cards, it fails every time but with different results.

Some times the printer just restarts the gcode, some times it either runs the same area several times or really over-extrudes as you can see in the pictures.
The print is only supposed to be 1 layer high, but as seen on the last pictures the printer thought it would be a good idea to print one of the squares over and over and over.. or with 10-times the extrusion (I have not had time to sit and watch the printer since I have to work with other things while printing)

Let me know if there is anything I can do to help you to solve this bug, I can email you the gcode-file I'm using, or set up a remote connection to the printer some how (via teamviewer and pronterface or something?), or if you want me to set up a octoprint-camera to a webcam so you can watch it print? (or a webcam via zoom or something)

[image]
[image]
[image]
[image]
[image]
[image]
[image]

djakkk, 2021-05-26:
@DRracer 
Here is the gcode I'm trying to print.

[2-3x3-bed-level-calibration-230x190-015mm_0.15mm_PLA_MK2.5S_30m.zip](https://github.com/prusa3d/Prusa-Firmware/files/6545568/2-3x3-bed-level-calibration-230x190-015mm_0.15mm_PLA_MK2.5S_30m.zip)

IntegersOfK, 2021-05-26:
>  (2kg of filament ;)

Well at least you have some cool chain-link fabric 👯 

>  it looks like something related to reading from the SD card + doing a second print

To be clear, I think it does happen on the "first print" after restarting the printer, I've just never seen it happen on the first print after freshly flashing the firmware. I guess it could be interesting to make a few more tests to compare if the frequency of occurrence truly differs between first/second print after flashing firmware vs first/second print after restarting printer, but what can this even tell us? Clearly we all agree there is an SD card reading situation and two of us with the issue have tried several brands and sizes of cards, all with the same results.

@djakkk Did you happen to notice if the print might work for the first print after a fresh flash of 3.10? It's possible that mine being intermittent was just a coincidence that each time the first one works after reflashing.

> we have not enough information what may be happening

What else can we do? Would it be useful to try reverting the SD Card changes and recompile it without that new SD card optimisation logic you mentioned? Or maybe there is a debug mode where we can capture logs related to the SD card reading?

djakkk, 2021-05-26:
> @djakkk Did you happen to notice if the print might work for the first print after a fresh flash of 3.10? It's possible that mine being intermittent was just a coincidence that each time the first one works after reflashing.

The problem happens 100% of the times when I print from SD-card, even after a firmware update, and factory reset+firmware update.

I does not seem to make any difference if it's the "1st print on the power-cycle" or not, I have printed from both a power-up and a reset, I have not printed after a successful print though, since I haven't had any successful prints from the SD-card at all.

IntegersOfK, 2021-05-26:
Wow that new SD card logic is indeed quite extensive. I thought it might be straight-forward to revert some obvious change to try to cut a custom version of 3.10 without those optimisations, but it's too complex for my experience.

I see lots of references in these changes to block caches, repeating, byte comparison, fetching cluster, filtering, etc. I know a lot of the diff is just moved from other files and stuff too but just to say I do not know how to break down these changes to help narrow down the function(s) that could be causing our problem. 😞 

https://github.com/prusa3d/Prusa-Firmware/commit/d275fe0e83e7d29397413e4804069dc0ef139670#diff-2b29d7fdceb2c4e4d73d37c9210e4c881c08b248edcbeb12637ae8877edeff9a

djakkk, 2021-05-26:
> Wow that new SD card logic is indeed quite extensive. 

Oh, too bad.

But it seems strange that there aren't more people having this problem though?

I've got an old LCD with a broken rotary encoder, I'm gonna replace the encoder later tonight and see if it makes any difference, if that doesn't help I was thinking of trying to shorten the cables from the Rambo to the LCD to 10cm or so and see if that makes it work.

wavexx, 2021-05-26:
To help finding out the case in a scenario like this, it's helpful to keep the FW in the same starting conditions. So if you have some gcode that happens already on the first layer @IntegersOfK , try to reset just before printing, select the file and start the print without doing anything else. If you do this a second time, does it happen in the same place?

If you don't have anything specific on that SD card that you don't mind sharing, could you make an image of it? This way we could check if there's something specific to the SD structure that's causing the issue to appear as well.

On linux dumping the card is quite easy. After unmounting it: ``dd if=/dev/[device] of=/path/to/file``

djakkk, 2021-05-26:
I replaced the LCD on my MK2.5S and tried again to print the same gcode from the same SD-card, but it did not work.
 
 @wavexx To answer your question to @IntegersOfK , my errors does not accur at the same place, it seems to be different places in the gcode every time, as you can see in this picture the print did not go far before it all of a sudden jumped back quite a bit in the code and started to print over the old layer (you can see the diagonal string where it jumped back this time, and compared to the older pictures I've posted  you can see it's always in different places.
 
 Is it possible for you to compile some kind of debuging firmware I can try that logs to putty or pronterface?
 
 Maybe we can find out if it's always the same amount of lines/bytes it jumps back in the gode if we can see the last working position and how far back it jumps?, if this would help you in some way to find the bug?
 Or if it's a specific gcode command that triggers the problem?
[image]

leptun, 2021-05-26:
We are preparing right now a debug firmware that could get us closer to the root of the issue. It seems this issue is not isolated to the SD handling and is just memory corruption. This can obviously lead to random behaviour such as seen here.

djakkk, 2021-05-26:
> We are preparing right now a debug firmware that could get us closer to the root of the issue. It seems this issue is not isolated to the SD handling and is just memory corruption. This can obviously lead to random behaviour such as seen here.

Yay, that sounds great, great work :)

Is this a debug firmware that I and @IntegersOfK can use or is it for internal use only?, and if it's for me and @IntegersOfK , is there any software or hardware we can prepare while waiting for the firmware?

IntegersOfK, 2021-05-26:
I think we're past that now, but just to say I can also confirm that I have never seen it occur at the same place twice; there is no consistency to when the "repeat" behaviour starts and begins looping. I've even had it happen 90% of the way through the print.

I tried making an image of my SD Card but my terminal didn't want to let me copy from the unmounted card. I will try again. But in the meantime, I had just formatted it fresh and it only has those two gcode files on it, so I would be very surprised if it has to do with some funky formatting or folder structure on the (multiple) cards.

Would be happy to try your debug firmware or anything else I can do to help solve this!

wavexx, 2021-06-12:
@djakkk @IntegersOfK I prepared a preliminary FW to help debugging this issue by combining pr #3180 and #3157 with the new crash dumping feature.

The branch is https://github.com/wavexx/Prusa-Firmware/tree/mk25_test_case, but I compiled the FW for the 2.5S already:

[mk25_test_case_fw.zip](https://github.com/prusa3d/Prusa-Firmware/files/6642291/mk25_test_case_fw.zip)

Select the hex file appropriately.
Which version of the Mini rambo do you have?

This is FW 3.10, but with a new debugging feature that allows to dump the memory at the instant of the crash so that we can see what is going on. On the 2.5S you need to have something reading the serial in order to get the log, since it cannot be saved in the printer itself.

A brief list of instructions (@leptun correct me if I'm wrong!):

- Flash this FW
- Add "D23" (without quotes) as the last instruction in the startup gcode OR modify the gcode to have D23 before the print actually starts.
- Copy the updated gcode to the SD card
- Enable the serial log in octoprint (the output is long, and we need all of it). Octoprint might need a restart after enabling it?
- Connect octoprint to the printer
- Start the print via the LCD. We use octoprint only to log the serial.

Instead of displaying the error message, the print will now suddenly stop on crash and start to beep. The crash log is written on the serial. After the crash the printer will auto-reset, and display that a crash has happened.

We need the full log generated by the printer. I rarely use octoprint, but in order to have the entire sequence you'll need to get it from the serial log. The log starts with:
````
D23 - emergency serial dump
````
and finishes with ``ok``.

This is a first attempt at memory debugging, so some things might need a bit of refinement here and there.

leptun, 2021-06-12:
Due to a bug in the bootloader of the miniRambo, the board might get stuck after the dump is finished and just show a blank screen. This is because the watchdog timer wasn't implemented properly in the bootloader back then and is a widespread issue that affects the Arduino bootloaders as well. The Einsy rambo is not affected by this bug.
This bug causes repeated watchdog resets inside the bootloader itself, basically never letting the firmware run. To get out of this state, an external reset is needed. This can be accomplished in multiple ways:
- pressing the reset button on the lcd.
- Reconnecting from octoprint.
- Power cycling the printer.

The bug itself is not dangerous as all heaters are disabled, but just something to keep in mind. After the manual reset, the error message about the firmware crash should appear.

djakkk, 2021-06-12:
I flashed the firmware, added D23 to the g-code file I printed earlier that didn't work a single time and started logging the printer with Putty.

I'm sorry to say... the print worked just fine... This is the first file I've managed to print from the SD-card using 3.10.

Do you want me to try again or have you changed anything in the debug-firmware that might have fixed the problem?

leptun, 2021-06-12:
This memory corruption issue seems to be greatly influenced by timing, so even a small skew sometimes fixes the issue. We're hoping however that we can get one debug build where the issue still happens. Otherwise, the debug code has to be left in the main build until the issue surfaces again.

djakkk, 2021-06-12:
> This memory corruption issue seems to be greatly influenced by timing, so even a small skew sometimes fixes the issue. We're hoping however that we can get one debug build where the issue still happens. Otherwise, the debug code has to be left in the main build until the issue surfaces again.

Is there anywhere I can connect a bus-sniffer on the minirambo to record something while printing with the original 3.10 firmware that could be helpful?

leptun, 2021-06-12:
Unfortunately no. The only debug interface would be the JTAG, but those pins are used for something else.

djakkk, 2021-06-12:
> Unfortunately no. The only debug interface would be the JTAG, but those pins are used for something else.

How about using the debugWire interface?

djakkk, 2021-06-12:
> > Unfortunately no. The only debug interface would be the JTAG, but those pins are used for something else.
> 
> How about using the debugWire interface?

Looking at the schematics of the Rambo13 the Reset/debugWire-pin is availible on the X17/ICSP-header, so if the firmware has debugWire enabled that shouldnt that work?

leptun, 2021-06-12:
Afaik the atmega2560 doesn’t have debugWire capability

djakkk, 2021-06-12:
> atmega2560

Ah, I read somewhere that the minirambo has a ATmega32U2, you are correct, the 2560 does not have debugWire :(

IntegersOfK, 2021-06-12:
Great, ok I turned on Serial Logging in OctoPrint, added a line that says D23 at the top of the gcode, flashed my Rambo (1a) with your .hex file. Now I'm in the middle of a first print with it, but I've often been able to get this particular print to work the "first time" so I intend to make quite a few over the weekend to try to catch the error in the logs. Will keep you posted, thanks!

wavexx, 2021-06-12:
As an extra info, after "D23" is issued, an extra item will appear in the Support menu: "Memory dump". This can be useful to know if the gcode is set up for logging (do _not_ select it in this case: it triggers a manual dump which will reset the printer just like a crash).

I added this feature just in case we still get the printer to repeat moves, but doesn't crash. In such cases you can trigger the dump manually as soon as it happens.

IntegersOfK, 2021-06-13:
Ok so it happened fairly soon after starting the print (layer 3 I think?) and it did the regular thing where it was repeating a section over and over. When I've let it repeat a few times, eventually it usually gives the "Static memory" error but it was just repeating many times with lots of plastic starting to scrape up against the nozzle so I triggered the manually "Memory dump" from the support menu.

This file should contain the entire print up to and including the memory dump, and I've pasted from a bit before until that:
[serial-AJW-firmware-debug-memory-dump.log](https://github.com/prusa3d/Prusa-Firmware/files/6644720/serial-AJW-firmware-debug-memory-dump.log)

```
2021-06-13 17:43:50,378 - Recv: T:210.9 /210.0 B:60.0 /60.0 T0:210.9 /210.0 @:72 B@:17 P:39.2
2021-06-13 17:43:50,579 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:50,583 - Recv: SD printing byte 1642830/11990356
2021-06-13 17:43:50,584 - Recv: 00:33
2021-06-13 17:43:50,585 - Recv: ok
2021-06-13 17:43:50,698 - Recv: NORMAL MODE: Percent done: 13; print time remaining in mins: 188; Change in mins: -1
2021-06-13 17:43:50,706 - Recv: SILENT MODE: Percent done: -1; print time remaining in mins: -1; Change in mins: -1
2021-06-13 17:43:51,340 - Send: M27
2021-06-13 17:43:51,530 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:51,534 - Recv: SD printing byte 1643850/11990356
2021-06-13 17:43:51,535 - Recv: 00:33
2021-06-13 17:43:51,536 - Recv: ok
2021-06-13 17:43:52,340 - Send: M27
2021-06-13 17:43:52,382 - Recv: T:210.1 /210.0 B:60.1 /60.0 T0:210.1 /210.0 @:85 B@:10 P:39.3
2021-06-13 17:43:52,504 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:52,508 - Recv: SD printing byte 1645153/11990356
2021-06-13 17:43:52,510 - Recv: 00:33
2021-06-13 17:43:52,511 - Recv: ok
2021-06-13 17:43:53,341 - Send: M27
2021-06-13 17:43:53,508 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:53,512 - Recv: SD printing byte 1646834/11990356
2021-06-13 17:43:53,513 - Recv: 00:33
2021-06-13 17:43:53,514 - Recv: ok
2021-06-13 17:43:54,342 - Send: M27
2021-06-13 17:43:54,389 - Recv: T:209.4 /210.0 B:60.0 /60.0 T0:209.4 /210.0 @:98 B@:15 P:39.1
2021-06-13 17:43:54,757 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:54,759 - Recv: SD printing byte 1647900/11990356
2021-06-13 17:43:54,760 - Recv: 00:33
2021-06-13 17:43:54,761 - Recv: ok
2021-06-13 17:43:55,343 - Send: M27
2021-06-13 17:43:55,540 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:55,541 - Recv: SD printing byte 1648666/11990356
2021-06-13 17:43:55,544 - Recv: 00:33
2021-06-13 17:43:55,544 - Recv: ok
2021-06-13 17:43:56,344 - Send: M27
2021-06-13 17:43:56,392 - Recv: T:209.7 /210.0 B:60.1 /60.0 T0:209.7 /210.0 @:92 B@:12 P:39.3
2021-06-13 17:43:56,461 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:56,465 - Recv: SD printing byte 1649422/11990356
2021-06-13 17:43:56,467 - Recv: 00:33
2021-06-13 17:43:56,468 - Recv: ok
2021-06-13 17:43:57,345 - Send: M27
2021-06-13 17:43:57,580 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:57,584 - Recv: SD printing byte 1650855/11990356
2021-06-13 17:43:57,584 - Recv: 00:33
2021-06-13 17:43:57,586 - Recv: ok
2021-06-13 17:43:58,345 - Send: M27
2021-06-13 17:43:58,399 - Recv: T:209.5 /210.0 B:60.1 /60.0 T0:209.5 /210.0 @:97 B@:0 P:39.3
2021-06-13 17:43:58,555 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:58,558 - Recv: SD printing byte 1652056/11990356
2021-06-13 17:43:58,560 - Recv: 00:33
2021-06-13 17:43:58,561 - Recv: ok
2021-06-13 17:43:59,348 - Send: M27
2021-06-13 17:43:59,779 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:43:59,783 - Recv: SD printing byte 1653012/11990356
2021-06-13 17:43:59,785 - Recv: 00:33
2021-06-13 17:43:59,786 - Recv: ok
2021-06-13 17:44:00,348 - Send: M27
2021-06-13 17:44:00,407 - Recv: T:210.2 /210.0 B:59.9 /60.0 T0:210.2 /210.0 @:86 B@:54 P:39.3
2021-06-13 17:44:00,554 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
2021-06-13 17:44:00,554 - Recv: SD printing byte 1653948/11990356
2021-06-13 17:44:00,557 - Recv: 00:33
2021-06-13 17:44:00,558 - Recv: ok
2021-06-13 17:44:01,349 - Send: M27
2021-06-13 17:44:01,463 - Recv: debug_Fabric_of_Thyme_2.0_10x10_0.35mm_PLA_MK2.5SMMU
[... 663 lines cut ...]
2021-06-13 17:44:28,317 - Recv: 20b0  b2 42 c7 24 99 42 62 00 c8 41 00 cc 4c 3d 20 14
2021-06-13 17:44:28,322 - Recv: 20c0  02 00 00 13 00 00 41 00 00 00 00 c8 b8 3c 00 00
2021-06-13 17:44:28,326 - Recv: 20d0  00 00 20 42 20 41 be 01 8e a2 01 8e 8c 00 76 89
2021-06-13 17:44:28,330 - Recv: 20e0  20 f3 01 00 49 9a c2 a8 44 88 b3 40 cd cc 0c 3f
2021-06-13 17:44:28,334 - Recv: 20f0  d8 08 77 00 fe be 00 bd 05 1a 00 00 7e 26 91 80
2021-06-13 17:44:28,338 - Recv: 2100  00 97 e8 21 1a 3f 0c cc cd 08 d8 42 20 00 00 be
2021-06-13 17:44:28,342 - Recv: 2110  9b 23 00 00 01 01 62 9b 00 00 00 02 3d da 7f 7e
2021-06-13 17:44:28,346 - Recv: 2120  80 00 00 00 00 00 01 8e a2 00 75 de 01 83 a6 83
2021-06-13 17:44:28,350 - Recv: 2130  80 b2 42 00 2e 32 be 00 00 00 20 42 01 34 31 21
2021-06-13 17:44:28,358 - Recv: 2140  00 01 00 49 00 95 a6 00 49 21 64 00 00 93 9a c2
2021-06-13 17:44:28,362 - Recv: 2150  21 64 01 00 49 9a c2 a8 44 88 b3 40 44 88 b2 65
2021-06-13 17:44:28,367 - Recv: 2160  04 0a 00 33 6c fe be 4a a9 11 b1 0c e7 4c 4a 2d
[... cut ...]
2021-06-13 17:44:28,387 - Recv: 21b0  32 b4 01 2f 11 00 4b 3b 09 9f 21 e1 09 65 01 93
2021-06-13 17:44:28,395 - Recv: 21c0  01 00 7d 74 01 86 82 e7 00 00 ea 60 00 21 e1 00
2021-06-13 17:44:28,399 - Recv: 21d0  00 0d a1 01 e9 00 00 05 01 00 c3 62 76 04 0a 01
2021-06-13 17:44:28,403 - Recv: 21e0  77 34 03 00 0a 00 00 00 7f 00 94 ff 21 02 03 00
2021-06-13 17:44:28,407 - Recv: 21f0  1b 77 40 42 70 00 00 68 00 e6 21 2e e7 00 2f 18
2021-06-13 17:44:28,408 - Recv: ok
2021-06-13 17:44:28,410 - Send: M27
2021-06-13 17:44:29,431 - Recv: ;S
```

wavexx, 2021-06-14:
@IntegersOfK  Going to have a look later today. I'm glad I did the "manual" dump.. it was almost kind of an afterthought.
Any chances you can also make an image of the sd card that you used? This way we can also resolve the reading locations inside the card.

IntegersOfK, 2021-06-14:
I caught the issue again and I was keen to let it continue looping until the "Static memory error" came up, so I took out the filament and let it continue. It went for a few hours before I finally gave up waiting for it. So I'm not sure what extra conditions were happening to have that error output, but either way the main symptoms remain the same (repeated section over and over).

I'm having a hard time getting Ubuntu to make a copy of the SD card (can't seem to read after it's unmounted) and I don't have a lot of experience with volume mounting or cloning. I will try again on my Raspberry Pi this time, but in the meantime I used 7zip in windows to make a copy of the drive as an archive. I realise that might not be as useful in terms of knowing the positions or whatnot on the card though so I will keep at it on a unix system. For now though, here's everything on the card at least:

[sdcard-prusadebug.zip](https://github.com/prusa3d/Prusa-Firmware/files/6648980/sdcard-prusadebug.zip)

wavexx, 2021-06-14:
The best way on ubuntu to image the file is to drop on the command line, and use ``dd``. You need to know the name of the device (usually /dev/sd*) which you can recover by typing ``sudo dmesg`` just after plugging the card. For example on my system:

```
[74221.582316] sd 0:0:0:0: Attached scsi generic sg0 type 0
[74221.802839] sd 0:0:0:0: [sda] 31277056 512-byte logical blocks: (16.0 GB/14.9 GiB)
[74221.803833] sd 0:0:0:0: [sda] Write Protect is off
[74221.803835] sd 0:0:0:0: [sda] Mode Sense: 21 00 00 00
[74221.804880] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[74221.823062]  sda: sda1
[74221.826198] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

The device in this case is "sda" (without the 1).
Then you can make an exact image with:

```
dd if=/dev/sda of=/home/[username]/sdcard-image.img
```

On windows I have no idea TBH. @leptun are you aware of some utility to make raw disk images?

djakkk, 2021-06-14:
> 
> On windows I have no idea TBH. @leptun are you aware of some utility to make raw disk images?

One windows tool to make raw copies is Win32 disk imager:

https://sourceforge.net/projects/win32diskimager/

IntegersOfK, 2021-06-14:
Ok I was able to do it with my RaspberryPi with your instructions. It's big obviously because it's a whole card image!

I made no changes to the card, it's exactly as it was when giving the error:
https://drive.google.com/file/d/1c2CFAzDe7WF1Gp5nalQ4kkYLewjeCZI7/view?usp=sharing

wavexx, 2021-06-17:
@IntegersOfK I went a bit further with the decoding.. for some reason the stack guard was still intact, but I call this a fluke: the stack was fully corrupted already by the time you triggered the dump.

I have a couple of theories.

Did you use "live Z" during the first layer of that print?
Was the filament already loaded during the whole print?

IntegersOfK, 2021-06-17:
Definitely did not touch the "Live Z" adjustment, because the calibration was already pretty good/consistent. I have had no need to edit the height of any layer since the last time I cleared the memory (which was a few dozen prints ago as an attempt to see if it would solve this problem). Haven't touched that in the menu since.

I have the MMU2S which loads the filament at the start of the print. Sometimes I do have to "nurse" it but I'm fairly sure with this particular print it was loaded normally by that start procedure. Sometimes that MMU confuses me with the order that it tries to load and overall feels inconsistent in itself, so I would tend to agree that when the filament is loaded (sometime before, after, or during the initial pre-print calibration) is unknown. In any case, by the time the initial filament purge began, it was loaded properly. Is there more to try here? Should I load "load to nozzle" before the print and try to capture the problem again?

Are more serial dumps helpful? I can make a few more for you to compare if you like.

wavexx, 2021-06-17:
Ok, that narrows it down further. I suspected an issue with the long-press handling, but the MMU handling is also .. uuhhhr, let's say not fantastic.

I'm going to prepare a new FW with the updated dumper in a couple of minutes.

I think it would be helpful if you could test it again, with the following setup:

- filament fully loaded to nozzle
- reset the printer (this ensures we start with a clean slate)
- start the print with D23 as did previously

If this still triggers the issue we know it's not the MMU filament loading code.

wavexx, 2021-06-17:
Ok, there's the updated FW (includes updated PR #3180, #3157 and #3191).

[1_75mm_MK25S-RAMBo13a-E3Dv6full.zip](https://github.com/prusa3d/Prusa-Firmware/files/6672545/1_75mm_MK25S-RAMBo13a-E3Dv6full.zip)

As said before, a reset just prior to print with the filament already loaded makes it much easier to debug.
Two scenarios would be useful:

- trigger the dump as soon as you see it's starting to repeat, without waiting
- let it continue until (and if) it self-triggers

:crossed_fingers:

IntegersOfK, 2021-06-17:
Understood, flashing it now and will try to catch it right away. Will report back the moment I have any results!

djakkk, 2021-06-17:
I can add that I have not had my mmu2 connected when I have had these problems.

wavexx, 2021-06-17:
@djakkk that's even better. Any chance you can also try to get it to crash and generate a dump using the above FW?

djakkk, 2021-06-17:
> @djakkk that's even better. Any chance you can also try to get it to crash and generate a dump using the above FW?

I have not had time to sit by the printer in a couple of days and I’ve got a lot of stuff I have to do in the next couple of days too, but if it gets to warm outside tomorrow I’ll try to spend breakfast lunch and dinner inside and I’ll try to do a couple of prints then, do you want me to use the latest debug firmware or the first one?

wavexx, 2021-06-17:
Please use the latest one. I've improved the dumper further.

djakkk, 2021-06-18:
> Please use the latest one. I've improved the dumper further.

Bad news again I'm afraid.. the print worked just fine :(

If I get some time later tonight I'll try to print the same file with the original 3.10 firmware to see if the problem appears again.

wavexx, 2021-06-18:
Please keep on printing with the new updated FW. The bug wasn't deterministic, so it might not happen all the time.
On the other side, if we get a crash, I have much better chances to find the cause with the new FW.

IntegersOfK, 2021-06-18:
I've printed four times now without seeing the issue, doing another one now. I'm sure I'll catch it here eventually, I think we've pretty much established that it's a very intermittent thing. Though I must say I did usually see the problem after the second print on the other firmware builds.

My only change in process has been that I have been making sure the filament has been "loaded to nozzle" before initiating the print from the menu. Will keep you posted.

djakkk, 2021-06-18:
> Please keep on printing with the new updated FW. The bug wasn't deterministic, so it might not happen all the time.
> On the other side, if we get a crash, I have much better chances to find the cause with the new FW.

I don’t have the time to sit next to the printer in the next couple of days unfortunately, but I will give it a go the next time I can sit in the office.

I want to add though that I had problems literally 100% of all prints I tried to do with the original 3.10, trying 4-5 different gcode-files.

wavexx, 2021-06-18:
@IntegersOfK  I'm somewhat positive the actual "printing" part is good. I had doubts around some of the code that managed the long-press and performed the autoreporting, but I've fixed both in this FW to exclude all problems.

I suspect it might have to do to something done before the actual printing, which is why makes it hard. MMU filament loading could be one. Maybe SD handling could be another. Just keep an eye on things you might be doing before the print, this can be very useful.

@djakkk No problem. Chasing these hard-to-reproduce bugs is always annoying, and I really appreciate that you're helping as much as you do.

djakkk, 2021-06-18:
> @IntegersOfK I'm somewhat positive the actual "printing" part is good. I had doubts around some of the code that managed the long-press and performed the autoreporting, but I've fixed both in this [...]

Regarding your thoughts about doing something before the print or changing Z during the first layer:
I haven’t been using this printer for almost a year, so I have been adjusting live-z during some of these prints, and it is possible that I have done so some time during all “printer power-ons”, but I have not changed the live-z during my tries with the debug-firmwares since I’m satisfied with the first layer…

I will do one more try in a couple of minutes and I will change the live-z a couple of times during the first layer to see if that triggers something.

djakkk, 2021-06-18:
Yay, it crashed!

The only thing I did differently this time was to adjust the live-z one time.

The last time I raised the Z-axis to the top before I started the print, and I did that with the long-press function, so that long-press does not seem to have done any damage.

I made a image file of my sd-card, the image, the g-code and the serial-log can be downloaded here:

https://1drv.ms/u/s!AnnDJmPZ499ygfcf9gJkQ-K1rzQl1w?e=ah8CJE

[50 more replies]

closed 2024-02-07 by 3d-gussner