prusa3d/Prusa-Firmware #3153: Starting SD print from LCD with queued moves causes crash

wavexx, 2021-05-20:
I found a similar bug that might be actually the same thing as this one. Instead of inserting filament, move Z axis over a long range (e.g. full top to full bottom) with quick knob rotation, exit the move screen, and while the axis is still moving, start a print from SD. The printer then restarts. I have found this by accident on 3.9.1, then repeated it to see if it's reproducible. It was, and I also saw an error message about static data being overwritten, before the restart. Later I noticed that my print stats became obviously corrupted - the filament counter was 12km instead of expected ~2km, and print time was changed by 50 days over cca 40 days (not running 24/7 anyway). I suspected this was related to the crashes, so third time (after an upgrade to 3.9.2) I took a picture of the stats before and after the crash, and also recorded the whole thing on video - https://photos.app.goo.gl/16GT5vvX96W5wycJ6 and indeed, both counters got a bump (800m of filament, 39 days of printing).
To sum up:
- crash is reproducible on 3.9.1 and 3.9.2
- how much stats get bumped seems to be nondeterministic

As my stats are already busted, I can try anything that involves their reset now. If there's a way to reset to non-zero values, I wouldn't mind doing that aftewards, to at least be somewhat close to reality (something like 50 days printing, 3000m of filament).

I can also try if the filament autoload variant screws the stats too.

_Originally posted by @tehcaster in https://github.com/prusa3d/Prusa-Firmware/issues/2651#issuecomment-751396442_

wavexx, 2022-08-20:
@tehcaster this was fixed as a result of https://github.com/prusa3d/Prusa-Firmware/pull/3551

It would be nice if you could double-check on your end and report back.

leptun, 2022-08-20:
I wonder if #3511 was the real cause of broken stats in general.

tehcaster, 2022-12-05:
Hello, finally got to checking the fix as I've updated the fw to 3.12.0-RC1. Before the update (from 3.11.1-RC1) I tried it one more time and reproduced the crash. Seems a crash dump was saved too. I can try dig it out if it's considered useful at this point, unless the subsequent firmware update deleted it already. I hope PrusaLink (which I'm using on an embedded rpi zero w 2) learns how to download the dumps soon so one doesn't need to use extra software for that?

Anyway, with 3.12.0-RC1, the crash didn't happen again. The gcode I attempted to print was checked as usual and then it was waiting (still showing 100% progress bar of the checking) for the Z axis to finish the move, and then went on printing without any crash. Thanks!

Prusa-Support, 2022-12-28:
Hi guys. Because 3.12.0-RC1 fixes this, can the issue be closed, or should we wait for 3.12.0 stable version?

Michele Moramarco
Prusa Research

wavexx, 2022-12-30:
I think we can close as resolved

closed 2022-12-30 by wavexx