prusa3d/Prusa-Firmware #3154: [BUG] Octoprint M601 pause resumes without nozzle temperature control

mjroux1984, 2021-05-20:
MK3S+
3.10.0

MMU2S
1.0.6

USB/Octoprint

**Describe the bug**
When gcode is run from Octoprint/USB with M601 pause embedded through Prusaslicer 2.3.1 layer height function, the printer will pause correctly, but after resuming from either the printer or the octoprint server, the printer will attempt to print without controlling temperature of the nozzle (nozzle will be cooled off to room temperature).  In addition, the LCD screen on the MK3S will go blank (backlight stays on, but not txt is shown).  This issue doesn't occur when same gcode is run from sd card.  When firmware is downgraded to 3.9.3, the issue does not occur over USB/Octoprint.
  
**To Reproduce**
Any gcode produced from Prusaslicer 2.3.1 with inserting pause at specific layer height
Print over USB/Octoprint
Use firmware 3.10.0 on MK3S+ and firmware 1.0.6 on MMU2S
  
**Expected behavior**
When resuming print, the printer would wait for nozzle temperature to reach required temperature.  Works as expected with firmware 3.9.3.
  
**G-code**
[pause_test_0.2mm_PETG_MK3SMMU2S_33m.zip](https://github.com/prusa3d/Prusa-Firmware/files/6517313/pause_test_0.2mm_PETG_MK3SMMU2S_33m.zip)

**Video**
none

wavexx, 2021-05-20:
Isn't this similar to https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-843442610 ?

mjroux1984, 2021-05-20:
> Isn't this similar to [#3148 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/3148#issuecomment-843442610) ?

Seems like it, my search for an existing bug wasn't adequate.  You may close it.

wavexx, 2021-05-21:
Can't close it, as a creator you should be able to.

mjroux1984, 2021-05-21:
This is a repeat of 3148.

closed 2021-05-21 by mjroux1984