prusa3d/Prusa-Firmware #3159: Implement Joseph's idea to allow fast SD card upload via SPI

marcosscriven, 2021-05-24:
**Is your feature request related to a problem? Please describe.**
Printing from SD card is more reliable, but involves having to constantly insert/remove it. Serial upload is possible via a Raspberry Pi, but very slow
  
**Describe the solution you'd like**
@josefprusa himself suggested a way to make this much faster: https://twitter.com/josefprusa/status/915315958545289222

Is there any chance this could be done please?

marcosscriven, 2021-05-24:
Note - this was a feature request over in the PrusaSlicer repo as well https://github.com/prusa3d/PrusaSlicer/issues/2067

scottmudge, 2021-05-25:
Doesn't it already read from the SD-card via the SPI interface anyway? I mean it's connected to the SPI pins. Besides, write speed doesn't really matter if you're just buffering the entire print to the SD card anyway.

The true benefit of printing from SD is not the speed, it's the fact that you don't have to ping-pong back every received command to tell the sender (PC/OctoPrint) that the command checksum was fine. SD card reads are just assumed correct, they lack checksums. 

The SPI read speeds from the SD card are, if I recall, somewhere around 250,000 baud. I can achieve this speed over the serial interface with my MeatPack plugin, and the print quality still isn't as smooth as the SD card, presumably because of the added latency of confirming every received command with a send-packet back to the PC. 

I believe there was already work done to allow sending files from the PC over serial to the SD card to buffer them, but I can't recall if that feature was ever finished.

leptun, 2021-05-26:
This would not be possible because the SPI lines on the J19 connector are not level shifted. That would mean the raspberry PI would get 5V to the pins whenever the atmega2560/tmc2130 would use the SPI bus. The PI might work just fine, or it could get fried since the GPIO is **not** 5V tolerant (https://raspberrypi.stackexchange.com/a/38974). Additional circuitry would be needed between the rPI and the Einsy Rambo to handle the level shifting on the SPI bus.
There is also the complication that there would be two SPI masters on that bus, so the firmware/pi would need to negotiate SPI bus access. This is possible to be implemented however as it's only software.

closed 2023-11-09 by github-actions[bot]