prusa3d/Prusa-Firmware #3160: [QUESTION] Long filename support on serial (M33)

aneillans, 2021-05-24:
I can see a number of issues / questions already raised around LFN support for OctoPrint, but no definitive closing answer, and given the current Prusa firmware doesnt support M33 command, is there currently any way to get the LFN on serial with any existing command?

If not, any objections to me adding support for the M33 command, and maybe some other specific "prusa specific lfn optimisations" that can then be added to OctoPrint as a prusa module?

leptun, 2021-05-24:
PrusaFirmware doesn't support the M33 command, but you can get the LFN from `M20 L` instead. Keep in mind that M23 requires the SFN. That's why both SFN and LFN are sent. Directories are handled in a different way compared to regular `M20`. Octoprint support for this L response is not here yet however.
https://reprap.org/wiki/G-code#M20:_List_SD_card
As for getting the LFN of a print started from the SD card, the M27 command gets a P parameter as well for LFN output.
Both of these features were added in 3.10

aneillans, 2021-05-24:
Thanks :) Definitely makes sense that command gets used instead of M33 looking at the notes! I'll have a look at the OctoPrint code and see about making a PR there to clear some complaints!

arekm, 2022-08-03:
````M20 L```` is supported starting from octoprint 1.7.0 (2021-11). This issue can be closed.

Prusa-Support, 2022-08-27:
Indeed the original question was answered and the issue has been silent for more than a year, so that it would be safe to close this issue now.
Thanks.

Michele Moramarco
Prusa Research

closed 2022-08-27 by Prusa-Support