prusa3d/Prusa-Firmware #3161: [BUG] Random hangs when printing

zeroflow, 2021-05-24:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.10.0

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** 1.0.6

**Octoprint**

**Describe the bug**
During printing, the printer stops for around a second, then continues agail

In the serial log, it looks like this:
```
Send: N19165 G1 X102.024 Y89.791 E0.17783*89
Recv: ok
Send: N19166 G3 X102.043 Y88.644 I5.888 J-0.470 E0.04426*127
Recv: ok
Send: N19167 G3 X102.436 Y88.286 I0.459 J0.109 E0.02158*91
Recv: T:206.1 /205.0 B:59.9 /60.0 T0:206.1 /205.0 @:83 B@:51 P:39.6 AN19167ok
Recv: T:206.5 /205.0 B:59.9 /60.0 T0:206.5 /205.0 @:78 B@:35 P:39.6 A:45.3
Recv: T:206.7 /205.0 B:59.9 /60.0 T0:206.7 /205.0 @:77 B@:33 P:39.6 A:45.3
Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N19168 M105*32
Recv: ok T:206.7 /205.0 B:59.9 /60.0 T0:206.7 /205.0 @:77 B@:33 P:39.6 A:45.3
Send: N19169 G1 X104.434 Y88.285 E0.07695*80
Recv: ok
Send: N19170 G2 X105.777 Y88.578 I1.510 J-3.684 E0.05313*112
Recv: ok
Send: N19171 G2 X106.361 Y88.284 I0.016 J-0.694 E0.02603*118
Recv: ok
Send: N19172 G1 X115.030 Y88.280 E0.33396*94
Recv: ok
```
  
**To Reproduce**
Print any file.
  
**Expected behavior**
Printer prints without short pauses / hangs

**G-code**
[Lamp_Bedside_Adapter_v4_0.6n_0.15mm_PLA_MK3SMMU2S_1h2m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/6531992/Lamp_Bedside_Adapter_v4_0.6n_0.15mm_PLA_MK3SMMU2S_1h2m.gcode.zip)

**Video**
https://youtu.be/rDNKoP70idY
Short hang at 0:50, bigger hang at 2:03

DRracer, 2021-05-26:
@zeroflow let's narrow your issue down, as this line looks very suspicious:

```
Recv: T:206.1 /205.0 B:59.9 /60.0 T0:206.1 /205.0 @:83 B@:51 P:39.6 AN19167ok
```

The temperature report is not terminated correctly and a confirmation of line 19167 is printed instead. It could be a bug in the firmware, a communication disturbance, or a bug in OctoPrint.

- what version of OctoPrint are you using?
- does this issue occur with previous firmwares
- does this issue occur with previous versions of OctoPrint?

zeroflow, 2021-05-27:
Yes, looked suspicious to me aswell.

After your questions, I had another closer look at OctoPrint, and noticed, that it was running at 250kbaud.
This was an oversight of mine, from experimenting with MeatPacker and the baudrate itself.
Thus, 3.10 with the baudrate of 250k was flashed instead of the stock 3.10 from Github.

Now, with stock 3.10, this is no longer happening.

While this problem did not occur on 3.9 + 250k, and is now occuring on 3.10 + 250k, I think, this is not something you have to support.
I leave it up to you @DRracer to decide if you want to close this issue.

DRracer, 2021-07-30:
@zeroflow ok, closing the issue in favour of future communication updates which may bring higher comm speeds.

closed 2021-07-30 by DRracer