prusa3d/Prusa-Firmware #3166: Printer stuck with dark LCD after Firmware Upgrade

U-B-Glade, 2021-05-26:
Till now the Printer works without any Trouble.
I just try to upgrade the FW from 3.9.3 to 3.10.0, all works fine but after the writeout "verifying ...100%" the LCD gets Black (lights out) and no Restart occours. 
After a Reset the Priter starts, the LCD gets dark with "Original Prusa i3 Prusa Research ..." then the LCD Light flashes one Time and thats it.

I try do downgrade to 3.9.3 and 3.9.2 successfull but with all FW the same issue.
Is this a known issue?
What can I do?

3d-gussner, 2021-05-27:
How did you flash the firmware? Please use PrusaSlicer to flash the printer.
If you still have the same issue, please post here the advanced output log from PrusaSlicer firmware updater. (you may want to replace your username with something else before posting it)

leptun, 2021-05-27:
What is the hardware revision written on the pcb of the lcd assembly?

U-B-Glade, 2021-05-27:
@leptun The Version next to "A2004" is V3.0 on the blue PCB I see no Version

@3d-gussner I flash with Slicer V2.3.1 over USB. 
Did you mean the Flash Tool output or is there any other Logfile?
The output was:
avrdude-slic3r -v -p atmega2560 -c wiring -P COM7 -b 115200 -D -U flash:w:0:C:\_Transfer\Prusa\prusa3d_fw_3_10_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK25S_3_10_0_4481_RAMBo10a_en-de.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on Apr 16 2021 at 15:52:10
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : COM7
                Using Programmer              : wiring
                Overriding Baud Rate          : 115200
                AVR Part                      : ATmega2560
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
                  flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : Wiring
                Description     : Wiring
                Programmer Model: AVRISP
                Hardware Version: 15
                Firmware Version Master : 2.10
                Vtarget         : 0.0 V
                SCK period      : 0.1 us

avrdude-slic3r: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude-slic3r: Device signature = 0x1e9801 (probably m2560)
avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: reading input file "C:\_Transfer\Prusa\prusa3d_fw_3_10_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK25S_3_10_0_4481_RAMBo10a_en-de.hex"
avrdude-slic3r: writing flash (223116 bytes):

Writing | ################################################## | 100% 38.84s

avrdude-slic3r: 223116 bytes of flash written
avrdude-slic3r: verifying flash memory against C:\_Transfer\Prusa\prusa3d_fw_3_10_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK25S_3_10_0_4481_RAMBo10a_en-de.hex:
avrdude-slic3r: load data flash data from input file C:\_Transfer\Prusa\prusa3d_fw_3_10_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK25S_3_10_0_4481_RAMBo10a_en-de.hex:
avrdude-slic3r: input file C:\_Transfer\Prusa\prusa3d_fw_3_10_0_MK3S_1_0_6_MMU2S\prusa3d_fw_MK25S_3_10_0_4481_RAMBo10a_en-de.hex contains 223116 bytes
avrdude-slic3r: reading on-chip flash data:

Reading | ################################################## | 100% 28.59s

avrdude-slic3r: verifying ...
avrdude-slic3r: 223116 bytes of flash verified

avrdude-slic3r: safemode: hfuse reads as D0
avrdude-slic3r: safemode: efuse reads as FD
avrdude-slic3r: safemode: Fuses OK (E:FD, H:D0, L:FF)

avrdude-slic3r done.  Thank you.

3d-gussner, 2021-06-01:
@U-B-Glade I thought you had a MK3/S/+ as you didn't fill out the bug report template.
From the PrusaSlicer firmware updater we can see that you have a MK2.5S.
Which miniRAMbo version do you have on your printer?
You flashed the MK2.5S version for the miniRAMbo 1.0.a (a very old board)

Please double check on your controller board which version it is. See picture https://help.prusa3d.com/en/article/blown-fuse-mk2s-mk2-5-mk2-5s_2111 close to the USB port.

U-B-Glade, 2021-06-01:
@3d-gussner
I try to find the bug report template, but didn't find it, here are the Printer Infos
Type - MK3s
SN - CZPX0120004XK63663
Packed -11.09.2020
MMU - MMU 2
Einsy Board - V1.2b

I double check the Fuses, once by viewing, second with a Multimeter they are all fine.

3d-gussner, 2021-06-01:
According to the Prusa Slicer Firmware log you flashed the firmware for the MK2.5S and not the MK3S hex file.
Please remove your Serial number from the previous comment.

Try again to flash the correct hex file. https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.10.0/prusa3d_fw_MK3S_MK3S+_3_10_0_4481.hex

3d-gussner, 2021-06-01:
Please let us know if it worked. Consider also then to close the issue.

U-B-Glade, 2021-06-02:
Hi,

it works fine, I close the Issue.

The mistake was that I load the "prusa3d_fw_MK25S_3_10_0_4481_RAMBo10a_en-de.hex" cause I wont the German language, In the new Zip there is only one *.hex.

"hardly you do it right, it works" thanks for the Support.

closed 2021-06-02 by U-B-Glade