prusa3d/Prusa-Firmware #3171: [BUG] MMU Eject requires nozzle to heat up

Pun-e, 2021-06-01:
**Printer type** - MK2.5S
**Printer firmware version** -  3.10
**MMU Upgrade** - MMU2S
**MMU upgrade firmware version** - 1.0.6

**Describe the bug**
When a user attempts to eject filament they first need to wait for the nozzle to heat up. This defeats the purpose of the eject function as the process of removing filament from the nozzle **with an MMU** is already handled by the Unload function.

Even if it's still decided that Eject must have the ability to run the Unload function before ejecting, it would be a simple matter to check the Filament Sensor first to save users 1-2 minutes every time they want to cut a fresh tip.

There is already a separate ticket at 2676 for an "eject all" function, but together these issues would mean that users can "eject all" and systematically cut all 5 tips in a matter of seconds versus the minutes it may require at present.
  
**Expected behavior**
For the printer to eject the filament, when I select the option to eject the filament.

DaReaLDeviL, 2021-06-02:
Good point annoyed me always as well. New behavior whould be nice.

shred, 2022-06-13:
It's the same for "cut filament". It should also not need to heat up the nozzle first.

Prusa-Support, 2022-06-25:
Hello! Thank you for your feedback and sorry for the late response from my side. I'll bring these aspects to our devs attention - no promises it would be implemented at this time, but it will be evaluated from our side. 

Alessandro Pantaleo
Prusa Research

VH-fixer, 2023-01-09:
Keeping the hotend working well and clean is crucial.  With an MMU2S the extra heat cycles without moving filament can lead to problems with the hotend.   This non-functional heat cycle has caused me problems with only a few months of use.  I had to replace the heatbreak as a result.    This is like saying you need to have the car running to change a tire. 

This is not for convenience, it is for system reliability. 

The filament sensor should be able to tell if filament is loaded in the hotend or not and avoid the heating cycle as needed with Unload and Eject with an MMU active.     I have found Eject is needed each time I change a spool, to cut the end and assure I can pull the filament out of the MMU.  I don't use unload anymore.

Prusa-Support, 2023-01-31:
Hello! Thank you for the ideas, all brought up to our devs with the newer idea from @VH-fixer - no guarantees that it might be implemented, but surely it will be taken in consideration. In case of more info from our side, we'll update in this thread. 

Alessandro Pantaleo
Prusa Research

gudnimg, 2023-06-22:
I believe we can close this issue after 3.13 is released. When FINDA sensor is NOT triggered, the printer does not try to preheat anymore. But if the FINDA sensor detects filament, the printer will preheat.

This also applies to the Cut filament operation.

gudnimg, 2023-07-31:
Closing as this is now resolved with FW 3.13.0 and MMU FW 3.0.0 :)

closed 2023-07-31 by gudnimg