prusa3d/Prusa-Firmware #3174: [BUG] Insert SD card with less than 3 files, does not always clear unused display lines.

tg73, 2021-06-02:
**MK3S**
**3.1.10**

**MMU2S**
**1.0.6 **

**SD card or USB/Octoprint**
  I use both, but in this case I believe that the RPi was plugged in via USB but octoprint was not connected.

**Describe the bug**
Insert an SD card with fewer than 3 files on it (including eg empty). The load from SD menu tries to show itself. However, sometimes, but not always, it will not clear the lines that are not filled with filenames.

Here's an example state after inserting an empty SD card (note that I had previously just remove the card then reinserted it, hence the "Card removed" message remaining):
[image]

Here's an example state after inserting an SD card with only 2 files:
[image]

**To Reproduce**
See above.
  
**Expected behavior**
The load from SD menu should always clear all lines of the display, even if fewer than 3 files loaded.

leptun, 2021-06-02:
Can you provide the SD image that generates this issue? I haven’t encountered such a scenario yet and it might be some specific edge case.

tg73, 2021-06-02:
@leptun I'll share the image with you privately. I'll email you via the email address on your github profile.

closed 2023-11-09 by github-actions[bot]