prusa3d/Prusa-Firmware #3175: [BUG] Crazy high nozzle temperature display with 3.10.0

tg73, 2021-06-02:
**MK3S**
**3.1.10**

**MMU2S**
**1.0.6 **

**SD card or USB/Octoprint**
Both, but in this case I think Octoprint was not connected.

**Describe the bug**
Generally, I've noticed some weird behaviour since updating to 3.10.0. Hard to pin down, but includes sometimes getting stuck at the boot screen and needing a power cycle. And also ended up in this state somehow, I think after doing some MMU load/unload from the menu after cancelling a print, but I can't be sure. It was a few days ago and I didn't have time to log the bug at the time, just took a photo. I remember that the temperature was gradually decreasing, so it looks like some weird multiplier got involved. Along with #3174 which I've just logged, and the changes announced in 3.10.0, the erratic behaviour smells to me (a professional software engineer) of memory corruption. Nothing like this has happened on my machine pre 3.10.0.

[image]

leptun, 2021-06-02:
Yes, we are chasing some memory corruption issue. Probably caused by the same bug that generates #3151 #3147

wavexx, 2021-06-06:
Indeed. The high temp which slowly reduces looks like some memory spraying happening in the PID regulation, which is another pointer going towards the temperature ISR. Which we also enable earlier in 3.10, which could also explain getting stuck early during boot.

Do you remember exactly when it got stuck? Can you reproduce it with the serial attached? (serial messages will show a little more info and could help pinpoint the moment).

tg73, 2021-06-06:
Sorry, I can't remember for sure the exact circumstances for it getting stuck during boot. Quite possibly it was when clicking "connect" in octoprint - so the printer does I guess a soft reboot (or at least, that's what it looks like). I'm not sure how to connect serial at the same time as diagnosing an interaction with octoprint which is connected via USB (running on an separate RPi).

wavexx, 2021-06-06:
In that case, what's the output of the serial log shown in octoprint? Completely empty?

wavexx, 2021-06-06:
Also.. @tg73, how easily can you reproduce this lock-on-boot issue? Does it happen >50% of the times you connect?

tg73, 2021-06-06:
It's happened only once. Right now I'm trying to reproduce, but so far no hang after around 50 reboots, a mix of reset button and octo disconnect/connect. Also trying a mix of SD card in/out, number of files on SD, guid vs near-identical names for files on SD etc. Now trying with preheat before reboot.

wavexx, 2021-06-06:
We have a few ideas in how we could fix this, but we're looking for a reproducible case so that we can ensure the bug is fixed and not just shifted into even rarer occasions.

tg73, 2021-06-06:
Ok, I've not been able to repro the hang on boot despite trying for an hour or so - but I have found a way to make the printer spontaneously reboot when listing  a specially-prepared (but perfectly valid) SD card. Might be completely unrelated, or might be a good way to trigger some of the memory corruption issues (@leptun - perhaps of interest?). I prepared an SD card with 400 files using the following powershell one-liner. Start with an empty SD, in powershell cd to SD card root, then:

```powershell
1..400 | %{ "# Hello World!" | Out-File "$(new-guid)$(new-guid)aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa.gcode" }
```

This will create 400 34-byte files named like:

```
01e0b130-b6f9-4e4f-b863-ac4d9def189506e4fdee-ff78-49db-a3db-51240a62dc71aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa.gcode
```

Insert this SD into printer. Accept "some files will not be sorted". When printer attempts to list SD contents, very slowly shows a couple of filenames and then reboots.

Clearly, not a realistic real-world scenario, but perhaps a useful torture test - as such, I won't log this as a separate issue unless requested to do so by a contributor.

leptun, 2021-06-06:
This shouldn't cause a crash. Will definitely test.

leptun, 2021-06-06:
Can confirm 3.10 crashes when trying to render the root of the fs. Not even disabling the sorting makes it work (same for alpha sorting). Maybe it just takes stupidly long and the WDT resets the printer.

leptun, 2021-06-06:
Aaaaand. Turns out the menu implementation can only handle up to 256 lines... So that's why items are overwritten.

tg73, 2021-06-24:
I've just seen the nozzle and bed temperatures showing a little down arrow symbol instead of the degrees symbol for a few seconds during menu-triggered preheat - with FW 3.9.3 (I had already backed off 3.10.0). Hopefully this will be covered by #3202.

wavexx, 2021-06-24:
I wonder how many one-time-issues happened due to #3202 in the past... :(

leptun, 2021-06-24:
@tg73 Nope. That's just a graphical bug. There are only 8 custom character "slots" available in the lcd memory, but we have someting like 11. Some banking is done for a few of them (picked so that they don't need to be shown at the same time on the screen). There is some broken banking handling with the status screen that I've fixed in one of my ancient PRs (actually, rewritten the hd44780 handling entirely), but the code isn't maintained anymore and the documentation is wrong. I might get back to it at some point. Just for reference: https://github.com/prusa3d/Prusa-Firmware/pull/2064

tg73, 2021-06-24:
Oh dear, it's going to be far too easy to blame everything on #3202...

wavexx, 2021-06-24:
> Oh dear, it's going to be far too easy to blame everything on #3202...

Not until the next FW is released, then every bug is genuine blame ;))

wavexx, 2021-06-29:
> Oh dear, it's going to be far too easy to blame everything on #3202...

While looking for other issues I regressed this bug to all versions down to FW 3.8.0 included.
I stopped because 3.8 is the last FW that I was able to rebuild from source...

DRracer, 2021-07-30:
@tg73 have you had a chance trying with FW 3.10.1-RC1? I know you reported a lot of similar bugs on your machine.

tg73, 2021-07-30:
I've been using 3.10.1-RC1 for a while. The only strange behaviour I've seen is the printer spontaneously setting the speed to 102% as described here https://github.com/prusa3d/Prusa-Firmware/issues/3225#issuecomment-885588564 and followed up (including dump) here https://github.com/prusa3d/Prusa-Firmware/issues/3225#issuecomment-885790455. I have also had quite a few hangs on the boot screen, fans at full power, when connecting octoprint - I'd say around 50% of the time - I've not made a record, but it feels quite common with 3.10.1-RC1, where previously it felt like quite a rare occurrence.

tg73, 2021-07-30:
Regarding the 102% speed thing - I've just noticed that when on the info screen (not printing, soon after turning printer on), moving the jog wheel changes the speed. I did not know that this was a feature. It's quite possible that I had unintentionally changed the speed by mistake myself, rather than the printer changing it spontaneously.

tg73, 2021-07-31:
Again regarding the spontaneous speed change - I noticed another print was running at 99% speed. I had started that print from octoprint. The only interaction with the printer since power on was to load filament (MMU, not to nozzle). I had not intentionally changed the speed. If this speed-change-from-info-screen is a new feature, it seems very easy to trigger accidentally, and my vote would be to remove the feature. Or alternatively, it was a spontaneous speed change not caused by me. I've never noticed this issue with previous firmware versions. Now it seems that I must add "check that the speed is 100%" to my checklist for every single print.

DRracer, 2021-08-02:
@tg73 these speed changes sound really suspicious - does this happen even without Octoprint? We have not seen anything like that so far.
Are you using the correct communication speed with Octoprint? We've been getting reports that 256kbaud does not work reliably.

closed 2023-11-09 by github-actions[bot]