prusa3d/Prusa-Firmware #3177: [BUG] Misleading Indentation if()

gudnimg, 2021-06-05:
I tried compiling the firmware and got a compiler warning `src\Marlin_main.cpp:3496:6: warning: this 'if' clause does not guard... [-Wmisleading-indentation]`

https://github.com/prusa3d/Prusa-Firmware/blob/a8a253c0ae115d9e4ec75adf62018436019a50d6/Firmware/Marlin_main.cpp#L3499

```cpp
if (result)
{
        if (calibration_status() == CALIBRATION_STATUS_Z_CALIBRATION)
	    // Shipped, the nozzle height has been set already. The user can start printing now.
	    calibration_status_store(CALIBRATION_STATUS_CALIBRATED);
	    final_result = true;
	    // babystep_apply();
}
```

My question is whether `final_result` should be set to `true` regardless of the if statement `if (calibration_status() == CALIBRATION_STATUS_Z_CALIBRATION)` because that is what is happening.

I propose to use brackets `{}` to improve readability and also fix the compiler warning. I can create a PR for this but would like someone to confirm how the bracket should be placed.

wavexx, 2021-06-05:
See https://github.com/prusa3d/Prusa-Firmware/pull/3016

gudnimg, 2021-06-05:
@wavexx thanks I’ll close this issue.

closed 2021-06-05 by gudnimg