prusa3d/Prusa-Firmware #3179: [BUG] Spurious filament change by firmware (not in GCODE)

bozzzzo, 2021-06-05:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version** -  prusa3d_fw_MK3S_3_10_0_4481.hex

**MMU Upgrade** - not installed
**MMU upgrade firmware version n/a

**SD card or USB/Octoprint** SD card FlashAir

**Describe the bug**
 I've observed unexpected filament change initiated by the printer in the middle of the print (like colorprint). At first I assumed I accidentally did that in the slicer, but I couldn't find any M600 commands in the Gcode. Today I'm printing a file that I printed ok twice on 3.9.? before the upgrade and with this exact same file  on the SD card I also got a filament change request. I'm sharing the shorter file as it's only a 2,5 hour print, the current one is 13 hours...
  
**To Reproduce**
Don't know how to reproduce.
  
**Expected behavior**
I expect to find the printer printing instead of waiting for an unwanted filament change :)
  
**G-code**
[balon-holder_0.2mm_ABS_MK3S_2h41m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/6603044/balon-holder_0.2mm_ABS_MK3S_2h41m.gcode.zip)

**Video**
No video available

JohnnyDeer, 2021-06-07:
Hi @bozzzzo, thank you for reporting. This issue may be related to #3147.

bozzzzo, 2021-06-07:
fwiw, after the third unsolicited filament change I turned off the filament sensor and the print completed without issues. I'm printing another one with filament sensor turned off. The print with the filament sensor disabled completed without issues.

JohnnyDeer, 2021-08-02:
Most likely you have faulty sensor. Please contact our Tech Support for help. Chat in right down corner https://shop.prusa3d.com/en/

closed 2023-11-14 by github-actions[bot]