prusa3d/Prusa-Firmware #3183: Print quality decrease since 3.10.0

m-cas, 2021-06-08:
MK3S, no MMU, via SD card

after installation of 3.10.0 parts that were previously printing perfectly started to show strange artifacts/extrusions and huge stringing.
Material: Prusament Urban Gray PETG.

GCODE and picture with bad result attached.

[image]

[holder.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/6615662/holder.gcode.zip)

tg73, 2021-06-08:
Just a thought (I'm not a FW dev): your gcode says it was sliced with PrusaSlicer 2.1.1, and contains linear advance v1.0 instructions (eg, `K900 K45`). You might want to try reslicing with PrusaSlicer 2.3.0 and see if that works better. Linear advance v1.5 was introduced in FW 3.9.0, and improved in 3.9.1. See the release notes [here](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.9.1). The firmware should automatically translate LA10 values to LA15, but if reslicing works, perhaps there's a regression in FW 3.10.0.

m-cas, 2021-06-09:
I've done the reslicing of the part using the latest PrusaSlicer (attached below).
The stringing is pretty much gone but the "warts", although smaller in size and less in number, are still there and they weren't with the previous FW, so something needs to be fixed.

[image]
[image]

[holder test 0.15mm_PETG_MK3S_2h9m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/6624361/holder.test.0.15mm_PETG_MK3S_2h9m.gcode.zip)

wavexx, 2021-06-10:
These staircase blobs usually form at the end of a long travel move when the retraction is insufficient. It's kind of odd though that these appeared now, and not earlier.

The settings in the gcode look normal. Could you provide also the 3mf project file of this piece?

It would help if you could also flash FW 3.9.3 to try to determine when the problem started.
You can cut the model in the last ~15mm from the top as shown to save testing time. Since these increase with travel distance, you just need (in this case) two pieces of the same part with some decent space between them to increase the problem.

m-cas, 2021-06-10:
The last version I had on the printer before 3.10 was 3.9.2 (and the printer complained about me skipping a round). This parts printed without the slightest stringing or warts/blobs.
I can try to find the time to test with 3.9.3 but I sort of wonder if this test makes sense at all... the code I've been used was sliced on Feb 7th 2020 and was printing flawlessly until 3.10...

wavexx, 2021-06-11:
@m-cas  LA1.5 (which is what _usually_ could have caused something like this) didn't change since 3.9.1 and I agree the gcode is sliced correctly.

But if this part works correctly with 3.9.2, it's helpful to know if there was a change introduced in 3.10 or in 3.9.3 that caused this (mostly because 3.10 includes a ton of changes).

wavexx, 2021-06-11:
Also, if you provide the 3mf file is easier to test

m-cas, 2021-06-11:
The 3mf is absolutely something I can provide if it helps fixing it faster (below). The test with 3.9.2 will have to wait a little (lot going on on the "real life" side) but I will try to do it.

[Holder (driver)_Side Extender Gen 1 v12.3mf.zip](https://github.com/prusa3d/Prusa-Firmware/files/6638626/Holder.driver._Side.Extender.Gen.1.v12.3mf.zip)

m-cas, 2021-06-25:
I finally found out time to install 3.9.3 - result exactly the same as 3.10 (warts/blobs and stringy all over).
I can confirm that with 3.9.2 it was working flawlessly.

wavexx, 2021-06-26:
I was checking for differences between 3.9.3 and 3.9.2, but couldn't spot anything that could influence this. On the printing/planning side nothing changed. We added temperature autoreporting, but you're clearly not using that when printing via SD so I would exclude any influence. The revised calibration shouldn't matter either.

Do you notice if the printer stutters at any point during the print?

I wonder if this isn't just due to heatcreep in the extruder. Anything obstructing airflow behind the noctua and around the heatsink?

I would flash again 3.9.2 and retest if it still prints perfectly.

tg73, 2021-06-26:
I've had similar problems when there's been a partial block in the nozzle, for me typically when switching between lower and higher temp materials (eg, PLA and ASA), then some PLA can get baked in, causing a mixture of under extrusion and blobs. If you have a spare clean nozzle, maybe try swapping, or do a cold pull. Also I've just started to suffer from seasonal hot end heat creep as the ambient summer temperature has risen (I'm in the UK) - it's my first summer of printing ASA, which seems to push things thermally. For me the heat creep was a strong candidate due to good-looking but oversized tips unloading with MMU2S, with molten material extending above the tip (and other causes ruled out). I've resorted to using the classic sunon fan instead of the noctua, but I've not had chance to print yet with the new fan.

wavexx, 2021-06-26:
True. I actually judge if the nozzle needs cleaning from the purge line. If the fat purge is not nicely flowing (typically thins out instead of getting larger), I run through some mildly abrasive cleaning filament at high temp. Silky pla does tend to make a lot of charring for me as well, so material swaps are not always the main cause. TPU is also especially nasty to clean off.

Still, before claiming anything, I'd try again a "known working" FW.

m-cas, 2021-06-28:
the purge line is consistently fine but I will due a cold pull just to make sure this is not the root cause. I did print with PLA and moved to PETG before the move to 3.10, so I will make sure that the two things didn't overlap. Will post feedback once done.

DRracer, 2021-07-29:
@m-cas any update on this issue?

JohnnyDeer, 2021-08-19:
This issue will be closed soon if autor will not respond.

m-cas, 2021-08-20:
I didn't have the time to do anything aside installing the FW 3.10.1-RC1 for another issue I did report. I will do a cold pull and try again soon. Feel free to close this if you need to, I will either reopen or create a new one if the issue represent itself.

DRracer, 2021-08-23:
@m-cas ok, fair enough, closing this issue, please feel free to file a new one if your issue still occurrs on 3.10.1-RC1

closed 2021-08-23 by DRracer