prusa3d/Prusa-Firmware #3184: Pause-Change Filament-Restart Print with 3.10.0 leaves gap in the object printed

m-cas, 2021-06-08:
mk3s, v3.10.0, no MMU, printing from SD card

I needed to change the filament as the old one was nearing the end, so I decided to:
- pause print
- unload filament
- load filament
  o the loading ejected a lot of material (a lot) before confirming the "color correctness"
- resume printing

The printing resumed leaving about 1cm gap in the layer being created. It seems that the material that was ejected before resuming was the "missing part"...

It worked well before 3.10.0

tg73, 2021-06-09:
I had a one-off issue with 3.10.0 with a scheduled pause (ie, added in PrusaSlicer) to allow a nut to be inserted. On resuming, "printing" continued, except the E axis stopped moving so no filament was extruded. I did check visually - the bondtech gear was not rotating at all, so not a jam etc. This was printing from octoprint. Subsequent prints from SD worked ok. However, I've since reverted to 3.9.3 due to other erratic behaviour (#3175, #3151, #3174) which the fimware devs suspect is due to memory corruption.

DRracer, 2021-07-30:
@m-cas could you please retest on FW 3.10.1-RC1? We (hopefully) found and fixed the memory corruption.

JohnnyDeer, 2021-08-19:
This issue will be closed soon if autor will not respond.

m-cas, 2021-08-20:
Installed FW 3.10.1-RC1 and issue cannot be reproduced.

JohnnyDeer, 2021-08-23:
@m-cas so that means issue is fixed and can be closed?

m-cas, 2021-08-23:
correct!

closed 2021-08-24 by JohnnyDeer