prusa3d/Prusa-Firmware #3186: [BUG] Spanish translation for MSG_UNLOAD_SUCCESSFUL is wrong

jfestrada, 2021-06-08:
**Printer type** - MK3S
**Printer firmware version**-  3.10

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
  The Spanish translation for MSG_UNLOAD_SUCCESSFUL is wrong
  
**To Reproduce**
  When the filament has run out and is detected by the IR sensor, the print will pause and the printer will ask to unload the filament, after the try of unloading the filament, the printer will ask "Was filament unload successful?" that translation is wrong on spanish because use "cargocon" instead of "descargo con", "cargocon" does not even exist on Spanish dictionary.
  
**Expected behavior**
 Show the following translation "Se descargo con exito el filamento?"
  
**Pull Request**
I have submitted the PR #3185 with the correct translation

JohnnyDeer, 2021-06-09:
Hi @jfestrada thank you for reporting and PR.

JohnnyDeer, 2021-07-29:
@jfestrada PR merged, thank you. I'm closing this issue as solved :)

closed 2021-08-02 by JohnnyDeer