prusa3d/Prusa-Firmware #3188: [BUG] Underextrusion on the first layer calibration with 0.6 and 0.8 nozzles

jfestrada, 2021-06-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3S
**Printer firmware version**-  3.10

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version 1.06

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
  Currently, the G-Code for the first layer calibration does not take care of the nozzle installed on the printer, so if you are using a 0.6 or 0.8 nozzle the first layer calibration menu is not valid because you will suffer under extrusion 
  
**To Reproduce**
 * Install a 0.6 or 0.8 nozzle
 * Execute the first layer calibration from the calibration menu
  
**Expected behavior**
  Adapt the flow to the installed nozzle in order to have a consistent first layer calibration
  
**PullRequest**
  I have created the PR #3187 with the implementation of one possible solution

JohnnyDeer, 2021-06-15:
Hi @jfestrada thank you for reproting. The nozzle size is not taken into account when calibrating the first layer.

jfestrada, 2021-06-15:
Hi @JohnnyDeer, I know that currently is not taken into account, because of that I submitted the PR #3187 in order to read the nozzle size from EEPROM and add the M221 to the queue just before starting to print the g-code for the first layer calibration.

gudnimg, 2023-06-22:
This can be closed after FW 3.13 is released :) Issue was fixed with https://github.com/prusa3d/Prusa-Firmware/pull/3312

Prusa-Support, 2023-08-31:
This issue has been addressed by FW [3.13.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.13.0), so will be closed now.

Michele Moramarco
Prusa Research

closed 2023-08-31 by Prusa-Support