prusa3d/Prusa-Firmware #3193: [BUG] Incorrect Nozzle Temperature into M112

trfunk, 2021-06-18:
Might be related to #3175.

**Printer type** - MK3S
**Printer firmware version**-  3.10.0-4481

**MMU Upgrade** - No

**SD card or USB/Octoprint**
  Octopi (rpi 4, connected via usb)

**Describe the bug**
Uploading files directly to the SD using octoprint (by dragging gcode on the right side of the octopi tab) displays very low (-600°C) and then very high numbers (13000°C+) both in octopi and on the printer screen. The software becomes sluggish aswell. After canceling once I get "M112 called emergency stop" and have to reset again. 

  
**To Reproduce**
1. Drag file on the right side of octoprint to upload
2. Wait for octopi to display "Streaming"
3. Temps go crazy
4. Press cancel button on printer to reset
5. "M112 called Emergency stop"
6. Press cancel again
7.  Back to normal
  
**Expected behavior**
File uploads and temps stay correct.
  
**G-code**
[test_Glasoffner_20SoL_0.8n_0.4mm_PETG_MK3S_1h35m.zip](https://github.com/prusa3d/Prusa-Firmware/files/6678608/test_Glasoffner_20SoL_0.8n_0.4mm_PETG_MK3S_1h35m.zip) (.gcode isn't supported).

**Video/Other Information**
https://user-images.githubusercontent.com/4313364/122601144-27a5ef00-d071-11eb-8da7-0c54485056b7.mp4
(the graphic artifacts are from my dying gpu)
https://user-images.githubusercontent.com/4313364/122601512-b1ee5300-d071-11eb-830e-6587ce84bb55.mp4
[systeminfo.txt](https://github.com/prusa3d/Prusa-Firmware/files/6678586/systeminfo.txt)
[octopi-log-cut.log](https://github.com/prusa3d/Prusa-Firmware/files/6678588/octopi-log-cut.log)
[image]
[image]
[image]

JohnnyDeer, 2021-06-21:
Hi @trfunk, thank you for reporting. This is definitely related to #3175 but here we have exact scenario. 
I'm trying to reproduce it, not success yet.

trfunk, 2021-06-21:
Hello,
I did some more investigating. Setting the printer to heat to specific temperature causes high temperature spikes. Same for colling. I can also smell that the filament is getting heated (prusament petg). The fan is also spinning that's what I tried to show at the end and failed at) although it doesn't seem to be 100% of the time. Sorry for the bad phone shaking. 

vid:

https://user-images.githubusercontent.com/4313364/122809704-bebbb280-d2ce-11eb-929e-bf89b4a05275.mp4

logs:
[serial_vid.log](https://github.com/prusa3d/Prusa-Firmware/files/6688698/serial_vid.log)
[octoprint_vid.log](https://github.com/prusa3d/Prusa-Firmware/files/6688699/octoprint_vid.log)

gcode:
[pikachu_3gen_flowalistik_0.8n_0.4mm_PETG_MK3S_4h10m.zip](https://github.com/prusa3d/Prusa-Firmware/files/6688708/pikachu_3gen_flowalistik_0.8n_0.4mm_PETG_MK3S_4h10m.zip) (any will do)

awenelo, 2021-06-21:
@JohnnyDeer, I can reproduce this by installing the [DisplayLayerProgress plugin](https://plugins.octoprint.org/plugins/DisplayLayerProgress/) on Octoprint, which it appears @trfunk has as well:
```
2021-06-20 16:44:11,855 - octoprint.plugin.core - INFO - 23 plugin(s) registered with the system:
...
|  DisplayLayerProgress Plugin (1.26.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress
```
In both my testing and in @trfunk's logs, the printer starts behaving weirdly when `N31 M117 INDICATOR-Layer1*40` is sent. This line doesn't appear in either of the gcode files attached, meaning that it was put in the gcode by a plugin. Running Octoprint in safe mode, or with the DisplayLayerProgress plugin disabled, causes this to not happen.

This appears to be two problems, one with the printer, leading it to report extreme temperatures, and one with the plugin, which put `M117 INDICATOR-Layer1` in the gcode but didn't change the output correctly (unless this is intentional behaviour?) Sending `M117 INDICATOR-Layer1` normally, both with the plugin enabled (changes it into `M117 0% L=1/-`) and with the plugin disabled doesn't cause issues, and the printer responds normally.

(Mentioning the author of the plugin - @OllisGit)

closed 2021-06-25 by DRracer