prusa3d/Prusa-Firmware #3197: [BUG] FW 3.10 Printer locked up changing mode during XYZ home.

vintagepc, 2021-06-21:
**Printer type** - MK3SMMU2
**Printer firmware version**-  3.10 + mods

**SD card or USB/Octoprint**
  SD card

**Describe the bug**
Printer was pre-heated and I (perhaps foolishly) went to Tune and changed from normal to stealth mode while it was in the process of doing the pre-MBL XYZ home. 

Printer completely locked up on the following screen and I had to use the manual reset button to recover. I did leave it for about 30 seconds since I expected to see a watchdog reset but did not. 
  
[image]

Checked temperatures after reset finished and both the heater and bed had cooled down - but that's not conclusive since I don't know if the freeze happened while both PWMs were off. It's entirely possible this might latch up the heater and should probably be investigated. 

Worth noting: I've merged a few of my open PRs into a custom 3.10 build. I do not think any of these would have directly caused this but worth noting just in case:

#2508 #2499 #2201 
  
**Expected behavior**
No lock-up

DRracer, 2021-07-23:
@vintagepc can you reproduce your issue in MK404 :wink: ? Or, is it reproducible at all?

vintagepc, 2021-07-23:
I have not had a chance to chase this down yet. Will have a go this weekend. :+1:

DRracer, 2021-08-09:
@vintagepc any progress on this issue :wink: ?

vintagepc, 2021-08-09:
I was not yet able to reproduce it with MK404 but it is certainly reproducible on real hardware. I've had some other priorities the last few days but intend to poke at it more soon.

wavexx, 2021-08-10:
I know why this happens. It's because the homing code 1) doesn't expect the mode to be changed - and I think that's to be expected and 2) changing mode calls st_synchronize(), however it's one of those cases where homing itself calls synchronize as well, so we're running two nested main loops.

It's probably not the only case where this can happen. All compound commands such as filament change sadly could lock up like this.

We can work-around this (had to do this for the power-panic while homing), but for a proper fix we'd need a way to improve the command queuing. We need to be able to schedule and wait for moves without affecting the main queue. The key thing is that we cannot change how st_synchronize works, or we need to change waaay to many spots in the code.

vintagepc, 2021-08-10:
That'd definitely explain it; if st_syncrhonize is not reentrant I'd expect undefined behaviour.

wavexx, 2021-08-10:
It's reentrant itself, but it doesn't guarantee which loop will unlock first, which breaks the code around it

ctrlbru, 2021-12-31:
I also changed to stealth during homing and instead of locking up it simply didn't home completely on the y axis, leaving about 1 cm from the bottom. It then hit the endstop on the other side during bed leveling, missed the screws with the pinda probe by a couple of mm and started printing at the wrong offset, with the priming line inside the print area. The print completed fine since it was small and centered.

closed 2023-11-14 by github-actions[bot]