prusa3d/Prusa-Firmware #3201: [BUG] MK3S FW 3.10 filament sensor not working anymore

nio101, 2021-06-23:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MK3s
**Printer firmware version**-  3.10

**SD card or USB/Octoprint**
SD card

**Describe the bug**
After upgrading to FW 3.10, my filament sensor isn't working anymore on my MK3S.
When switching back to FW 3.9.3 it works fine!
  
**To Reproduce**
upgrade to FW 3.10, insert filament... nothing happens.
  
**Expected behavior**
preheat options should be shown (or material/autoload options/dialogs)

nio101, 2021-06-23:
well it seems that flashing again FW 3.10 after that solved my problem... very odd... :s

closed 2021-06-23 by nio101

jlshelby, 2021-09-11:
Having the same issue. My MMU2S ran out of filament. I then loaded the new filament and it started printing but wasnt fully feeding the filament. When the filament from the MMU2S to the nozzle ran out..it continued to print with no filament and didnt stop as expected. It was basically shooting blanks and ruined my print instead of stopping.