prusa3d/Prusa-Firmware #3207: [BUG]<Layer shift after brief power interruption - MK3S (upgraded to MK3S+) (FW 3.10.0) plus MMU2S upgrade (FW 1.06)>

Tanischacat2009, 2021-06-24:
Yesterday, I was working on a 2 and a half day print and I was about 48 hours into it when a very brief power outage occurred, just enough to cause everything to flicker off and on, clocks and computers etc. to reset and of course the printer.  The power panic kicked in and the printer restarted and resumed right away.  I was printing my project in mmus2 single mode at the time.  After my printer restarted, all elements of the project had a 1 cm layer shift to the right.  The print had worked perfect up to that point and probably would have continued to work perfectly after that hadn't I canceled the print.  Of course, I had to restart.  I simply can't trust the power panic feature.  Fortunately, I do have a UPS where another printer is plugged into in another location, so I now have both printers plugged into the UPS.  That's what I get for living dangerously!

wavexx, 2021-06-25:
Where you there to see what was happening when it occurred, or did you just hear it?
Do you have a picture of the layer shift? Did it happen on X, Y, both?

Tanischacat2009, 2021-06-25:
I wasn't home at the time, but I was able to verify the time when the outage occurred and estimate the amount of printing that had been done, after the outage and when the shift started after the printer rebooted.  My UPS, which was connected to my other printer registered the outage as only 3 seconds, which is normal when the power just flickers off and back on again.

Unfortunately, I didn't think of snapping a picture before removing the damaged print from the plate and discarding it.  The shift only happened on the X axis, 1 cm to the right.

wavexx, 2021-06-25:
I would have loved to see that. Described like that, it sounds almost like homing on X didn't work properly for some reason.
@leptun do you know if when printing in single mode we still have the problem that the mmu motors drain the power too quickly?

wavexx, 2021-06-25:
@Tanischacat2009 if you have some spare filament, could you try to print a small test object in the same way, and pull the plug in order to make a test? (for convenience, do it after the 2-3rd layer, otherwise it's annoying to pull off the bed).

Make a video, and watch the extruder actively especially when resuming to see if there's something preventing the homing for some reason (the cable bundle being the usual suspect).

Tanischacat2009, 2021-06-25:
Thank you, I will try that when the printer is again free, which won't be till Monday at least.  Unfortunately, I'm doing the same project for the third time because this morning, I grossly underestimated the amount of filament left on the roll that I was using and not long after I left for work, the printer started "air printing" and continued to do so until I got home earlier this afternoon because the filament sensor didn't detect a runout.

It is quite possible that the cable bundle may have interfered with the homing process, as you mentioned, but I just posted this issue to make people aware of a possible bug in the firmware as I've had power interruptions with other prints in the past with printers not plugged into a UPS and there was only very slight layer shifting or none at all in those cases.

leptun, 2021-06-26:
> @leptun do you know if when printing in single mode we still have the problem that the mmu motors drain the power too quickly?

Could be. Even in single mode at least two of the steppers are active.

DRracer, 2021-07-23:
The MMU can drain the power and unless you hear a beep just before the power outage, the printer didn't manage to finish saving the data.

However - if the printer didn't manage to save the PP data, it wouldn't resume on power up, thus wouldn't continue printing - which is obviously not this case.

The issue described by the author sounds very much like a homing problem - I'd recommend checking the cables (do a few belt tests to see if the X-axis has proper length measured) and also the X-axis' bearings.

JohnnyDeer, 2021-08-19:
This issue will be closed soon if autor will not respond.

closed 2021-08-23 by JohnnyDeer