prusa3d/Prusa-Firmware #3216: [BUG]<Knob click doesn't work, won't load any menus>

ericflor, 2021-06-30:
**Printer type** - [ MK3]
**Printer firmware version**-  [3.10.0]

SD CARD

After upgrading printer firmware to 3.10.0, the knob for clicking through menus was no longer working. It would only work when doing a factory reset and would allow me to choose the language but after that, nothing. So after doing a few factory resets and refreshing the firmware to try and fix the issue, the screen then got stuck on the welcome title and never moved on to the main menu. Replaced both the LCD screen and Einsy board and got it to work again but then when I flashed the firmware to 3.10, the same issue happened. So downgraded to 3.9.3 and it's working fine. Definitely an issue with the firmware.  
  

https://user-images.githubusercontent.com/46897396/124012233-cac20700-d995-11eb-88d3-0420eabd99d7.mp4

leptun, 2021-06-30:
Would anything else in the firmware work? Just looking at the video, it appears that the entire firmware locks up after startup. The temperatures on the screen don't seem to update, which could be related to the encoder not registering.
Have you tried connecting to the printer via the usb port to see if anything interesting pops up there? Or maybe if it really is the entire firmware that locks up and not just the UI part.

ericflor, 2021-06-30:
https://user-images.githubusercontent.com/46897396/124014277-2ab9ad00-d998-11eb-8336-cfcf530e9c5d.mp4

ericflor, 2021-06-30:
ya it seems like the entire firmware, just posted another video where it doesn't even let me get past the welcome screen. Downgrading to 3.9.3 however fixes the issue

ericflor, 2021-06-30:
https://user-images.githubusercontent.com/46897396/124014735-bf240f80-d998-11eb-8d1e-fd5be67fcb71.MOV

leptun, 2021-06-30:
Can you still repeat the case where it entirely locks up before booting the status screen is reached? If yes, please create a log with everything sent by the printer on the serial port during boot. I'm curious where it gets stuck. There has been a similar case where it was getting stuck right after "adc_init".

ericflor, 2021-06-30:
here's a screenshot of the log, most recent connection is at the bottom 
[image]

ericflor, 2021-06-30:
here's a screenshot of the log, most recent connection is at the bottom, thanks!
[image]

ericflor, 2021-06-30:
looks like it makes it all the way through so this is a weird bug

wavexx, 2021-06-30:
That's actually far enough to be #3202.

@ericflor could you try this updated firmware (current version for the MK3):

[1_75mm_MK3-EINSy10a-E3Dv6full-1279a6c.zip](https://github.com/prusa3d/Prusa-Firmware/files/6743421/1_75mm_MK3-EINSy10a-E3Dv6full-1279a6c.zip)

ericflor, 2021-07-01:
got the same result on this firmware version where it just got stuck on the welcome screen like in the second video I sent.

wavexx, 2021-07-01:
Ok.. that's new. And quite puzzling.
@leptun I'd try next with the hwtim PR, unless you have better ideas?

wavexx, 2021-07-01:
@ericflor what happens if you remove the SD card before resetting? Any change?

ericflor, 2021-07-07:
sorry for the late reply here. tested this out, removing the SD card, refreshing with the new firmware and I'm still getting the same issue. actually in the process of downgrading the rest of our 70 printers to 3.9.2 since we've been having a lot of issues the last several weeks with print quality. I'll create a new bug report with that data today.

wavexx, 2021-07-19:
@ericflor Any chance we could try to debug this interactively somehow? The knob not working might be hinting at some other low-level issue that really needs to be fixed properly.

AcidWinter, 2021-08-08:
I'm also experiencing this issue on my MK3S when upgrading to 3.10.0.

I see only the loading screen and main info screen, but can't click into any menus. I still hear the click noise when I press the wheel.

I do see a difference when loading with/without an SD card. When loading with an SD I get to the same point in the logs as @ericflor 

```Connecting...
start
Printer is now online.
echo: 3.10.0-4481
echo: Last Updated: May  6 2021 13:52:17 | Author: (none, default config)
Compiled: May  6 2021
echo: Free Memory: 2054  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Extruder fan type:
NOCTUA
CrashDetect ENABLED!
FSensor DISABLED (sensor board revision:unknown state)
Sending 0xFF
echo:SD card ok
echo:busy: processing
LCD status changed
```

If I remove the SDcard during boot, I get a different error and never get to the info screen:

```start
echo: 3.10.0-4481
echo: Last Updated: May  6 2021 13:52:17 | Author: (none, default config)
Compiled: May  6 2021
echo: Free Memory: 2054  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Extruder fan type:
NOCTUA
CrashDetect ENABLED!
FSensor DISABLED (sensor board revision:unknown state)
echo:SD init fail
```

wavexx, 2021-08-08:
On Sun, Aug 08 2021, AcidWinter wrote:
> I do see a difference when loading with/without an SD card. When loading
> with an SD I get to the same point in the logs as @ericflor

Can you try to do both (with/without SD) using the latest 3.10.1-RC1
firmware?

JohnnyDeer, 2021-08-23:
This issue will be closed soon if autor will not respond.

wavexx, 2021-08-23:
Please don't close this prematurely. This is rare, but needs to be fixed.
@AcidWinter any news?

AcidWinter, 2021-08-23:
Checking now. Will update shortly.

AcidWinter, 2021-08-23:
Still seeing the same behavior in 3.10.1-RC1.

with the SD:
```
Connecting...
start
Printer is now online.
echo: 3.10.1-RC1-4587
echo: Last Updated: Jul 16 2021 07:49:31 | Author: (none, default config)
Compiled: Jul 16 2021
echo: Free Memory: 2047  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Extruder fan type:
NOCTUA
CrashDetect ENABLED!
FSensor DISABLED (sensor board revision:unknown state)
Sending 0xFF
echo:SD card ok
LCD status changed
```

and without the SD:

```
start
Printer is now online.
echo: 3.10.1-RC1-4587
echo: Last Updated: Jul 16 2021 07:49:31 | Author: (none, default config)
Compiled: Jul 16 2021
echo: Free Memory: 2047  PlannerBufferBytes: 1760
echo:Stored settings retrieved
adc_init
Extruder fan type:
NOCTUA
CrashDetect ENABLED!
FSensor DISABLED (sensor board revision:unknown state)
echo:SD init fail
```

LR-J, 2021-11-10:
I try to update my MK3 with the 3.10.0 firmware and I got the discribed behaviour. LCD screen locked to the start message.
Downgrad to 3.9.3 is ok.

closed 2023-11-14 by github-actions[bot]