prusa3d/Prusa-Firmware #3218: [QUESTION] should we remove ` EEPROM_save_B()`  and ` EEPROM_read_B()` ?

gudnimg, 2021-07-01:
**What is your question?**

I am thinking about if we can remove these two functions from the code? This is little endian if I am not mistaken.

```cpp
// Save a single axis babystep value.
void EEPROM_save_B(int pos, int* value)
{
  eeprom_update_byte((unsigned char*)pos, (unsigned char)((*value) & 0xff));
  eeprom_update_byte((unsigned char*)pos + 1, (unsigned char)((*value) >> 8));
}

// Read a single axis babystep value.
void EEPROM_read_B(int pos, int* value)
{
  *value = (int)eeprom_read_byte((unsigned char*)pos) | (int)(eeprom_read_byte((unsigned char*)pos + 1) << 8);
}
```
and use `eeprom_update_word()` instead.

Doing a search on the entire code I see these two functions are use for these EEPROM variables:

| Variable | Type| 
|------------------------------------|---------------|
| `EEPROM_BABYSTEP_X`               |   `uint16_t`   |
| `EEPROM_BABYSTEP_Y`               |   `uint16_t`   |
| `EEPROM_BABYSTEP_Z`               |   `uint16_t `  | 
| `EEPROM_BABYSTEP_Z0 `            |    `uint16_t`  | 
| `EEPROM_BOWDEN_LENGTH`     |    `uint32_t ` |
| `EEPROM_PROBE_TEMP_SHIFT`   |    `int16_t` |

After I change:

`EEPROM_read_B()` --> `eeprom_read_word()`
`EEPROM_save_B()` --> `eeprom_update_word()`

SRAM usage is 8 bytes lower
Flash usage is 258 bytes lower

Btw I think listing `EEPROM_BOWDEN_LENGTH` as `uint32_t` is misleading. It is not consistent with how  `EEPROM_PROBE_TEMP_SHIFT`  is defined which includes 5 ints.

wavexx, 2021-07-02:
I think, just like EEPROM_read/save in your PR #3206 these are duplicate functions which were just forgotten.
IMHO getting rid of them is a good idea also to avoid new code using them by mistake

wavexx, 2021-07-02:
Also EEPROM_BOWDEN_LENGTH is not really used, as it's part of SNMM which is essentially not supported anymore.

gudnimg, 2021-07-02:
> I think, just like EEPROM_read/save in your PR #3206 these are duplicate functions which were just forgotten.
> IMHO getting rid of them is a good idea also to avoid new code using them by mistake

I can add these changes to that PR (https://github.com/prusa3d/Prusa-Firmware/pull/3206 ) to keep the changes within the same PR review/place.

wavexx, 2021-07-02:
Sounds great to me :)

gudnimg, 2021-07-05:
Closing issue since I’ve made a PR and question is answered.

commit 13477c65 (2022-02-15): Remove two functions EEPROM_read_B() and EEPROM_save_B()

This resolves #3218
closed 2021-07-05 by gudnimg