prusa3d/Prusa-Firmware #3222: [BUG] FW 3.10.10 continue to print with nozzle temperature set to 0°

antimix, 2021-07-04:
**Printer type** - MK3S
**Printer firmware version**-  prusa3d_fw_MK3S_3_10_0_4481

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version 1.0.6

**SD card : FlashAir

Just happened. After a long print of 5h, with about another hour to finish, I was still around the printer, before going bed.
Suddenly I heard some strange click noise. I went to have a look and:

- The printer was still printing without any error on the LCD.
- The last 6 mm in height of the objects were just very lite filament balls & mess.
- The bed was at 60°/60°
- **The nozzle was 150°/0°** dropping down the temperature, since it was becoming cold.

For an unknown reason the printer behave as it someone (gcode ???) set the nozzle temperature to 0°.
The firmware did not noticed that the nozzle temperature was set to 0°, and instead of giving an alarm about an inconsistent parameter or situation, it was continuing to extrude PLA, and print normally , even if it was becoming hard in the nozzle,  blocking the printer.
I paused the printer, unscrewed the extruder screw and performed a filament cold pull, since the nozzle was still around 140°.
Then I aborted the print.

I checked the gcode, and I could not find any temperature set to 0 in the middle of the print. The gcode was produced by PrusaSlider 2.3.1 win64. 
This has lowered the reliability of the Prusa regarding leaving it unattended during nightly long prints.

**To Reproduce**
No idea if this can be reproduced or it was a crazy firmware moment.
  
**Expected behavior**
At least an error message on the LCD and the print paused.
  
**G-code**
You can find it attached.

**Video**
I forgot to film :(

[Tubi Fotografici_0.2mm_MK3SMMU2S_5h47m.zip](https://github.com/prusa3d/Prusa-Firmware/files/6760717/Tubi.Fotografici_0.2mm_MK3SMMU2S_5h47m.zip)

leptun, 2021-07-05:
That should not have been possible. The firmware should have stopped extruding if the nozzle temperature dropped below 175C. It could very well be related to the memory corruption issue that has been observed in this fw version. If this is the case, a fix is on its way. You can read more about this fix here: #3202. Here is a patched version of the firmware that in theory could fix this. It is 3.10.1 before release. https://github.com/prusa3d/Prusa-Firmware/issues/3147#issuecomment-869203070

antimix, 2021-07-05:
I just downloaded the 3.10.1 test. I will REPRINT the same exact g-code. I will see what happens.

antimix, 2021-07-07:
Ok, flashing the 3.10.1 and reprinting THE SAME g-code, the printer was able to complete the 5hrs print.
It looks good.

broodro0ster, 2021-07-14:
Similar issue here with 3.10. I printed a part with a pause halfway. After hitting resume the printer wasn't extruding anymore and making print moves in the air. 
I thought it was heatcreep at first, but then I noticed my nozzle wasn't heated anymore and it didn't print a single layer after the pause.
I will try 3.10.1 as well and print the same g-code again.

JohnnyDeer, 2021-07-29:
@broodro0ster have you tried 3.10.1? What is the result?

broodro0ster, 2021-07-29:
> @broodro0ster have you tried 3.10.1? What is the result?

Not yet. I had to finish a few parts where I needed the pause and went back to 3.9.3. I need to test 3.10.1 when I have some more time.

JohnnyDeer, 2021-08-19:
This issue will be closed soon if there won't be responses.

Breakingnewgtty, 2021-08-21:
I have the same problem, I use Octoprint, and after upgrading to the new FW 3.10.1  the problem is not solved.

SgtSeppel, 2021-09-07:
Could this be related to a pause command?
I have a repeating issue with 3.10.1: When using a pause command from PrusaSliver, to add some magnets into the print, the print is paused and the Etemp is set to 0.
I have to manually resume the print and the temp will stay at 0. Only setting the temp via Octoprint (Settings Menu not visible in paused mode), I can somewhat successfully resume the print.

kean-stump, 2022-01-20:
I see the same behavior reported by antimix with MK3S+ FW 3.10.1-4697.

CarrollDiehl, 2022-06-14:
I had the same behaviour at my Prusa i3 MK3S+. I initially thought that it would have timed out, but on the second test M601 immediately set the temperature of the printer to 0. Used Octoprint to fix the issue. The printer turned unresponsive (blue screen) and continued printing. For me, the extruder stopped extruding at 170°C but the printhead still continued its program.
Very strange bug :o

Prusa-Support, 2022-06-14:
Hello! Thank you for the reports, so it did occur with 3.10.1 - was it also with FlashAir SD card? Or "normal" SD card, or Octoprint? @CarrollDiehl was your issue with 3.11.0 or still previous? Would updating to 3.11.0 solve the issue? In case, can you please send me the GCodes where the issue occurred? 

Alessandro Pantaleo
Prusa Research

CarrollDiehl, 2022-06-17:
Oh I'm sorry, will come back to you asap!

closed 2023-11-14 by github-actions[bot]