prusa3d/Prusa-Firmware #3224: [ENHANCEMENT] Add repetition limit on enforcing Z calibration during mesh bed leveling

jtebert, 2021-07-06:
During mesh bed leveling, if there is an error with the z-height [("Enforcing the Z calibration")](https://help.prusa3d.com/en/article/mesh-bed-leveling_112163/#enforcing-the-z-calibration), this can go on indefinitely. For example, if the magnetic bed is misaligned and not flat (eg, it is accidentally on a screw in the back of the print bed), mesh bed leveling will continuously re-try and enforce the z calibration. Today, I can into work and found that the printer had been running the mesh bed/z-enforcement for 20 hours.

The obvious solution to this is a timeout or limit on the repetitions. For example, it will enforce the Z calibration no more than five times. If it hits that limit, it turns off the heaters and prompts the user on the screen.

(This occurred on a MK3S with firmware version 3.10.0-4481)

mix579, 2023-05-13:
Just ran into this today and can't believe this is still open. Started a print in the evening, checked on it this morning and found it was mesh bed leveling. Huh? Noticed that the still sheet was slightly misaligned, so it ran into a calibration error, enforced Z-leveling, and apparently had been doing this for 7 hours ☹️

Please add a simple counter, stop the print, and throw and error after 3 attempts.

FW 3.12.0.

leptun, 2023-05-13:
I am preparing some significant changes to the MBL for fw 3.14. I'll keep this issue in mind and address it alongside the other changes.

leptun, 2023-05-13:
I added a fix for this to my PR (#4183). It is now limited to 2 failed alignments at the top.

closed 2023-09-13 by 3d-gussner