prusa3d/Prusa-Firmware #3225: [BUG] 3.10.1-pre weird surface imperfection at specific point on model

tg73, 2021-07-07:
MK3S (with superpinda)
FW 3.10.1 pre-release from https://github.com/prusa3d/Prusa-Firmware/issues/3147#issuecomment-869203070
MMU2S FW 1.0.6
Octoprint

**Describe the bug**
I updated from 3.9.3 to 3.10.1-pre. Using the exact same gcode, the last print I did with 3.9.3 and the first with 3.10.1-pre, I get a surface imperfection at a specific point on each instance of the model, which is repeated in a 3x3 grid on the plate. The imperfection appears at the same place on all 9 of the models in the grid. The 3.9.3 print had no such imperfection. I've reverted to 3.9.3 and am trying the print again. gcode and 3mf attached. Using Prusament PC blend.

A selection from the 9 parts from one print with 3.10.1-pre:
[image]

A selection from the 9 parts from one print with 3.9.3:
[image]

[Files.zip](https://github.com/prusa3d/Prusa-Firmware/files/6775568/Files.zip)

tg73, 2021-07-11:
See also possibly related #3226,

DRracer, 2021-07-19:
@tg73 could you please verify this with FW 3.10.0. From the changes done between 3.9.3 and 3.10.1-RC1 we don't see any hint what may have caused this issue.

@wavexx any ideas?

wavexx, 2021-07-19:
@tg73 can you reproduce this on multiple printers by chance?

tg73, 2021-07-19:
I should be able to look at this tomorrow. I will try to reproduce with
something other than PC as I have a very limited supply. I will try with
3.9.3, 3.10.0 and 3.10.1-RC1. Unfortunately I only have one printer.

tg73, 2021-07-19:
If I can reproduce it, then my first hunch based solely on the release
notes and general experience rather than familiarity with the code would be
an unintended rounding, quantisation or boundary condition introduced by
some of the data type changes/cleanup.

wavexx, 2021-07-19:
We didn't change much on the planning side since 3.9.2 AFAIK. The only thing that comes to mind on this front is the support for separate travel acceleration (pr #2838) which however shouldn't really do much unless the gcode is changed.

I'm going to try to print your gcode and see if I can get the same though. Does it also happen if you just print the model a single time?

tg73, 2021-07-19:
I don't know re single object. I was under time pressure to produce the
parts in quantity, so I just went back to 3.9.3 and printed several further
batches with no artifacts. I was going to try a single object tomorrow. But
the artifact occurs at exactly the same location in all the parts when
printed in a 3x3 grid, so hopefully a single object at the centre position
might reproduce the issue - but I don't know for sure

wavexx, 2021-07-19:
If you can test if this reproduces with a single object it would be great, as it does simplify testing a lot. Testing the model from #3226 is much harder, as we could have some extra things going on there (cooling for example).

It would also be helpful to know if the object in the dead center has the issue, or you can also get it if you shift it slightly, or what happens if you just rotate it (does the spot rotate too)?

tg73, 2021-07-20:
A single object with PETG, with 3.10.1-pre (see OP for link) is clean, which is unfortunate. I'm contemplating what to try next. If I had PC blend to spare, I'd test with that, but I don't. Maybe I should try with the PC profile, but with temperatures reduced. It will be stringy, but hopefully the gcode will be closer to the PC gcode.

wavexx, 2021-07-20:
On Tue, Jul 20 2021, Tom Glastonbury wrote:
> A single object with PETG, with 3.10.1-pre (see OP for link) is clean,
> which is unfortunate. I'm contemplating what to try next. If I had PC
> blend to spare, I'd test with that, but I don't. Maybe I [...]

If we go with the FW related issue, just change the temperature and
print PLA or PETG. The issue *should* remain.

tg73, 2021-07-20:
A single object printed in PETG with PC profile but reduced temps (set in PS) - clean. I will try printing the exact 3x3 grid PC gcode as attached in OP, but tune the temps down on the printer and use PETG. I'll start with 3.10.1-pre as this is the FW that I saw the problem with.

tg73, 2021-07-21:
A 3x3 grid printed from the same gcode as in the original post, with FW 3.10.1-pre, using PETG and manually tuning the temperatures to PETG via LCD - clean, all parts good. I'm not sure what to try next.

tg73, 2021-07-21:
@DRracer The imperfection does not correspond to a seam or apparently anything of geometric significance. The imperfection appears at around 45 deg clockwise from the modelled indented rectangle (towards the bottom of the image below), in the layers immediately above the modelled indent. It's almost like an echo of the modelled indent. While there is an infill feature in the right area, similar infill features occur all over the model, but the imperfection only occurs in one place.

[image]

tg73, 2021-07-21:
@wavexx I also just tried doing 4 objects in PC with FW 3.10.1-pre, and it was clean. So unfortunately this does not seem to be an easily-repeatable behaviour. I will be printing more of these objects in PC for a customer job, which I'll do with 3.10.1-RC1. If I see the issue happening during a print (I can't guarantee to watch all print runs at the right time though) I'll do a memory dump. Any other suggestions welcome.

DRracer, 2021-07-22:
@tg73 ok, thank you for the additional explanation. If it is caused by the firmware, it must be visible on simulated print. I'll run a session during the day to see, if there is a 45 deg offsetted "echo" of the modelled rectangular inset and report back.

DRracer, 2021-07-23:
@tg73 We managed to run the simulations yesterday, but couldn't find any trace of your issue with either FW 3.10.1-RC1 or 3.9.3. Moreover, the simulated extrusion chunks came out almost identical in both firmwares, i.e. no missing material/echo at those 45deg.

Just to rule out the influence of Octoprint, could you try printing just from the SD card without Octoprint attached?

tg73, 2021-07-23:
@DRracer Given that I've been unable to reproduce the issue even *with* octoprint, which was in use for the original problematic print, I'm not sure that printing from SD will help. But as noted [above](https://github.com/prusa3d/Prusa-Firmware/issues/3225#issuecomment-884175344), I've got a few repeats of the original print coming up, and I'll do those from SD so that if there is a repeat, we know that it wasn't octoprint's fault.

Also, below is a macro closeup of the issue. I'm not used to looking at closeups like this, but to my untrained eye it kind of looks like there was a small glitch across all axes (XYZE) on each of the affected layers - you see apparent deviation in X, Y and Z, and some pulses of overextrusion and perhaps underextrusion. Very odd.

[image]

tg73, 2021-07-23:
Interestingly, I just started a 3x3 PC print from SD, and I noticed on LCD that tuned speed had set itself to 102% (I noticed during 1st layer, all brims printed, was on 1st or 2nd object 1st layer, I don't know exactly when it become 102%). I had not tuned anything. I've set it back to 100% and let the print continue. FW 3.10.1-RC1.

DRracer, 2021-07-23:
@tg73 thank you for your update and the detailed photo - this looks like the extruder is hitting something - the layer is not straight ... if you get to this spot - please have a look at PINDA and fan shroud if they are accidentally hitting anything.

So far we have confirmed the simulation doesn't show any computational errors and there is no seam in the sliced G-code.

tg73, 2021-07-23:
The 3x3 PC print from SD mentioned in my last post completed successfully, no problem. Note that I normally take the fan shroud off for PC prints, although there is a very small chance that it was in place for the original problematic print - I can't remember and don't have any photos to check. But I agree, it would in theory explain the behaviour. As regards PINDA, it does not catch. I need to print another 4 plates of 9 parts, in PC, which I'll do from SD. If there's no repeat of the problem during those prints, I suggest that it's then worth closing the issue on the likelihood that it was a fan shroud issue.

I did take a memory dump from around the point in the print where a problem might occur. I've attached it in case it sheds any light on the spontaneous 102% speed behaviour mentioned above. Interestingly, the printer hung on the boot screen when I tried to connect octoprint, fans at full speed. After a while I tried connecting again, and the printer rebooted and connected successfully.

[crashdump1.txt](https://github.com/prusa3d/Prusa-Firmware/files/6870297/crashdump1.txt)

DRracer, 2021-07-29:
@tg73 any update on this issue?

tg73, 2021-07-29:
I had to divert to printing replacement parts for the printer itself in PCCF as PETG parts were deforming due to sustained high temperature printing (y-axis belt clamp, extruder parts). I still expect to complete the remaining 3x3 grid PC prints as per my last comment before the end of the weekend, but so far no repeat of the problem.

DRracer, 2021-07-30:
OK, thank you for update on this issue, we'll wait for your next report if everything is still ok.

JohnnyDeer, 2021-08-19:
This issue will be closed soon if autor will not respond.

tg73, 2021-08-19:
Apologies for the delay. I've been unable to repeat the problem and will close the issue.

closed 2021-08-19 by tg73