prusa3d/Prusa-Firmware #3226: [BUG]<3.10 firmware causing poor print quality>

ericflor, 2021-07-07:
**Printer type** - MK3/S (40-50 printers with the same data)
**Printer firmware version**-  3.10.0
**SD card**
 
A few weeks ago we started getting a lot of issues with print quality, specifically a hopper we sell through our business. As seen in the picture, it has some relatively steep angles but over the last year and a half that we've been selling it, we've been able to get perfect quality out of these almost every time. We hit a pretty big heat wave two weeks ago so assumed that was the problem with the drooping over those holes. We tested a lot of different parameters; nozzle temp, speed, bridging, orientation, walls, infill, and even brought 4 of our printers into a temperature/humidity controlled room where we played with ambient temperatures between 68-72F and humidity between 20-35%. We also tested several different filaments and still no luck. Our last shot in the dark was to try and downgrade the firmware and use our original slice of this model. Came into the shop this morning to find every hopper with the 3.9.2 firmware and original slice printed totally fine. Went through and downgraded the rest of our 74 machines today and am now getting almost 100% clean prints again. Found a previous bug report talking about poor print quality with the new firmware so can only assume we're dealing with the same thing here. 

We're very curious as to what specifically is happening that would cause these errors so any insight would be much appreciated. 

[image]
[image]
[image]

  
**G-code**
[PER_HOP_BACK_V.21_14h54m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware/files/6778979/PER_HOP_BACK_V.21_14h54m.gcode.zip)

DRracer, 2021-07-19:
@ericflor this needs further investigation as there have been no changes in terms of material flow computations or movement planning between 3.9.1 and 3.10.1.

If you said transition from 3.9.0 and 3.9.1 I'd completely agree, 3.9.1 brought a fix for a bug in the linear advance algorithm present in 3.9.0. But that is apparently not your case.

Potentially related changes introduced in 3.10.0:
- double edge stepping (but if there were errors, the print will be ruined)
- new g-code filtering code when reading from the SD card (could influence timing - faster reading)
- Scaling extruder motor current based on speed - unless you really tried to enable this feature, it is disabled - i.e. no change from 3.9.3

Just to be sure we are not hitting a memory corruption bug - could you please try to run a few samples on the latest FW 3.10.1-RC1 released a couple days ago and report back? Thank you.

DRracer, 2021-07-20:
@ericflor We printed one batch of your G-code over night with stock MK3S/+ machines and FW 3.9.2.
Actually only one of the machines managed to make the inner overhangs correctly and only on one side of the print.
Now running the same batch with 3.10.1-RC1 to check for any differences

[image]

The primary issue here is the fact, that the Slicer creates starts of the paths in the overhangs' corners, which cannot result in a good result.

DRracer, 2021-07-22:
@ericflor FW 3.10.1-RC1 produced a similar result on the very same machine. Now, we are running simulations to check for any firmware-related changes if any.
Out of curiosity - can you post a photo of a successful print of your part? I'd like to see how the overhangs came out.

DRracer, 2021-07-22:
@ericflor simulations finished. The simulator (MK404) produced **identical** PLY exports (not considering the fact, that the prints were stopped at slightly different heights). I'm surprised the exports were identical, I was expecting some slight timing disrepancies in the simulator, but no - the stepping of the motors was identical both in FW 3.9.3 and 3.10.1-RC1.

After such a result I'd be surprised if the firmware itself was the real cause of your issue, in other words - we are sorry, we cannot reproduce your issue.

ericflor, 2021-08-18:
> @ericflor FW 3.10.1-RC1 produced a similar result on the very same machine. Now, we are running simulations to check for any firmware-related changes if any.
> Out of curiosity - can you post a photo of a successful print of your part? I'd like to see how the overhangs came out.

[image]

ericflor, 2021-08-18:
> @ericflor We printed one batch of your G-code over night with stock MK3S/+ machines and FW 3.9.2.
> Actually only one of the machines managed to make the inner overhangs correctly and only on one side of the print.
> Now running the same batch with 3.10.1-RC1 to check for any differences
> 
> [image]
> 
> The primary issue here is the fact, that the Slicer creates starts of the paths in the overhangs' corners, which cannot end up well.

Thanks for trying all this out, we got the same pattern with the right side typically doing fine on the overhangs but not on the left. So how could I create a custom slicing profile so that the start of the path isn't on the corner of the overhang? It's a little bizarre though still because prior to a couple months ago, we were able to print these with relatively no errors for the past year and a half that we've been selling the hopper. The only things that really changed were updating firmware and upgrading the slicer. Also yes I work as the print tech at Out Of Darts :)

DRracer, 2021-08-23:
@ericflor that's a nice print. 

Regarding the overhangs - we noticed during the print, that the G-code you posted earlier not only tries to start all paths on the overhangs, but each perimeter starts on a different overhang on one side of the print.

Personally, I'd hand-draw a seam (in PrusaSlicer 2.3+) in the inside middle of the front vertical wall (between the holes) which should instruct the slicer to avoid messing up the start of the paths on the overhangs. This seam will be hidden anyway.

_The only things that really changed were updating firmware and upgrading the slicer_ - the upgrade of the Slicer would have been my first suspicion for the real cause of this issue, but you also wrote that you tried an older G-code with the same result.

Honestly, you report is still puzzling to us as we didn't do any changes to stepping in FW 3.10.x and we also haven't been able to print your part cleanly like you posted recently.

MrWho900913, 2022-02-25:
I personally encountered issues with printer hitting already printed parts on several czech FB groups, and i suspect an issue in leveling. I bet theres some problem with segmentation of those moves and I personally think it is definitely related. Try looking at measured leveling data from printers(G81 V1 i guess). Maybe trying this model with something like fade height function in marlin could point at it more accurately. I unfortunately dont have any prusa printer to help with that.

closed 2023-11-09 by github-actions[bot]