prusa3d/Prusa-Firmware #3227: [BUG] building the firmware with SNMM enabled fails with multiple errors

gudnimg, 2021-07-10:
When I enable the SNMM and try to build the firmware it just fails. It seems to me the code just hasn't been updating in a while and now we're left with dead code.

Is there any reason to keep this code? Can we remove all SNMM code if it doesn't work anyways?

```
C:\Users\Gudni\OneDrive\Desktop\Prusa-Firmware\Firmware\mmu.cpp: In function 'void extr_adj(uint8_t)':
mmu.cpp:1045:2: error: 'lcdDrawUpdate' was not declared in this scope
  lcdDrawUpdate = 2;
  ^~~~~~~~~~~~~
C:\Users\Gudni\OneDrive\Desktop\Prusa-Firmware\Firmware\mmu.cpp:1045:2: note: suggested alternative: 'lcd_draw_update'
  lcdDrawUpdate = 2;
  ^~~~~~~~~~~~~
  lcd_draw_update
In file included from C:\Users\Gudni\OneDrive\Desktop\Prusa-Firmware\Firmware\Configuration.h:60:0,
                 from C:\Users\Gudni\OneDrive\Desktop\Prusa-Firmware\Firmware\Marlin_main.cpp:47:
C:\Users\Gudni\OneDrive\Desktop\Prusa-Firmware\Firmware\Marlin_main.cpp: In function 'void setup()':
Configuration_prusa.h:161:23: error: array must be initialized with a brace-enclosed initializer
 #define BOWDEN_LENGTH 433 //default total length for filament fast loading part; max length for extrusion is 465 mm!; this length can be adjusted in service menu
                       ^
C:\Users\Gudni\OneDrive\Desktop\Prusa-Firmware\Firmware\Marlin_main.cpp:1504:13: note: in expansion of macro 'BOWDEN_LENGTH'
    int _z = BOWDEN_LENGTH;
             ^~~~~~~~~~~~~
C:\Users\Gudni\OneDrive\Desktop\Prusa-Firmware\Firmware\ultralcd.cpp: In function 'void lcd_commands()':
ultralcd.cpp:919:16: error: 'count_e' was not declared in this scope
   float extr = count_e(0.2, width, length);
                ^~~~~~~
ultralcd.cpp:1169:24: error: cannot convert 'int' to 'LcdCommands' in assignment
    lcd_commands_type = 0;
                        ^
exit status 1
'lcdDrawUpdate' was not declared in this scope

```

I tried fixing some of the errors manually, but new errors pop up. Undefined variables, scope issues.

gudnimg, 2021-07-10:
I am seeing similar issues when I enable `ENABLE_AUTO_BED_LEVELING`. I cannot build the firmware due to errors.

wavexx, 2021-07-13:
There are many parts enabled by conditionals which weren't maintained since a long time. SNMM is the old mixing MMU implementation. This is definitely not supported anymore: many assumptions around SNMPP got broken over time. Making SNMM working again would be a major feat of testing, and I don't think it's worth the effort unless somebody is actively maintaining it.

gudnimg, 2021-07-14:
> There are many parts enabled by conditionals which weren't maintained since a long time. SNMM is the old mixing MMU implementation. This is definitely not supported anymore: many assumptions around [...]

Is `ENABLE_AUTO_BED_LEVELING` also an old setting which mesh bed leveling has replaced?

leptun, 2021-07-14:
I think so. If I remember correctly, the auto bed leveling only uses 3 points and compensates only linearly across the whole bed. It is leftover code from the old marlin.

gudnimg, 2021-07-14:
Awesome, I would be interested to help you trim the code. It's very fun 😄 I could start by removing the broken code that fails to compile as a first step. The broken code is a good indicator for me of what is definately OK to remove.

I think we can split this into two issues/tasks:

* Remove `ENABLE_AUTO_BED_LEVELING`
    * 24 #ifdef's 
    * Possibly remove `AUTO_BED_LEVELING_GRID`
    * Possibly remove `Z_PROBE_REPEATABILITY_TEST`
    * Remove files `qr_solve.h` and `qr_solve.cpp`
    * Remove files `vector_3.h` and `vector_3.cpp`
* Remove  `SNMM`
    * 33 #ifdef's

leptun, 2021-07-14:
https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/uni_avr_rpi.h Should be deleted and deprecated as well.

gudnimg, 2021-07-14:
Is `NUM_SERVOS` used for anything today? I suspect we can remove `Servo.cpp` and `Servo.h` as well.

gudnimg, 2021-07-21:
Some more thoughts on dead code:

## SWSPI
I looked into https://github.com/prusa3d/Prusa-Firmware/blob/MK3/Firmware/uni_avr_rpi.h and it is used by `swspi.cpp` and `swispi.h` which are only available if `__SWSPI` is defined, but it is never defined anywhere.

The functions in `swispi.cpp` are used in `pat9125.c` but the code tries to use the functions if `PAT9125_SWSPI `is defined, but `__SWPI` is never defined so the code won't work.

## LCD_TEST

The #ifdef `LCD_TEST` is also never defined anywhere.
https://github.com/prusa3d/Prusa-Firmware/blob/400f673fe0b898d0e3ec8c43fd2011f9320cf20c/Firmware/ultralcd.cpp#L6331

GilesBathgate, 2021-12-10:
The unix tool `unifdef` works really well for this kind of thing.

`grep SNMM * -Rl | xargs unifdef -USNMM -m`

closed 2023-01-30 by gudnimg