prusa3d/Prusa-Firmware #3231: Brim runs over skirt. Skirt redundant when using brim

Spinnetti, 2021-07-15:
MK3
3.10.0.4481
SD

When selecting both brim and skirt, the skirt prints and the brim prints (or can print) over the skirt
  
**Expected behavior**
  Either:
     a) the skirt should print outside of the brim range, or 
     b) (preferred) the skirt is suppressed since the brim will do the job of the skirt anyway.

3d-gussner, 2021-07-15:
@Spinnetti This is PrusaSlicer related and not Firmware, please close this issue here and open it in https://github.com/prusa3d/PrusaSlicer/issues

neophyl, 2021-07-15:
There is already a report for that in PS and it is intended operation apparently as it's under user control from their setting choices.

DRracer, 2021-07-19:
Not firmware related, closing

closed 2021-07-19 by DRracer