prusa3d/Prusa-Firmware #3233: Material change does not set temp to new material

Spinnetti, 2021-07-15:
MK3
3.10.0.4481
SD

**Describe the bug**
If I unload material e.g. PLA and then load ABS (or other), when I click load the ABS, it makes a double click noise and tries to load the ABS even though the temp isn't right and jams. 
  
**To Reproduce**
Switch dissimilar materials
  
**Expected behavior**
It should ask what material I'm loading and prep the temp for that

3d-gussner, 2021-07-15:
I think there is already an reported issue. Please search the issues before creating a new one.

Danafra, 2021-08-20:
I haven't found the duplicate to this. Can someone give the issue number?

Spinnetti, 2021-08-20:
I didn't see a duplicate either. There's several undesirable behaviors when
I unload material
1. It heats the bed and the extruder instead of just the extruder -
shouldn't assume I want to print: function should be atomic and just do
what I asked.
2. After I reload, it stays hot even if I'm not printing anything. Again,
shouldn't assume I immediately want to print; should shut off.
3. Most critically, I unload usually when I change material types, but if I
unload, remove then immediately reload, it beeps twice and immediately
tries to reload without asking me which material type, jamming up the
extruder.

closed 2023-11-09 by github-actions[bot]