prusa3d/Prusa-Firmware #3234: Selecting any function after selecting load filament, the firmware crashes

Spinnetti, 2021-07-15:
MK3
3.10.0.4481
SD

**Describe the bug**
If I do an action like load filament, then pick another action, the software crashes and restarts (need to confirm specific action sequences). 
  
**To Reproduce**
From cold start, select load filament, then as its heating, select another function
  
**Expected behavior**
 Should have an option to cancel, and  disallow any actions other than <Cancel> till the current operation completes

3d-gussner, 2021-07-15:
I think there is already an reported issue. Please search the issues before creating a new one.

DRracer, 2021-07-19:
Please try on the latest firmware: 3.10.1-RC1

Spinnetti, 2021-07-20:
Will do, thanks

DRracer, 2021-08-09:
@Spinnetti did FW 3.10.1-RC1 solve your issue?

Spinnetti, 2021-08-09:
Thanks for the follow up. Sorry, haven't installed yet, will go download.

JohnnyDeer, 2021-08-23:
This issue will be closed soon if autor will not respond.

Spinnetti, 2021-08-23:
This is resolved. Now only allows cancel function as one would expect.

Spinnetti, 2021-08-29:
With the latest firmware, still crashes. To Reproduce:

1. Select Load filament
2. While loading, I selected "Print from SD"

Result: Firmware crash and reboot.

leptun, 2021-08-29:
@wavexx Could this be stack smashing again because of st_sync?

wavexx, 2021-08-30:
Yeah might be. While the filament is loading, we're running another loop which allows menu selection.
If I understand right correct sequence, it's:

- Load filament
- If preheat is needed, allow it to finish
- Filament starts to load
- Select "Print from SD"

Spinnetti, 2021-11-20:
Latest update fixed.. Only option after preheat is cancel which is a "good enough" solution.

closed 2021-11-20 by Spinnetti

wavexx, 2022-09-06:
This should be handled correctly with #3564 (feedback would be welcome)

Spinnetti, 2022-09-06:
Thanks for that... Will check.