prusa3d/Prusa-Firmware #3235: Material unload has unnecessary actions

Spinnetti, 2021-07-15:
MK3
3.10.0.4481
SD

**Describe the bug**
If I select "unload filament" it heats both the nozzle and bed - why? It presupposes I want to print now, and with the same material, and heating the bed takes longer (and wastes energy). Just heat the nozzle. Further, when its done, it doesn't shut off, presupposing I am going to print immediately and with the same material and will just sit there at temp for both the nozzle and bed till it hits timeout wasting energy and potentially increasing a chance of nozzle clogging with the hot filament in the heat zone for an extended period. 
  
**To Reproduce**
  Please describe steps to reproduce the behavior.
  
**Expected behavior**
a) Heat only the nozzle then follow the usual steps
b) Once the user has confirmed unloading, shut off the heating elements and return to the main menu

3d-gussner, 2021-07-15:
I think there is already an reported issue. Please search the issues before creating a new one.

JohnnyDeer, 2021-07-15:
Hi @Spinnetti, as @3d-gussner wrote this issue is duplicate of #422 #2295. I´m closing this one.

closed 2021-07-15 by JohnnyDeer